esphome/issues #552: Camera not initializing after deep sleep

roofuskit, 2019-07-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu Server Running Latest HA 96.5 in a docker container, ESPHome Dashboard is also running in docker on a different machine Also Running Ubuntu Server.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP-32s The Geekcreit ESP32-cam board from Bangood, operates the same as the AI-Thinker board and that is the code I am using.

**Affected component:**

Camera

https://esphome.io/components/esp32_camera.html

**Description of problem:**
Sometimes when coming out of deep sleep the camera never loads. There are no errors in the log and I cannot find any consistency in the issue except that it happens sometimes when coming out of deep sleep.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
# Example configuration entry
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

  # Image settings
  name: Front Door
  # ....
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
[15:44:24][C][wifi:372]: WiFi:
[15:44:24][C][wifi:254]:   SSID: [redacted]
[15:44:24][C][wifi:255]:   IP Address: 192.168.1.244
[15:44:24][C][wifi:257]:   BSSID: [redacted]
[15:44:24][C][wifi:258]:   Hostname: 'doorbellcam'
[15:44:24][C][wifi:262]:   Signal strength: -44 dB ▂▄▆█
[15:44:24][C][wifi:263]:   Channel: 1
[15:44:24][C][wifi:264]:   Subnet: 255.255.255.0
[15:44:24][C][wifi:265]:   Gateway: 192.168.1.1
[15:44:24][C][wifi:266]:   DNS1: 0.0.0.0
[15:44:24][C][wifi:267]:   DNS2: 0.0.0.0
[15:44:24][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'doorbell'
[15:44:24][C][gpio.binary_sensor:016]:   Pin: GPIO12 (Mode: INPUT_PULLUP, INVERTED)
[15:44:24][C][gpio.output:010]: GPIO Binary Output:
[15:44:24][C][gpio.output:011]:   Pin: GPIO4 (Mode: OUTPUT)
[15:44:24][C][logger:137]: Logger:
[15:44:24][C][logger:138]:   Level: DEBUG
[15:44:24][C][logger:139]:   Log Baud Rate: 115200
[15:44:24][C][logger:140]:   Hardware UART: UART0
[15:44:24][C][light:174]: Light 'Doorcam Light'
[15:44:24][C][esp32_camera:043]: ESP32 Camera:
[15:44:24][C][esp32_camera:044]:   Name: Front Door
[15:44:24][C][esp32_camera:045]:   Board Has PSRAM: YES
[15:44:24][C][esp32_camera:047]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[15:44:24][C][esp32_camera:048]:   VSYNC Pin: 25
[15:44:24][C][esp32_camera:049]:   HREF Pin: 23
[15:44:24][C][esp32_camera:050]:   Pixel Clock Pin: 22
[15:44:24][C][esp32_camera:051]:   External Clock: Pin:0 Frequency:20000000
[15:44:24][C][esp32_camera:052]:   I2C Pins: SDA:26 SCL:27
[15:44:25][C][esp32_camera:053]:   Reset Pin: -1
[15:44:25][C][esp32_camera:074]:   Resolution: 640x480 (VGA)
[15:44:25][C][esp32_camera:097]:   JPEG Quality: 10
[15:44:25][C][esp32_camera:099]:   Contrast: 0
[15:44:25][C][esp32_camera:100]:   Brightness: 0
[15:44:25][C][esp32_camera:101]:   Saturation: 0
[15:44:25][C][esp32_camera:102]:   Vertical Flip: ON
[15:44:25][C][esp32_camera:103]:   Horizontal Mirror: ON
[15:44:25][C][esp32_camera:120]:   Test Pattern: NO
[15:44:25][C][ota:029]: Over-The-Air Updates:
[15:44:25][C][ota:030]:   Address: 192.168.1.244:3232
[15:44:25][C][api:103]: API Server:
[15:44:25][C][api:104]:   Address: 192.168.1.244:6053
[15:44:25][C][mqtt:051]: MQTT:
[15:44:25][C][mqtt:053]:   Server Address: 192.168.1.213:1883 (192.168.1.213)
[15:44:25][C][mqtt:054]:   Username: [redacted]
[15:44:25][C][mqtt:055]:   Client ID: [redacted]
[15:44:25][C][mqtt:057]:   Discovery prefix: 'homeassistant'
[15:44:25][C][mqtt:058]:   Discovery retain: YES
[15:44:25][C][mqtt:060]:   Topic Prefix: 'doorbellcam'
[15:44:25][C][mqtt:062]:   Log Topic: 'doorbellcam/debug'
[15:44:25][C][mqtt:065]:   Availability: 'home/doorbell/motion'
[15:44:25][C][mqtt:551]: MQTT Message Trigger:
[15:44:25][C][mqtt:552]:   Topic: 'home/doorbell/active'
[15:44:25][C][mqtt:553]:   QoS: 0
[15:44:25][C][mqtt.light:053]: MQTT Light 'Doorcam Light':
[15:44:25][C][mqtt.light:054]:   State Topic: 'doorbellcam/light/doorcam_light/state'
[15:44:25][C][mqtt.light:054]:   Command Topic: 'doorbellcam/light/doorcam_light/command'
[15:44:25][C][mqtt:551]: MQTT Message Trigger:
[15:44:25][C][mqtt:552]:   Topic: 'home/doorbell/active'
[15:44:25][C][mqtt:553]:   QoS: 0
[15:44:25][C][mqtt.binary_sensor:018]: MQTT Binary Sensor 'doorbell':
[15:44:25][C][mqtt.binary_sensor:019]:   State Topic: 'doorbellcam/binary_sensor/doorbell/state'
[15:44:25][C][deep_sleep:020]: Setting up Deep Sleep...
[15:44:25][C][deep_sleep:026]:   Run Duration: 120000 ms
[15:44:25][C][deep_sleep:030]:   Wakeup Pin: GPIO13 (Mode: INPUT, INVERTED)
```

**Additional information and things you've tried:**

Due to a lack of errors I'm not sure how to resolve this issue, it seems like it's a problem with ESPHome and not any of my settings.

OttoWinter, 2019-07-28:
Some debug logs over USB could be useful - the OTA logs don't contain all info from the startup process, so connect the device via USB and replicate it through that please. Probably the esp camera driver is just not meant to be used with deep sleep functionality, but that would be an external issue in https://github.com/espressif/esp32-camera

glmnet, 2019-09-14:
@pavolholes are you using deep sleep component? Can you share yaml In case there is something relevant?

roofuskit, 2020-05-04:
> Some debug logs over USB could be useful - the OTA logs don't contain all info from the startup process, so connect the device via USB and replicate it through that please. Probably the esp camera [...]

`[0;32m[I][deep_sleep:067]: Beginning Deep Sleep[0m
ets Jun  8 2016 00:22:57
07:32:18.324 -> 
07:32:18.324 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
07:32:18.324 -> configsip: 0, SPIWP:0xee
07:32:18.324 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
07:32:18.324 -> mode:DIO, clock div:2
07:32:18.358 -> load:0x3fff0018,len:4
07:32:18.358 -> load:0x3fff001c,len:1044
07:32:18.358 -> load:0x40078000,len:8896
07:32:18.358 -> load:0x40080400,len:5828
07:32:18.358 -> entry 0x400806ac
[0;32m[I][logger:156]: Log initialized[0m
07:32:20.208 -> [0;35m[C][ota:364]: There have been 0 suspected unsuccessful boot attempts.[0m
[0;32m[I][app:028]: Running through setup()...[0m
07:32:20.242 -> [0;36m[D][binary_sensor:034]: 'doorbell': Sending initial state OFF[0m
07:32:20.242 -> [0;35m[C][light:097]: Setting up light 'Doorcam Light'...[0m
07:32:20.242 -> [0;36m[D][light:264]: 'Doorcam Light' Setting:[0m
[0;35m[C][wifi:033]: Setting up WiFi...[0m
[0;32m[I][wifi:193]: WiFi Connecting to 'Ashley'...[0m
[0;32m[I][wifi:423]: WiFi Connected![0m
07:32:21.510 -> [0;35m[C][wifi:283]:   SSID: [5m'Ashley'[6m[0m
07:32:21.510 -> [0;35m[C][wifi:284]:   IP Address: 192.168.1.244[0m
07:32:21.510 -> [0;35m[C][wifi:286]:   BSSID: [5m24:05:88:1F:A2:17[6m[0m
07:32:21.510 -> [0;35m[C][wifi:287]:   Hostname: 'doorbellcam'[0m
07:32:21.510 -> [0;35m[C][wifi:291]:   Signal strength: -51 dB [0;33m▂▄▆[0;37m█[0m[0m
07:32:21.544 -> [0;35m[C][wifi:295]:   Channel: 1[0m
07:32:21.544 -> [0;35m[C][wifi:296]:   Subnet: 255.255.255.0[0m
07:32:21.544 -> [0;35m[C][wifi:297]:   Gateway: 192.168.1.1[0m
07:32:21.544 -> [0;35m[C][wifi:298]:   DNS1: 0.0.0.0[0m
07:32:21.544 -> [0;35m[C][wifi:299]:   DNS2: 0.0.0.0[0m
07:32:21.544 -> [0;36m[D][wifi:432]: Disabling AP...[0m
07:32:21.544 -> ***ERROR*** A stack overflow in task framebuffer_tas has been detected.
07:32:21.544 -> abort() was called at PC 0x40091aa0 on core 1
07:32:21.578 -> 
07:32:21.578 -> Backtrace: 0x40091858:0x3ffba280 0x40091a89:0x3ffba2a0 0x40091aa0:0x3ffba2c0 0x4008eb6c:0x3ffba2e0 0x400907d0:0x3ffba300 0x40090786:0x40001d48
07:32:21.578 -> 
07:32:21.578 -> Rebooting...
07:32:21.578 -> ets Jun  8 2016 00:22:57
07:32:21.578 -> 
07:32:21.578 -> rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
07:32:21.578 -> configsip: 0, SPIWP:0xee
07:32:21.578 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
07:32:21.578 -> mode:DIO, clock div:2
07:32:21.954 -> load:0x3fff0018,len:4
07:32:21.954 -> load:0x3fff001c,len:1044
07:32:21.954 -> load:0x40078000,len:8896
07:32:21.954 -> load:0x40080400,len:5828
07:32:21.954 -> entry 0x400806ac
[0;32m[I][logger:156]: Log initialized[0m
07:32:23.463 -> [0;35m[C][ota:364]: There have been 1 suspected unsuccessful boot attempts.[0m
07:32:23.463 -> [0;32m[I][app:028]: Running through setup()...[0m
07:32:23.463 -> [0;36m[D][binary_sensor:034]: 'doorbell': Sending initial state OFF[0m
07:32:23.463 -> [0;35m[C][light:097]: Setting up light 'Doorcam Light'...[0m
07:32:23.497 -> [0;36m[D][light:264]: 'Doorcam Light' Setting:[0m
[E][camera.c:1049] camera_probe(): Detected camera not supported.
07:32:23.567 -> [E][camera.c:1249] esp_camera_init(): Camera probe failed with error 0x20004
07:32:23.567 -> [1;31m[E][esp32_camera:017]: esp_camera_init failed: ERROR[0m
07:32:23.567 -> [1;31m[E][component:092]: Component was marked as failed.[0m
07:32:23.567 -> [0;35m[C][wifi:033]: Setting up WiFi...[0m
[0;32m[I][wifi:193]: WiFi Connecting to 'Ashley'...[0m
[0;32m[I][wifi:423]: WiFi Connected![0m
07:32:24.597 -> [0;35m[C][wifi:283]:   SSID: [5m'Ashley'[6m[0m
07:32:24.597 -> [0;35m[C][wifi:284]:   IP Address: 192.168.1.244[0m
07:32:24.597 -> [0;35m[C][wifi:286]:   BSSID: [5m24:05:88:1F:A2:17[6m[0m
07:32:24.597 -> [0;35m[C][wifi:287]:   Hostname: 'doorbellcam'[0m
07:32:24.597 -> [0;35m[C][wifi:291]:   Signal strength: -49 dB [0;32m▂▄▆█[0m[0m
07:32:24.632 -> [0;35m[C][wifi:295]:   Channel: 1[0m
[... 54 lines cut ...]
07:32:25.144 -> [0;35m[C][mqtt:051]: MQTT:[0m
07:32:25.144 -> [0;35m[C][mqtt:053]:   Server Address: 192.168.1.213:1883 (192.168.1.213)[0m
07:32:25.179 -> [0;35m[C][mqtt:054]:   Username: [5m''[6m[0m
07:32:25.179 -> [0;35m[C][mqtt:055]:   Client ID: [5m'doorbellcam-246f2816f7d8'[6m[0m
07:32:25.179 -> [0;35m[C][mqtt:057]:   Discovery prefix: 'homeassistant'[0m
07:32:25.179 -> [0;35m[C][mqtt:058]:   Discovery retain: YES[0m
07:32:25.179 -> [0;35m[C][mqtt:060]:   Topic Prefix: 'doorbellcam'[0m
07:32:25.179 -> [0;35m[C][mqtt:062]:   Log Topic: 'doorbellcam/debug'[0m
07:32:25.179 -> [0;35m[C][mqtt:065]:   Availability: 'home/doorbell/motion'[0m
07:32:25.214 -> [0;35m[C][mqtt:551]: MQTT Message Trigger:[0m
07:32:25.214 -> [0;35m[C][mqtt:552]:   Topic: 'home/doorbell/active'[0m
07:32:25.214 -> [0;35m[C][mqtt:553]:   QoS: 0[0m
07:32:25.214 -> [0;35m[C][mqtt.light:053]: MQTT Light 'Doorcam Light':[0m
07:32:25.248 -> [0;35m[C][mqtt.light:054]:   State Topic: 'doorbellcam/light/doorcam_light/state'[0m
07:32:25.248 -> [0;35m[C][mqtt.light:054]:   Command Topic: 'doorbellcam/light/doorcam_light/command'[0m
07:32:25.248 -> [0;35m[C][mqtt:551]: MQTT Message Trigger:[0m
[... cut ...]
07:32:25.282 -> [0;35m[C][mqtt.binary_sensor:019]:   State Topic: 'doorbellcam/binary_sensor/doorbell/state'[0m
07:32:25.282 -> [0;35m[C][deep_sleep:020]: Setting up Deep Sleep...[0m
07:32:25.316 -> [0;35m[C][deep_sleep:026]:   Run Duration: 35000 ms[0m
07:32:25.316 -> [0;35m[C][deep_sleep:030]:   Wakeup Pin: GPIO13 (Mode: INPUT, INVERTED)[0m
[0;36m[D][api.connection:583]: Client 'Home Assistant 0.109.3 (192.168.1.213)' connected successfully![0m
[0;32m[I][deep_sleep:067]: Beginning Deep Sleep[0m`

closed 2020-12-05 by stale[bot]