esphome/issues #554: ESP32 Mqtt not pushing temperature value

Asinke, 2019-07-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-Wrover 4Gb

**Affected component:**

https://esphome.io/components/mqtt.html

**Description of problem:**

Temperature and humidity values not published to MQTT Broker

**Problem-relevant YAML-configuration entries:**
```yams
esphome:
  name: temperatures
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "TGV"
  password: "0651345721"

mqtt:
  broker: m24.cloudmqtt.com
  port: 16193
  username: ernfgddp
  password: 6VY6d9j6nQ5X
  topic_prefix: xiaomi/sensors

esp32_ble_tracker:

sensor:
  - platform: xiaomi_mijia
    mac_address: '4C:65:AB:DD:16:7E'
    temperature:
      name: living temperature
    humidity:
      name: living humidity
    battery_level:
      name: living battery Level
 
# Enable logging
logger:

ota:

```

**Logs (if applicable):**

```
[17:50:43][C][logger:137]: Logger:
[17:50:43][C][logger:138]:   Level: DEBUG
[17:50:43][C][logger:139]:   Log Baud Rate: 115200
[17:50:43][C][logger:140]:   Hardware UART: UART0
[17:50:43][C][esp32_ble_tracker:459]: BLE Tracker:
[17:50:43][C][esp32_ble_tracker:460]:   Scan Interval: 300 s
[17:50:43][C][xiaomi_mijia:012]: Xiaomi Mijia
[17:50:43][C][xiaomi_mijia:013]:   Temperature 'living temperature'
[17:50:43][C][xiaomi_mijia:013]:     Unit of Measurement: '°C'
[17:50:43][C][xiaomi_mijia:013]:     Accuracy Decimals: 1
[17:50:43][C][xiaomi_mijia:013]:     Icon: 'mdi:thermometer'
[17:50:43][C][xiaomi_mijia:014]:   Humidity 'living humidity'
[17:50:43][C][xiaomi_mijia:014]:     Unit of Measurement: '%'
[17:50:43][C][xiaomi_mijia:014]:     Accuracy Decimals: 1
[17:50:43][C][xiaomi_mijia:014]:     Icon: 'mdi:water-percent'
[17:50:43][C][xiaomi_mijia:015]:   Battery Level 'living battery Level'
[17:50:43][C][xiaomi_mijia:015]:     Unit of Measurement: '%'
[17:50:43][C][xiaomi_mijia:015]:     Accuracy Decimals: 0
[17:50:43][C][xiaomi_mijia:015]:     Icon: 'mdi:battery'
[17:50:43][C][ota:029]: Over-The-Air Updates:
[17:50:43][C][ota:030]:   Address: temperatures.local:3232
[17:50:43][C][mqtt:051]: MQTT:
[17:50:43][C][mqtt:053]:   Server Address: m24.cloudmqtt.com:16193 (108.129.17.116)
[17:50:43][C][mqtt:054]:   Username: 'ernfgddp'
[17:50:43][C][mqtt:055]:   Client ID: 'temperatures-30aea41c742c'
[17:50:43][C][mqtt:057]:   Discovery prefix: 'homeassistant'
[17:50:43][C][mqtt:058]:   Discovery retain: YES
[17:50:43][C][mqtt:060]:   Topic Prefix: 'xiaomi/sensors'
[17:50:43][C][mqtt:062]:   Log Topic: 'xiaomi/sensors/debug'
[17:50:43][C][mqtt:065]:   Availability: 'xiaomi/sensors/status'
[17:50:43][C][mqtt.sensor:024]: MQTT Sensor 'living temperature':
[17:50:43][C][mqtt.sensor:028]:   State Topic: 'xiaomi/sensors/sensor/living_temperature/state'
[17:50:43][C][mqtt.sensor:024]: MQTT Sensor 'living humidity':
[17:50:43][C][mqtt.sensor:028]:   State Topic: 'xiaomi/sensors/sensor/living_humidity/state'
[17:50:43][C][mqtt.sensor:024]: MQTT Sensor 'living battery Level':
[17:50:43][C][mqtt.sensor:028]:   State Topic: 'xiaomi/sensors/sensor/living_battery_level/state'
[17:50:45][D][esp32_ble_tracker:470]: Found device 

```

**Additional information and things you've tried:**

b2un0, 2019-07-29:
you have to wait some minutes for log entries like this

```plain
[D][xiaomi_ble:118]: Got Xiaomi Mi Jia (58:2D:34:31:8A:60):
[D][xiaomi_ble:121]:   Temperature: 27.6°C
[D][xiaomi_ble:124]:   Humidity: 64.1%
[D][sensor:092]: 'studio_temperature': Sending state 27.60000 °C with 1 decimals of accuracy
[D][sensor:092]: 'studio_temperature': Sending state 27.60000 °C with 1 decimals of accuracy
[D][sensor:092]: 'studio_humidity': Sending state 64.10000 % with 1 decimals of accuracy
```

```plain
[D][esp32_ble_tracker:147]: Starting scan...
[D][xiaomi_ble:118]: Got Xiaomi Mi Jia (4C:65:A8:D8:AD:7D):
[D][xiaomi_ble:121]:   Temperature: 25.4°C
[D][xiaomi_ble:124]:   Humidity: 78.2%
[D][sensor:092]: 'bathroom_temperature': Sending state 25.40000 °C with 1 decimals of accuracy
[D][sensor:092]: 'bathroom_humidity': Sending state 78.20000 % with 1 decimals of accuracy
[D][sensor:092]: 'bathroom_humidity': Sending state 78.20000 % with 1 decimals of accuracy
[D][esp32_ble_tracker:470]: Found device 58:A2:07:C2:88:2C RSSI=-62y

```

Asinke, 2019-07-29:
I never ever see the ‘ sensor:092 ‘ lines as per below example, I guess > 30 minutes should do?
Will this event also trigger the sending of the data through MQTT?

b2un0, 2019-07-29:
exactly! if the ble tracker nothing found, nothing sends to MQTT..

check the mac address from your sensor,
remove the quotes in the yaml 
and put the sensor beside the esp for testing purpose.

Asinke, 2019-07-29:
Problem solved!
The issue was in white space in the friendly name for the sensor

Incorrect:
platform: xiaomi_mijia
    mac_address: 4C:65:A8:DD:16:7E
    temperature:
      name: living temperature

Correct:
platform: xiaomi_mijia
    mac_address: 4C:65:A8:DD:16:7E
    temperature:
      name: living_temperature

Now the sensor is correctly identified and the sensor state is sent to mqtt

Thanks for the quick and excellent support!

glmnet, 2019-07-29:
White space is certainly allowed on devices names

OttoWinter, 2019-10-13:
Yes, whitespace is allowed in sensor names. But I *think* you just listened on the wrong topic. Anyway, I will close this issue since OP has solved his issue, if there are others please open new issues (though I doubt this is really an issue, as I said wrong topic maybe).

closed 2019-10-13 by OttoWinter