esphome/issues #556: DS18B20 Problem - Sensor not found

rubenzori86, 2019-07-29:
I'v placed a DS18B20 sensor that works perfect on tasmota, When Im try to migrate to ESPHome i get **[07:20:22][W][dallas.sensor:075]:   Found no sensors!**

My yaml

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

1.13.6 Jul 29

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml

substitutions:
  devicename: lolin_acuario
  friendly_name: lolin_acuario
  lower_devicename: lolin_acuario
  upper_devicename: Lolin_Acuario
  
esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m
  #board_flash_mode: dout

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: on
  manual_ip:
    static_ip: 192.168.10.58
    gateway: 192.168.10.254
    subnet: 255.255.255.0
    dns1: 192.168.10.254
    dns2: 8.8.8.8

web_server:
  port: 80

# Enable logging
logger:
  baud_rate: 0 #need this to free up UART pins

# Enable Home Assistant API
api:

ota:

dallas:
  - pin: GPIO5
      
switch:
  - platform: gpio
    name: "$friendly_name Relay 01"
    pin: 
      number: GPIO12
      inverted: True

  - platform: gpio
    name: "$friendly_name Relay 02"
    pin: 
      number: GPIO13
      inverted: True    
  
  - platform: gpio
    name: "$friendly_name Relay 03"
    pin: 
      number: GPIO16
      inverted: True
  
  - platform: gpio
    name: "$friendly_name Relay 04"
    pin: 
      number: GPIO14
      inverted: True

sensor:
  - platform: wifi_signal
    name: "$friendly_name WiFi Signal"
    update_interval: 60s
  - platform: uptime
    name: "$friendly_name Uptime"
  - platform: dallas
    index: 0
    name: "Pool"  
    update_interval: 1s
#  - platform: dallas
#    name: ${upper_devicename} Temperature
#    id: ${lower_devicename}_temperature
#    address: 0x38041750ED77FF29  
  
text_sensor:
  - platform: version
    name: "$friendly_name ESPHome Version"

```

**Logs (if applicable):**

```

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.10.58 using esphome API
INFO Connecting to 192.168.10.58:6053 (192.168.10.58)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.10.58: timed out). Re-Trying in 1 seconds
INFO Connecting to 192.168.10.58:6053 (192.168.10.58)
INFO Successfully connected to 192.168.10.58
[07:20:22][I][app:096]: esphome version 1.13.6 compiled on Jul 29 2019, 07:12:50
[07:20:22][C][wifi:372]: WiFi:
[07:20:22][C][wifi:254]:   SSID: [redacted]
[07:20:22][C][wifi:255]:   IP Address: 192.168.10.58
[07:20:22][C][wifi:257]:   BSSID: [redacted]
[07:20:22][C][wifi:258]:   Hostname: 'lolin_acuario'
[07:20:22][C][wifi:262]:   Signal strength: -64 dB ▂▄▆█
[07:20:22][C][wifi:263]:   Channel: 11
[07:20:22][C][wifi:264]:   Subnet: 255.255.255.0
[07:20:22][C][wifi:265]:   Gateway: 192.168.10.254
[07:20:22][C][wifi:266]:   DNS1: 192.168.10.254
[07:20:22][C][wifi:267]:   DNS2: 8.8.8.8
[07:20:22][C][switch.gpio:042]: GPIO Switch 'lolin_acuario Relay 01'
[07:20:22][C][switch.gpio:043]:   Pin: GPIO12 (Mode: OUTPUT, INVERTED)
[07:20:22][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[07:20:22][C][switch.gpio:042]: GPIO Switch 'lolin_acuario Relay 02'
[07:20:22][C][switch.gpio:043]:   Pin: GPIO13 (Mode: OUTPUT, INVERTED)
[07:20:22][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[07:20:22][C][switch.gpio:042]: GPIO Switch 'lolin_acuario Relay 03'
[07:20:22][C][switch.gpio:043]:   Pin: GPIO16 (Mode: OUTPUT, INVERTED)
[07:20:22][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[07:20:22][C][switch.gpio:042]: GPIO Switch 'lolin_acuario Relay 04'
[07:20:22][C][switch.gpio:043]:   Pin: GPIO14 (Mode: OUTPUT, INVERTED)
[07:20:22][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[07:20:22][C][logger:137]: Logger:
[07:20:22][C][logger:138]:   Level: DEBUG
[07:20:22][C][logger:139]:   Log Baud Rate: 0
[07:20:22][C][logger:140]:   Hardware UART: UART0
[07:20:22][C][dallas.sensor:070]: DallasComponent:
[07:20:22][C][dallas.sensor:071]:   Pin: GPIO5 (Mode: INPUT)
[07:20:22][C][dallas.sensor:072]:   Update Interval: 1.0s
[07:20:22][W][dallas.sensor:075]:   Found no sensors!
[07:20:22][C][dallas.sensor:085]:   Device 'Pool'
[07:20:22][C][dallas.sensor:085]:     Unit of Measurement: '°C'
[07:20:22][C][dallas.sensor:085]:     Accuracy Decimals: 1
[07:20:22][C][dallas.sensor:085]:     Icon: 'mdi:thermometer'
[07:20:22][C][dallas.sensor:087]:     Index 0
[07:20:22][E][dallas.sensor:089]: Couldn't find sensor by index - not connected. Proceeding without it.
[07:20:22][C][version.text_sensor:015]: Version Text Sensor 'lolin_acuario ESPHome Version'
[07:20:22][C][version.text_sensor:015]:   Icon: 'mdi:new-box'
[07:20:22][C][web_server:125]: Web Server:
[07:20:22][C][web_server:126]:   Address: 192.168.10.58:80
[07:20:22][C][ota:029]: Over-The-Air Updates:
[07:20:22][C][ota:030]:   Address: 192.168.10.58:8266
[07:20:22][E][dallas.sensor:248]: Reading scratchpad from Dallas Sensor failed
[07:20:22][C][api:103]: API Server:
[07:20:22][C][api:104]:   Address: 192.168.10.58:6053
[07:20:22][C][wifi_signal.sensor:009]: WiFi Signal 'lolin_acuario WiFi Signal'
[07:20:22][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[07:20:22][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[07:20:22][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[07:20:23][E][dallas.sensor:248]: Reading scratchpad from Dallas Sensor failed
[07:20:24][E][dallas.sensor:248]: Reading scratchpad from Dallas Sensor failed
[07:20:25][E][dallas.sensor:248]: Reading scratchpad from Dallas Sensor failed

```

**Additional information and things you've tried:**

glmnet, 2019-07-29:
Do you have a pull-up resistor?
I don’t know if trasmota would configure the pin as pull-up and work. Esphome seems not to do it.

rubenzori86, 2019-07-29:
Yes, have the resistor fine.

brandond, 2019-07-29:
The dallas component is pretty stable and seems to work well when everything's wired up right. What are you pulling it up to, and what value resistor are you using? Are you using 2 or 3 wires?

rubenzori86, 2019-07-29:
I'm using this
https://s.click.aliexpress.com/e/btQhvY6y
That comes prepared with resistor.

brandond, 2019-07-29:
Have you tried another pin? For some reason the dallas component doesn't take a normal pin schema so you can't configure pull-up, but I know that GPIO 4 through 11 don't have pull-up enabled by default, while 0-3 and 12-15 do. I have a couple nodes working with GPIO13.

rubenzori86, 2019-07-30:
Yes, I tried all this pins with no luck :(

glmnet, 2019-07-30:
Have you tried a minimal esphome config?
Just Dallas and logger. No WiFi no web server no OTA. Etc.

rubenzori86, 2019-08-01:
Tried with minimal,

Here the yaml

```
substitutions:
  devicename: lolin_acuario
  friendly_name: lolin_acuario
  lower_devicename: lolin_acuario
  upper_devicename: Lolin_Acuario
  
esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m
  #board_flash_mode: dout

# wifi:
#   ssid: !secret wifi_ssid
#   password: !secret wifi_password
#   fast_connect: on
#   manual_ip:
#     static_ip: 192.168.10.58
#     gateway: 192.168.10.254
#     subnet: 255.255.255.0
#     dns1: 192.168.10.254
#     dns2: 8.8.8.8

#web_server:
#  port: 80

# Enable logging
logger:
  #baud_rate: 0 #need this to free up UART pins
  level: DEBUG

# Enable Home Assistant API
#api:

#ota:

dallas:
  - pin: GPIO13
    update_interval: 1s
      
# switch:
#   - platform: gpio
#     name: "$friendly_name Relay 01"
#     pin: 
#       number: GPIO12
#       inverted: True

#   - platform: gpio
#     name: "$friendly_name Relay 02"
#     pin: 
#       number: GPIO4
#       inverted: True    
  
#   - platform: gpio
#     name: "$friendly_name Relay 03"
#     pin: 
#       number: GPIO16
#       inverted: True
  
#   - platform: gpio
#     name: "$friendly_name Relay 04"
#     pin: 
#       number: GPIO14
#       inverted: True

# sensor:
#   - platform: wifi_signal
#     name: "$friendly_name WiFi Signal"
#     update_interval: 60s
#   - platform: uptime
#     name: "$friendly_name Uptime"
#  - platform: dallas
#    index: 0
#    name: "Pool"  

#  - platform: dallas
#    name: ${upper_devicename} Temperature
#    id: ${lower_devicename}_temperature
#    address: 0x38041750ED77FF29  
  
# text_sensor:
#   - platform: version
#     name: "$friendly_name ESPHome Version"
```

And here the logs

```
Done! Flashing is complete!

Showing logs:
[16:26:35]rl[I][logger:116]: Log initialized
[16:26:35][I][app:028]: Running through setup()...
[16:26:35][C][dallas.sensor:032]: Setting up DallasComponent...
[16:26:35][W][dallas.sensor:043]: Dallas device 0x7502FF98D3060EFF has invalid CRC.
[16:26:35][I][app:060]: setup() finished successfully!
[16:26:35][I][app:096]: esphome version 1.13.6 compiled on Aug  1 2019, 14:22:23
[16:26:35][C][logger:137]: Logger:
[16:26:35][C][logger:138]:   Level: DEBUG
[16:26:35][C][logger:139]:   Log Baud Rate: 115200
[16:26:35][C][logger:140]:   Hardware UART: UART0
[16:26:35][C][dallas.sensor:070]: DallasComponent:
[16:26:35][C][dallas.sensor:071]:   Pin: GPIO13 (Mode: INPUT)
[16:26:35][C][dallas.sensor:072]:   Update Interval: 1.0s
[16:26:35][W][dallas.sensor:075]:   Found no sensors!
```

mpaneth, 2019-08-29:
You don't seem to have started the I2C component which drives the comms.

I use
i2c:
  sda: GPIO4
  scl: GPIO5
  scan: True

OttoWinter, 2019-08-29:
@mpaneth ds18b20 sensor does not use i2c for communication.

rubenzori86, 2019-08-29:
@mpaneth can you showme complete yaml?

@OttoWinter my sensor still not working, what can be wrong??

OttoWinter, 2019-08-29:
@rubenzori86 Honestly, I don't know, as has been said before, dallas has been very solid for quite a while. And with the information here there's really no way to diagnose the issue.

rubenzori86, 2019-08-29:
Maybe I'm doing something wrong...

It's possible to send me a complete yaml for lolin v3 and I try to flash it?

mpaneth, 2019-08-29:
I have 3 DS18B20 connected in parallel which I have labled 0, 1 & 2 as I am trying to see how to callibrate them (so far no luck). I used https://esphome.io/components/sensor/dallas.html to generate the code.

I also have a SHT30 for comparative purposes. I was confused and this uses the I2C bus.

The yaml code is
# For the SHT30
i2c:
  sda: GPIO4
  scl: GPIO5
  scan: True

dallas:
  - pin: GPIO0
    update_interval: 10s

sensor:
  - platform: dallas
    index : 0
    name: "Work Room Temperature"
    id: d_temp
  - platform: dallas
    index : 1
    name: "Work Room Temp1"
    id: d_temp1
  - platform: dallas
    index : 2
    name: "Work Room Temp2"
    id: d_temp2

  - platform: sht3xd
    temperature:
      name: "Temperature"
      id: i_temp
    humidity:
      name: "Humidity"
      id: i_humid
    address: 0x45
    update_interval: 10s

Log snip
INFO Reading configuration...
INFO Detected timezone 'AEST' with UTC offset 10 and daylight savings time from 10/06/19 02:00:00 to 04/05/20 03:00:00
INFO Starting log output from temp.local using esphome API
INFO Connecting to temp.local:6053 (192.168.0.10)
INFO Successfully connected to temp.local
[08:40:25][I][app:096]: esphome version 1.13.6 compiled on Aug 18 2019, 21:56:01
[08:40:25][C][wifi:372]: WiFi:
[08:40:25][C][wifi:254]:   SSID: 'm-kbridge'[redacted]
[08:40:25][C][wifi:255]:   IP Address: 192.168.0.10
[08:40:25][C][wifi:257]:   BSSID: 10:0D:7F:D4:9C:50[redacted]
[08:40:25][C][wifi:258]:   Hostname: 'temp'
[08:40:25][C][wifi:262]:   Signal strength: -43 dB ▂▄▆█
[08:40:25][C][wifi:263]:   Channel: 6
[08:40:25][C][wifi:264]:   Subnet: 255.255.255.0
[08:40:25][C][wifi:265]:   Gateway: 192.168.0.1
[08:40:25][C][wifi:266]:   DNS1: 198.142.152.164
[08:40:25][C][wifi:267]:   DNS2: 198.142.152.165
[08:40:25][D][time:030]: Synchronized time: Fri Aug 30 08:40:25 2019
[08:40:25][C][i2c:028]: I2C Bus:
[08:40:25][C][i2c:029]:   SDA Pin: GPIO4
[08:40:25][C][i2c:030]:   SCL Pin: GPIO5
[08:40:25][C][i2c:031]:   Frequency: 50000 Hz
[08:40:25][I][i2c:033]: Scanning i2c bus for active devices...
[08:40:25][I][i2c:040]: Found i2c device at address 0x3C
[08:40:25][I][i2c:040]: Found i2c device at address 0x45
[08:40:26][C][logger:137]: Logger:
[08:40:26][C][logger:138]:   Level: DEBUG
[08:40:26][C][logger:139]:   Log Baud Rate: 115200
[08:40:26][C][logger:140]:   Hardware UART: UART0
[08:40:26][C][sht3xd:034]: SHT3xD:
[... 48 lines cut ...]
[08:40:39][D][dallas.sensor:144]: 'Work Room Temp1': Got Temperature=16.4°C
[08:40:39][D][sensor:092]: 'Work Room Temp1': Sending state 16.43750 °C with 1 decimals of accuracy
[08:40:39][D][dallas.sensor:144]: 'Work Room Temp2': Got Temperature=16.8°C
[08:40:39][D][sensor:092]: 'Work Room Temp2': Sending state 16.81250 °C with 1 decimals of accuracy
[08:40:44][D][sht3xd:059]: Got temperature=18.28°C humidity=44.20%
[08:40:44][D][sensor:092]: 'Temperature': Sending state 18.27878 °C with 1 decimals of accuracy
[08:40:44][D][sensor:092]: 'Humidity': Sending state 44.19928 % with 1 decimals of accuracy
[08:40:49][D][dallas.sensor:144]: 'Work Room Temperature': Got Temperature=17.3°C
[08:40:49][D][sensor:092]: 'Work Room Temperature': Sending state 17.31250 °C with 1 decimals of accuracy
[08:40:49][D][dallas.sensor:144]: 'Work Room Temp1': Got Temperature=16.4°C
[08:40:49][D][sensor:092]: 'Work Room Temp1': Sending state 16.43750 °C with 1 decimals of accuracy
[08:40:49][D][dallas.sensor:144]: 'Work Room Temp2': Got Temperature=16.8°C
[08:40:49][D][sensor:092]: 'Work Room Temp2': Sending state 16.75000 °C with 1 decimals of accuracy
[08:40:54][D][sht3xd:059]: Got temperature=18.28°C humidity=44.16%
[08:40:54][D][sensor:092]: 'Temperature': Sending state 18.27878 °C with 1 decimals of accuracy
[08:40:54][D][sensor:092]: 'Humidity': Sending state 44.16266 % with 1 decimals of accuracy
[08:40:59][D][dallas.sensor:144]: 'Work Room Temperature': Got Temperature=17.3°C
[08:40:59][D][sensor:092]: 'Work Room Temperature': Sending state 17.31250 °C with 1 decimals of accuracy
[08:40:59][D][dallas.sensor:144]: 'Work Room Temp1': Got Temperature=16.4°C
[08:40:59][D][sensor:092]: 'Work Room Temp1': Sending state 16.43750 °C with 1 decimals of accuracy
[08:40:59][D][dallas.sensor:144]: 'Work Room Temp2': Got Temperature=16.8°C
[08:40:59][D][sensor:092]: 'Work Room Temp2': Sending state 16.81250 °C with 1 decimals of accuracy
[08:41:04][D][sht3xd:059]: Got temperature=18.29°C humidity=44.18%
[08:41:04][D][sensor:092]: 'Temperature': Sending state 18.29213 °C with 1 decimals of accuracy
[08:41:04][D][sensor:092]: 'Humidity': Sending state 44.17640 % with 1 decimals of accuracy

Littleislandbrewing, 2019-08-31:
@rubenzori I have wired about 30 DS18b20 in my brewery and have noticed that they are not so reliable when you take them out of one situation and try to reuse them in another. The CRC error you see when it detects the serial no tells me it is now broken and faulty. I have had this w exact problem brand new sensors as well. I have not been able to resurrect any sensors w a CRC error.

Put the sensor back in the original working node and see if it works. (Unlikely)

Additionally I have had random success if you do not connect have a connection to wifi first (may be a bug but not proven) but not related to your problem - the CRC error is the problem.

rubenzori86, 2019-08-31:
@Littleislandbrewing but I'm can get it working with tasmota on same GPIO :S

Littleislandbrewing, 2019-08-31:
You can read about my trials w a ds18b20 in the brewery at https://habrewery.home.blog/

I ended up building a test bench for my sensors.

rubenzori86, 2019-08-31:
@Littleislandbrewing there is any other way to get address of sensor??

OttoWinter, 2019-08-31:
@rubenzori86 No, if you can't read the address you also won't be able to communicate with them.

MiloszP, 2019-09-05:
So you are using esp-01 when flashing the yaml. Are you sure thats your board model ? esp-01 has limited number of pin exposed. If you are for example using weemos board the gpio are numbered using D01 what is different pin than gpio0. Could you share link to your exact board model ?

rubenzori86, 2019-09-05:
Sure, I'm using lolin v3

https://www.theengineeringprojects.com/wp-content/uploads/2018/10/Introduction-to-NodeMCU-V3-2.png

MiloszP, 2019-09-05:
@rubenzori86 You should choose NodeMCU or Lolin in that case instead of esp-01
`esphome:
  name: $devicename
  platform: ESP32
  board: lolin_d32
`
for example

rubenzori86, 2019-09-05:
Could not find board 'lolin_d32'. Valid boards are d1, d1_mini, d1_mini_lite, d1_mini_pro, esp01, esp01_1m, esp07, esp12e, esp210, esp8285, esp_wroom_02, espduino, espectro, espino, espinotee, espresso_lite_v1, espresso_lite_v2, gen4iod, heltec_wifi_kit_8, huzzah, modwifi, nodemcu, nodemcuv2, oak, phoenix_v1, phoenix_v2, sparkfunBlynk, thing, thingdev, wifi_slot, wifiduino, wifinfo, wio_link, wio_node, xinabox_cw01.

MiloszP, 2019-09-05:
Strange i took it straight from my instance of esphome. Lolin seems to be very similar to nodemcu so you can try it

rubenzori86, 2019-09-05:
Tried but doesn't working too

MiloszP, 2019-09-06:
Have you tried using "d1" in config instead of "gpio15"?
Does your relay working on this chip ? 
Please attach your yaml after changes

Littleislandbrewing, 2019-09-06:
@rubenzori86 as @ottowinter said - unfortunately with a CRC error the sensor is dead and there is no rectification. I ended up buying heaps of them because they weren't so 'out of the box' reliable. Once you get it up and running then don't touch it. It's when you decide to repurpose them that they fail.

rubenzori86, 2019-09-06:
@MiloszP tried that you say without success.  Here you have yaml

```
substitutions:
  devicename: lolin_acuario
  friendly_name: lolin_acuario
  lower_devicename: lolin_acuario
  upper_devicename: Lolin_Acuario
  
esphome:
  name: $devicename
  platform: ESP8266
  board: nodemcu
  #board_flash_mode: dout

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: on
  manual_ip:
    static_ip: 192.168.10.58
    gateway: 192.168.10.254
    subnet: 255.255.255.0
    dns1: 192.168.10.254
    dns2: 8.8.8.8

web_server:
  port: 80

# Enable logging
logger:
  #baud_rate: 0 #need this to free up UART pins
  level: DEBUG

# Enable Home Assistant API
api:

ota:

dallas:
  pin: D1
  update_interval: 10s
      
switch:
  - platform: gpio
    name: "$friendly_name Relay 01"
    pin: 
      number: GPIO12
      inverted: True

  - platform: gpio
    name: "$friendly_name Relay 02"
    pin: 
      number: GPIO4
      inverted: True    
  
  - platform: gpio
    name: "$friendly_name Relay 03"
    pin: 
      number: GPIO16
      inverted: True
  
  - platform: gpio
    name: "$friendly_name Relay 04"
    pin: 
      number: GPIO14
      inverted: True

sensor:
  - platform: wifi_signal
    name: "$friendly_name WiFi Signal"
    update_interval: 60s
  - platform: uptime
    name: "$friendly_name Uptime"
    unit_of_measurement: dias
    update_interval: 300s
    filters:
      - multiply: 0.000011574      
 
#  - platform: dallas
#    name: ${upper_devicename} Temperature
#    id: ${lower_devicename}_temperature
#    address: 0x38041750ED77FF29  
  
text_sensor:
  - platform: version
    name: "$friendly_name ESPHome Version"

```

lwqcz, 2019-11-17:
@rubenzori86 have you tried another 1wire sensor? Maybe from another delivery batch? It may be bad or get dead. Most of these sensors are actually rejects from the quality control stage in the factory...

rubenzori86, 2019-11-17:
Ey, sorry . finally I fixed it changing wires and making better soldering.

Now is working fine

lwqcz, 2019-11-17:
I'm glad to hear that this issue is solved now :-) you can close this issue then ...

closed 2019-11-17 by rubenzori86