esphome/issues #562: Flicker Light Effect Moves Doesn't Respect Primary Color-Changes to White

jumblies, 2019-08-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker - RPi3B

**ESP (ESP32/ESP8266, Board/Sonoff):**

H801/ESP8266
https://esphome.io/cookbook/h801.html?highlight=h801
**Affected component:**

https://esphome.io/components/light/index.html#light-effects

**Description of problem:**
When on flicker light effect, ESP does not respect primary color and turns to white.  Any change to primary color is ignored on flicker.  The light does flicker, but always white.  Have tried setup for RGB and RGBW with same effect.  

Per the docs : 
"This effect “hovers” around the active color of the light and flickers each color channel a bit."  
but it doesn't use active color at all.  Logs reflect color changes but output color is the same.
This effect “hovers” around the active color of the light and flickers each color channel a bit.
**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: h801_2_light
  platform: ESP8266
  board: esp01_1m
wifi:
  ssid: 'banana'
  password: 'hammock'
  manual_ip:
    static_ip: 10.10.10.178
    gateway: 10.10.10.1
    subnet: 255.255.255.0
logger:
api:
ota:
sensor:
  - platform: wifi_signal
    name: "WiFi Signal h801_2_light"
    update_interval: 60s
#RGBWarm
output:
  - platform: esp8266_pwm
    pin: 12
    frequency: 1000 Hz
    id: pwm_b
  - platform: esp8266_pwm
    pin: 13
    frequency: 1000 Hz
    id: pwm_g
  - platform: esp8266_pwm
    pin: 15
    frequency: 1000 Hz
    id: pwm_r
  - platform: esp8266_pwm
    pin: 14
    frequency: 1000 Hz
    id: pwm_w
light:
  - platform: rgbw
    name: "H801 Light"
    red: pwm_r
    green: pwm_g
    blue: pwm_b
    white: pwm_w
    effects:
      - random:
      - flicker:
      - flicker:
          name: Flicker Effect With Custom Values
          alpha: 95%
          intensity: 5%
          
web_server:
  port: 80

```

**Logs (if applicable):**

```
INFO Reading configuration...
INFO Starting log output from 10.10.10.178 using esphome API
INFO Connecting to 10.10.10.178:6053 (10.10.10.178)
INFO Successfully connected to 10.10.10.178
[15:09:22][I][app:096]: esphome version 1.13.6 compiled on Aug  1 2019, 15:03:45
[15:09:22][C][wifi:372]: WiFi:
[15:09:22][C][wifi:254]:   SSID: 

[redacted]
[15:09:22][C][wifi:255]:   IP Address: 10.10.10.178
[15:09:22][C][wifi:257]:   BSSID: 

[redacted]
[15:09:22][C][wifi:258]:   Hostname: 'h801_2_light'
[15:09:22][C][wifi:262]:   Signal strength: -38 dB ▂▄▆█
[15:09:22][C][wifi:263]:   Channel: 11
[15:09:22][C][wifi:264]:   Subnet: 255.255.255.0
[15:09:22][C][wifi:265]:   Gateway: 10.10.10.1
[15:09:22][C][wifi:266]:   DNS1: 0.0.0.0
[15:09:22][C][wifi:267]:   DNS2: 0.0.0.0
[15:09:22][C][esp8266_pwm:022]: ESP8266 PWM:
[15:09:22][C][esp8266_pwm:023]:   Pin: GPIO12 (Mode: OUTPUT)
[15:09:22][C][esp8266_pwm:024]:   Frequency: 1000.0 Hz
[15:09:22][C][esp8266_pwm:022]: ESP8266 PWM:
[15:09:22][C][esp8266_pwm:023]:   Pin: GPIO13 (Mode: OUTPUT)
[15:09:22][C][esp8266_pwm:024]:   Frequency: 1000.0 Hz
[15:09:22][C][esp8266_pwm:022]: ESP8266 PWM:
[15:09:22][C][esp8266_pwm:023]:   Pin: GPIO15 (Mode: OUTPUT)
[15:09:22][C][esp8266_pwm:024]:   Frequency: 1000.0 Hz
[15:09:22][C][esp8266_pwm:022]: ESP8266 PWM:
[15:09:22][C][esp8266_pwm:023]:   Pin: GPIO14 (Mode: OUTPUT)
[15:09:22][C][esp8266_pwm:024]:   Frequency: 1000.0 Hz
[15:09:22][C][logger:137]: Logger:
[15:09:22][C][logger:138]:   Level: DEBUG
[15:09:22][C][logger:139]:   Log Baud Rate: 115200
[15:09:22][C][logger:140]:   Hardware UART: UART0
[15:09:22][C][light:174]: Light 'H801 Light'
[15:09:22][C][light:176]:   Default Transition Length: 1.0s
[15:09:22][C][light:177]:   Gamma Correct: 2.80
[15:09:22][C][web_server:125]: Web Server:
[15:09:22][C][web_server:126]:   Address: 10.10.10.178:80
[15:09:22][C][ota:029]: Over-The-Air Updates:
[15:09:22][C][ota:030]:   Address: 10.10.10.178:8266
[15:09:22][C][api:103]: API Server:
[15:09:22][C][api:104]:   Address: 10.10.10.178:6053
[15:09:22][C][wifi_signal.sensor:009]: WiFi Signal 'WiFi Signal h801_2_light'
[15:09:22][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[15:09:22][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[15:09:22][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[15:09:27][D][light:261]: 'H801 Light' Setting:
[15:09:27][D][light:283]:   Red=100%, Green=0%, Blue=0%
[15:09:27][D][light:300]:   Transition Length: 1.0s

```

**Additional information and things you've tried:**
Checked wiring.  Used 2 different H801 units with same result.  Standard colors work fine.

DarkFox, 2019-08-20:
This seems to apply to the strobe effect as well, which as far as I rememeber, used to flash in the selected colour.

It seems like a recent change causes it to simply ignore the primary colour.

jumblies, 2019-08-22:
@DarkFox Can you remember how long ago it worked so we can look at the diff?

Today I'm going to see if I can revert esphome to a different version and try it.  

To me the lambdas are confusing because I suck at C++ and I don't know what I can pass to lambda functions, but I extrapolated the code from esphome and came up with this crude flicker which does respect color.  I'm sure something similar can be done with strobe.

It unfortunately fixes brightness at 70%.  If I could figure out how to pass brightness to the lambda, then I could make the cubic random variation do +/- each iteration (as opposed to my simpleton reset to .7 each update)

```
globals:
   - id: global_brightness
     type: double
     restore_value: no
     initial_value: '0.7'
...
      - lambda:
          name: randomflicker
          update_interval: 200 ms
          lambda: |-
            auto call = id(h801_2_rgb).turn_on();
            float r = random_float() * 2.0f - 1.0f; 
            r =  r * r * r;
            call.set_transition_length(r);
            static double br=id(global_brightness);
            call.set_brightness(br + r/10);
            call.set_publish(false);
            call.set_save(false);
            call.perform();
```

DarkFox, 2019-08-22:
Unfortunately I haven't been keeping up with every update, but I'm fairly certain that it worked before 1.12.1, so it broke at some point after that.

jumblies, 2019-08-22:
@DarkFox 
I am unable to find any info on specifying version in order to revert and look at working code.  I'll see if I can find the diff on the light effects. 

Do you have any idea how to pass a light state to a global var? I don't know how to get access to the class members so bring their data into my lambda.

DarkFox, 2019-09-15:
@OttoWinter This applies to all lights, not just addressable. It's a pity, as I was using the strobe effect with different colors for notifications, but now I've got a choice of white or white.

jumblies, 2019-09-15:
Martin,  as a temporary workaround, you can use lambda functions which do
respect the primary color.  I've looked at the temolated src and esphome
but I'm not sophisticated enough to figure out where it is set to white

closed 2019-10-17 by OttoWinter