esphome/issues #563: Unhandled debug exception - Boot loop

xstrex, 2019-08-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip / MacOS 10.14.6

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-nodemcu-32s & ESP32-featheresp32

**Affected component:**

https://esphome.io/components/sensor/homeassistant.html

**Description of problem:**
When trying to add an additional Home-Assistant sensor, I receive a unhandled debug exception / boot loop. Remove the additional sensor, and the code works fine. Add ANY additional Home-Assistant sensor, and the exception returns (Doesn't matter which sensor you add) - Possibly running out of resources? 

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: wifi_signal
    id: int_wifi
    update_interval: 60s
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_temperature
    id: ds_temp
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_humidity
    id: ds_humid
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_wind_speed
    id: ds_speed
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_wind_gust
    id: ds_gust
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_precip_probability
    id: precip_prob
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_precip_intensity
    id: precip_inten
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_apparent_temperature_1h
    id: ds_temp_1h
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_apparent_temperature_2h
    id: ds_temp_2h
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_apparent_temperature_3h
    id: ds_temp_3h
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_apparent_temperature_4h
    id: ds_temp_4h
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_apparent_temperature_5h
    id: ds_temp_5h
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_apparent_temperature_6h
    id: ds_temp_6h
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_apparent_temperature_7h
    id: ds_temp_7h
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_humidity_1h
    id: ds_humid_1h
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_humidity_2h
    id: ds_humid_2h
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_humidity_3h
    id: ds_humid_3h
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_humidity_4h
    id: ds_humid_4h
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_humidity_5h
    id: ds_humid_5h
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_humidity_6h
    id: ds_humid_6h
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_humidity_7h
    id: ds_humid_7h
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_daytime_high_apparent_temperature_1d
    id: ds_temp_hi_1d
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_daytime_high_apparent_temperature_2d
    id: ds_temp_hi_2d
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_daytime_high_apparent_temperature_3d
    id: ds_temp_hi_3d
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_daytime_high_apparent_temperature_4d
    id: ds_temp_hi_4d
    internal: true
  # - platform: homeassistant
  #   entity_id: sensor.dark_sky_daytime_high_apparent_temperature_5d
  #   id: ds_temp_hi_5d
  #   internal: true
  # - platform: homeassistant
  #   entity_id: sensor.dark_sky_daytime_high_apparent_temperature_6d
  #   id: ds_temp_hi_6d
  #   internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_overnight_low_apparent_temperature_1d
    id: ds_temp_lo_1d
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_overnight_low_apparent_temperature_2d
    id: ds_temp_lo_2d
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_overnight_low_apparent_temperature_3d
    id: ds_temp_lo_3d
    internal: true
  - platform: homeassistant
    entity_id: sensor.dark_sky_overnight_low_apparent_temperature_4d
    id: ds_temp_lo_4d
    internal: true
  # - platform: homeassistant
  #   entity_id: sensor.dark_sky_overnight_low_apparent_temperature_5d
  #   id: ds_temp_lo_5d
  #   internal: true
  # - platform: homeassistant
  #   entity_id: sensor.dark_sky_overnight_low_apparent_temperature_6d
  #   id: ds_temp_lo_6d
  #   internal: true
```

If I add in a commented sensor, the exception happens- comment it out, exception goes away. 
**Logs (if applicable):**

```
[17:28:31]ets Jun  8 2016 00:22:57
[17:28:31]
[17:28:31]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:28:31]configsip: 0, SPIWP:0xee
[17:28:31]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:28:31]mode:DIO, clock div:2
[17:28:31]load:0x3fff0018,len:4
[17:28:31]load:0x3fff001c,len:928
[17:28:31]ho 0 tail 12 room 4
[17:28:31]load:0x40078000,len:9280
[17:28:31]load:0x40080400,len:5860
[17:28:31]entry 0x40080698
[17:28:31][I][logger:116]: Log initialized
[17:28:31][C][ota:364]: There have been 2 suspected unsuccessful boot attempts.
[17:28:31][I][app:028]: Running through setup()...
[17:28:31][C][spi:084]: Setting up SPI bus...
[17:28:31][D][binary_sensor:033]: 'EDSW Status': Sending state OFF
[17:28:32][C][wifi:029]: Setting up WiFi...
[17:28:32]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[17:28:32]Debug exception reason: Stack canary watchpoint triggered (loopTask)
[17:28:32]Core 1 register dump:
[17:28:32]PC      : 0x4008195c  PS      : 0x00060236  A0      : 0x3ffb0110  A1      : 0x3ffb0050  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d prod/edsw -t idedata
[17:28:37]A2      : 0x3ffb7dd4  A3      : 0x00000000  A4      : 0x3ffb7e1c  A5      : 0x00000000  
[17:28:37]A6      : 0x00000001  A7      : 0x00000004  A8      : 0x8008f8b8  A9      : 0x3ffb00f0  
[17:28:37]A10     : 0x00000001  A11     : 0x3ffc1564  A12     : 0x3ffc1568  A13     : 0x00000001  
[17:28:37]A14     : 0x00060220  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000001  
[17:28:37]EXCVADDR: 0x00000000  LBEG    : 0x40001609  LEND    : 0x4000160d  LCOUNT  : 0x00000000  
[17:28:37]
[17:28:37]Backtrace: 0x4008195c:0x3ffb0050 0x3ffb010d:0x3ffb0150 0x4010a8a2:0x3ffb0170 0x40082c2e:0x3ffb0190 0x400825f6:0x3ffb01b0 0x4010fc75:0x3ffb0220 0x4010e6f5:0x3ffb0240 0x4010eb43:0x3ffb0260 0x4010ecd1:0x3ffb02d0 0x4010d7ed:0x3ffb0350 0x4010d89c:0x3ffb03d0 0x4010c962:0x3ffb0430 0x4010ce99:0x3ffb0470 0x40177327:0x3ffb0490 0x40177512:0x3ffb04e0 0x40177565:0x3ffb0510 0x4013bdf4:0x3ffb0530 0x4013c002:0x3ffb0560 0x4010bafd:0x3ffb0590 0x400f1c99:0x3ffb05b0 0x400dcacd:0x3ffb06b0 0x400dbf11:0x3ffb06e0 0x400de76a:0x3ffb0710 0x400de4f1:0x3ffb0730 0x400eedc2:0x3ffb0780 0x400fa067:0x3ffb1fb0 0x4008eb8d:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
[17:28:37]
[17:28:37]Rebooting...
```
https://pastebin.com/c9nApXwG

**Additional information and things you've tried:**

I believe I've hit some kind of a sensor limit, which makes me thing it's running out of resources. If I remove 1 sensor, the issue disappears, if I add 1 sensor, it returns. 

I'm happy to attach project, in its entirety, if needed.

andrelung, 2019-09-05:
I am encountering the same issue. Have you had any success fixing it?
My setup:
- pip (github-lastest) / MacOS 10.14.6 / python via brew
- ESP-WROOM-32 / esp32dev
- wifi_signal is the only sensor we have in common

xstrex, 2019-09-05:
@andrelung I have not, unfortunately. But it is nice to know I'm not the only one! 

How many sensors does your config currently have? 
I'm running 27 that fall into the "sensor:" category. If I leave it at 27 the issue is not present, but if I add 1 more, making it 28, the issue returns.

andrelung, 2019-09-06:
Currently I use 12 sensors:
- 1 display
- 4 text_sensor
- 7 sensor
+ logger, wifi, ota, font, i2c, mqtt, time, esp32_ble_tracker
I think it is not a fixed number, it kind of relates to the ressources the sensors use. For example, if I remove the display, I can add many more sensors. Recently I also added time via sntp and I had to remove 2 sensors for that. I experimented a lot, but I can't figure out a scheme. Also I have an unverified feeling, that the order of the sensors might have an impact.

OttoWinter, 2019-10-14:
I agree it looks like a resource problem.

> Stack canary watchpoint triggered

That sounds to me like something is a) corrupting the stack or b) the heap is overfilling into the stack (if that can even happen with the malloc implementation here)

The fact that it appears to happen right after WiFi is initialized also points to that - the internal espressif APIs allocate a whole bunch of memory when wifi is initialized.

BLE certainly uses a lot of memory too, as does a display. All in all I think this is just a constraint of the system, and I don't expect a system with WiFi+BLE+Display to work well.

closed 2019-10-14 by OttoWinter