esphome/issues #564: ESP-7: OTA update fails with "ERROR[9]: new Flash config wsong real: 1048576"

mopagemo, 2019-08-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu Desktop 19.04, Kernel 5.0.0-21-generic

**ESP (ESP32/ESP8266, Board/Sonoff):** 

ESP8266 ESP-07

**Affected component:**

https://esphome.io/components/ota.html

**Description of problem:**
OTA update fails whereas update via USB (using FTDI module) works fine.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: main_house
  platform: esp8266
  board: esp07

wifi:
  ssid: "xxx"
  password: "xxx"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxx"

ota:
  password: "xxx"
```

**Logs (if applicable):**

OTA logs:
```
INFO Successfully compiled program.
INFO Resolving IP address of main_house.local
INFO  -> 192.168.xx.xx
INFO Uploading main_house/.pioenvs/main_house/firmware.bin (334192 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information.
```

While USB had logging running:
```
[15:10:50][D][ota:072]: Starting OTA Update from 192.168/88.100...
[15:10:50][D][ota:243]: OTA in progress: 0.2%
[15:10:52][D][ota:243]: OTA in progress: 27.1%
[15:10:53][D][ota:243]: OTA in progress: 39.3%
[15:10:54][D][ota:243]: OTA in progress: 53.7%
[15:10:55][D][ota:243]: OTA in progress: 67.5%
[15:10:56][D][ota:243]: OTA in progress: 82.2%
[15:10:57][D][ota:243]: OTA in progress: 96.5%
[15:10:58][D][ota:243]: OTA in progress: 99.4%
[15:10:58][W][ota:276]: Update end failed! Error: ERROR[9]: new Flash config wsong real: 1048576
```

**Additional information and things you've tried:**

manju-rn, 2019-08-12:
Caution:  I might be wrong here. So please wait for other before doing the below. 

Maybe an issue with flash size.  You might want to change the platformio options to provide the correct flash size. The ESP-07 has a 4 MB flash size

https://docs.platformio.org/en/latest/platforms/espressif8266.html#flash-size

From the Examples - you would need to get the correct entry for board...
esphome:
  # ...
  platformio_options:
    upload_speed: 115200
    board_build.f_flash: 80000000L

Drew765, 2019-08-18:
Having the same issue, did you get anywhere?

mopagemo, 2019-08-18:
Haven't had a chance yet to try changing the settings, but will do so over the next couple days and then report back.

mopagemo, 2019-08-19:
The settings already look fine to me in the platform file (https://github.com/platformio/platform-espressif8266/blob/master/boards/esp07.json). I explicitly set the values in the esphome configuration yaml, but got the same error.

brandond, 2019-08-19:
IIRC, doing the update OTA will not correct the flash layout, as it just rewrites the sketch (user code) portion of memory. If you need to change the size or layout you need to connect it via UART again and flash via the serial bootloader.

Drew765, 2019-08-22:
This may be entirely useless to you, but I've had issues with some esp's having incorrect flash sizes. I  just thought I'd try OTA this 'esp07' as 'esp01_1m' in the board description, and voila, it's working and I can OTA update it. I know changing the board description in the ESPhome OTA update is not supposed to do anything, but this has worked for me.

scsi050, 2019-08-31:
same OTA error for me here.  The board says it is an ESP-07, and pytool says it has a 1MB flash.

xx@xx:~ $ esptool.py --chip auto --port /dev/ttyAMA0 --baud 57600 write_flash --flash_size=detect -fm dout 0 <bin>
esptool.py v2.6
Serial port /dev/ttyAMA0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:ba:a1:4d
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Flash params set to 0x0320
Compressed 617880 bytes to 402086...
Wrote 617880 bytes (402086 compressed) at 0x00000000 in 72.9 seconds (effective 67.8 kbit/s)...
Hash of data verified.

closed 2019-09-01 by OttoWinter