esphome/issues #565: Migrated from virtualenv with python to docker image (esphome/esphome-armhf) but now things won't compile.

LandonStewart, 2019-08-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker running on a Raspberry Pi (using docker compose)

```
version: "3.4"
services:
  esphome:
    container_name: esphome
    environment:
    - TZ=${TZ}
    hostname: ${DOCKERHOSTNAME}
    image: esphome/esphome-armhf:latest
    network_mode: host
    ports:
    - 6052:6052
    restart: unless-stopped
    volumes:
    - /etc/localtime:/etc/localtime:ro
    - ${DOCKERCONFDIR}/esphome:/config:rw
```

**ESP (ESP32/ESP8266, Board/Sonoff):**

platform: espressif8266@1.8.0; framework: arduino; board: nodemcuv2

**Affected component:**

I'm sorry I don't know what component it is.  The error shows components/switch/switch.h if that helps.

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: multisensor01
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: 'RAINMAKER'
  password: 'XXXXXXXXX'
  manual_ip:
    static_ip: 192.168.1.101
    gateway: 192.168.1.1
    subnet: 255.255.255.0

# Enable logging
logger:

sensor:
  - platform: dht
    pin: D5
    model: dht22
    temperature:
      name: "SenseNode1 Temperature"
    humidity:
      name: "SenseNode1 Humidity"
    update_interval: 30s

binary_sensor:
  - platform: gpio
    pin: D6
    name: "SenseNode1 Motion"
    device_class: motion

ota:
  password: 'XYXYXYXYX'

api:
[... 143 lines cut ...]
^
compilation terminated.
In file included from src/esphome/components/api/util.h:5:0,
from src/esphome/components/api/api_message.h:4,
from src/esphome/components/api/api_message.cpp:1:
src/esphome/core/controller.h:23:46: fatal error: esphome/components/switch/switch.h: No such file or directory
#include "esphome/components/switch/switch.h"
^
In file included from src/esphome/components/api/api_server.h:4:0,
from src/esphome/components/api/api_server.cpp:3:
src/esphome/core/controller.h:23:46: fatal error: esphome/components/switch/switch.h: No such file or directory
#include "esphome/components/switch/switch.h"
^
compilation terminated.
compilation terminated.
*** [.pioenvs/multisensor01/src/esphome/components/api/command_messages.cpp.o] Error 1
*** [.pioenvs/multisensor01/src/esphome/components/api/api_message.cpp.o] Error 1
*** [.pioenvs/multisensor01/src/esphome/components/api/api_server.cpp.o] Error 1
*** [.pioenvs/multisensor01/src/esphome/components/api/basic_messages.cpp.o] Error 1
========================== [ERROR] Took 11.50 seconds ==========================

```

**Additional information and things you've tried:**
I tried using the docker image from https://hub.docker.com/r/esphome/esphome but of course that doesn't work on a Raspberry PI.

glmnet, 2019-08-05:
Try deleting the folders next to your yaml files (make a backup first)
Esphome should create folders again, you shouldn't lose any info.

LandonStewart, 2019-08-05:
> Try deleting the folders next to your yaml files (make a backup first)
> Esphome should create folders again, you shouldn't lose any info.

Ahhhh yes.  Thank you!  It's compiling fine now.

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/multisensor01
Processing multisensor01 (platform: espressif8266@1.8.0; framework: arduino; board: nodemcuv2)
--------------------------------------------------------------------------------
LibraryManager: Installing id=305 @ 1.2.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
LibraryManager: Installing v1.12.2
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=3837 @ 5.13.3
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphome-core> 1.12.2
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/multisensor01/src/main.cpp.o
Generating LD script .pioenvs/multisensor01/ld/eagle.app.v6.common.ld
Compiling .pioenvs/multisensor01/lib4d9/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling .pioenvs/multisensor01/lib4d9/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling .pioenvs/multisensor01/lib4d9/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling .pioenvs/multisensor01/lib4d9/ESP8266WiFi/ESP8266WiFiAP.cpp.o
Compiling .pioenvs/multisensor01/lib4d9/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
Compiling .pioenvs/multisensor01/lib4d9/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Compiling .pioenvs/multisensor01/lib4d9/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
Compiling .pioenvs/multisensor01/lib4d9/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Compiling .pioenvs/multisensor01/lib4d9/ESP8266WiFi/WiFiClient.cpp.o
Compiling .pioenvs/multisensor01/lib4d9/ESP8266WiFi/WiFiClientSecureAxTLS.cpp.o
Compiling .pioenvs/multisensor01/lib4d9/ESP8266WiFi/WiFiClientSecureBearSSL.cpp.o
Compiling .pioenvs/multisensor01/lib4d9/ESP8266WiFi/WiFiServer.cpp.o
Compiling .pioenvs/multisensor01/lib4d9/ESP8266WiFi/WiFiServerSecureAxTLS.cpp.o
Compiling .pioenvs/multisensor01/lib4d9/ESP8266WiFi/WiFiServerSecureBearSSL.cpp.o
Compiling .pioenvs/multisensor01/lib4d9/ESP8266WiFi/WiFiUdp.cpp.o
Compiling .pioenvs/multisensor01/lib0d3/ESP8266mDNS/ESP8266mDNS.cpp.o
Compiling .pioenvs/multisensor01/lib0e4/ESPAsyncTCP_ID305/AsyncPrinter.cpp.o
Compiling .pioenvs/multisensor01/lib0e4/ESPAsyncTCP_ID305/ESPAsyncTCP.cpp.o
Archiving .pioenvs/multisensor01/lib4d9/libESP8266WiFi.a
Compiling .pioenvs/multisensor01/lib0e4/ESPAsyncTCP_ID305/ESPAsyncTCPbuffer.cpp.o
Compiling .pioenvs/multisensor01/lib0e4/ESPAsyncTCP_ID305/SyncClient.cpp.o
Compiling .pioenvs/multisensor01/lib0e4/ESPAsyncTCP_ID305/tcp_axtls.c.o
Archiving .pioenvs/multisensor01/lib0d3/libESP8266mDNS.a
Compiling .pioenvs/multisensor01/lib6f5/Hash/Hash.cpp.o
Compiling .pioenvs/multisensor01/lib6f5/Hash/sha1/sha1.c.o
[... 271 lines cut ...]
[18:42:24][C][sensor.dht:036]:   Pin: GPIO14 (Mode: OUTPUT)
[18:42:24][C][sensor.dht:042]:   Model: DHT22 (or equivalent)
[18:42:24][C][sensor.dht:045]:   Update Interval: 30000 ms
[18:42:24][C][sensor.dht:047]:   Temperature 'SenseNode1 Temperature'
[18:42:24][C][sensor.dht:047]:     Unit of Measurement: '°C'
[18:42:24][C][sensor.dht:047]:     Accuracy Decimals: 1
[18:42:24][C][sensor.dht:048]:   Humidity 'SenseNode1 Humidity'
[18:42:24][C][sensor.dht:048]:     Unit of Measurement: '%'
[18:42:24][C][sensor.dht:048]:     Accuracy Decimals: 0
[18:42:24](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[18:42:24](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[18:42:24](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[18:42:24][C][ota:127]: Over-The-Air Updates:
[18:42:24][C][ota:128]:   Address: 192.168.1.101:8266
[18:42:24][C][ota:130]:   Using Password.
[18:42:29][D][api:546]: Client 'Home Assistant 0.96.5 (192.168.1.199)' connected successfully!
[18:42:41][D][sensor.dht:066]: Got Temperature=22.3°C Humidity=33.8%
[18:42:44][D][binary_sensor:037]: 'SenseNode1 Motion': Sending state ON
[18:42:46][D][binary_sensor:037]: 'SenseNode1 Motion': Sending state OFF
[18:43:11][D][sensor.dht:066]: Got Temperature=22.3°C Humidity=33.9%
[18:43:41][D][sensor.dht:066]: Got Temperature=22.3°C Humidity=33.8%
[18:44:11][D][sensor.dht:066]: Got Temperature=22.3°C Humidity=33.7%
[18:44:15][I][ota:152]: Boot seems successful, resetting boot loop counter.
[18:44:41][D][sensor.dht:066]: Got Temperature=22.3°C Humidity=33.7%
```

closed 2019-08-05 by LandonStewart