esphome/issues #566: Call on_turn_on with restore_mode

Anonym-tsk, 2019-08-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Docker

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP8266

**Affected component:** GPIO Output, GPIO Switch

**Description of problem:**
I want to make LED indicator of relay. Turn on LED if relay is turned on.
I configured `on_turn_on` action of `switch` to turn on led. But if i use `restore_mode: RESTORE_DEFAULT_ON`, `on_turn_on` is not called on start and led is off.

**Problem-relevant YAML-configuration entries:**
```yaml
output:
  - platform: gpio
    pin: GPIO00
    inverted: true
    id: led

switch:
  - platform: gpio
    pin: GPIO15
    id: relay
    name: 'Relay'
    restore_mode: RESTORE_DEFAULT_ON
    on_turn_on:
      - output.turn_on: led
    on_turn_off:
      - output.turn_off: led
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2019-08-05:
I wouldn't expect it to work that way.

You could have a [on_boot](https://esphome.io/components/esphome.html#on-boot) automation for that.

Anonym-tsk, 2019-08-05:
But i think, it's trivial and needed feature.
We have `interlock`, but don't have `lock`.

glmnet, 2019-08-05:
The default option is `RESTORE_DEFAULT_OFF`
do you think the `on_turn_off` should be called in the case it turned off when it "restores" ?

Anonym-tsk, 2019-08-05:
Well, maybe `restore_mode` for output component is better solution?

glmnet, 2019-08-05:
in that case you can use the already available restore mode for light (your LED is a light)

```yaml
light:
  - platform: binary
    id: led
    restore_mode: RESTORE_DEFAULT_ON
    output: output_gpio0 # you'll have to rename the id of the output to this
````

Anonym-tsk, 2019-08-05:
Yes, but i don't need led. In this case i have to create output and internal led )

glmnet, 2019-08-05:
you mean you don't need a led in Home Assistant?
if you don't provide `name` then ESPHome will not show this as a light to HA.
It is the same as setting `internal: true`

Anonym-tsk, 2019-08-12:
Ok. Light is workaround

closed 2019-08-12 by Anonym-tsk