esphome/issues #567: After adding the esp32_ble_tracker component the ESP32 board is  not reachable

blademckain, 2019-08-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32

**Affected component:**
esp32_ble_tracker

**Description of problem:**
After adding the esp32_ble_tracker component the ESP32 board is  not reachable

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test_ble
  platform: ESP32
  board: nodemcu-32s
  
wifi:
  ssid: "WifiDomotica"
  password: "lolololololoo"
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.14.38
    gateway: 192.168.14.254
    subnet: 255.255.255.0
    dns1: 8.8.8.8
    dns2: 8.8.4.4
  reboot_timeout: 1440min

# Enable logging
logger:
#  level: NONE
  level: VERBOSE
  
# Enable Home Assistant API
api:
 #reboot_timeout: 1440min
 
ota:

web_server:
  port: 80

# Example configuration entry
esp32_ble_tracker:
  scan_interval: 600s

binary_sensor:
  - platform: ble_presence
    mac_address: E0:EB:78:D3:9B:2D
    name: "ESP32 BLE Tracker Google Home Mini"

```

**Logs (if applicable):**

```
Using 'COM4' as serial port.
Writing at 0x000e4000... (100 %)Wrote 1505008 bytes (874227 compressed) at 0x00010000 in 77.6 seconds (effective 155.1 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[18:20:40]ets Jun  8 2016 00:22:57
[18:20:40]
[18:20:40]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:20:40]configsip: 0, SPIWP:0xee
[18:20:40]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:20:40]mode:DIO, clock div:2
[18:20:40]load:0x3fff0018,len:4
[18:20:40]load:0x3fff001c,len:952
[18:20:40]load:0x40078000,len:6084
[18:20:40]load:0x40080000,len:7944
[18:20:40]entry 0x40080310
[18:20:40][I][logger:116]: Log initialized
[18:20:40][E][Preferences.cpp:457] getBytes(): nvs_get_blob len fail: 0 NOT_FOUND
[18:20:40][V][preferences:204]: getBytes failed!
[18:20:40][C][ota:364]: There have been 0 suspected unsuccessful boot attempts.
[18:20:40][I][app:028]: Running through setup()...
[18:20:40][V][app:029]: Sorting components by setup priority...
[18:20:41][D][esp32_ble_tracker:147]: Starting scan...
[18:20:41][C][wifi:029]: Setting up WiFi...
[18:20:41][V][wifi_esp32:033]: Enabling STA.
[18:20:41]I (857) wifi: wifi driver task: 3ffe4fa0, prio:23, stack:3584, core=0
[18:20:41]I (1810) wifi: wifi firmware version: 6b44342
[18:20:41]I (1814) wifi: config NVS flash: enabled
[18:20:41]I (1818) wifi: config nano formating: disabled
[18:20:41]I (1845) wifi: Init dynamic tx buffer num: 32
[18:20:41]I (1845) wifi: Init data frame dynamic rx buffer num: 10
[18:20:41]I (1845) wifi: Init management frame dynamic rx buffer num: 10
[18:20:41]I (1849) wifi: Init static rx buffer size: 1600
[18:20:41]I (1853) wifi: Init static rx buffer num: 4
[18:20:41]I (1857) wifi: Init dynamic rx buffer num: 10
[18:20:41]I (1871) wifi: mode : softAP (80:7d:3a:ba:e7:a9)
[18:20:41]I (1872) wifi: Init max length of beacon: 752/752
[18:20:41]I (1872) wifi: Init max length of beacon: 752/752
[18:20:41][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[18:20:41][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[18:20:41][Vwifi: mode : sta (80:7d:3a:ba:e7:a8)
[18:20:41]I (1886) wifi: mode : sta (80:7d:3a:ba:e7:a8)
[18:20:41][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[18:20:41][V][wifi_esp32:297]: Event: WiFi STA start
[18:20:41][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[18:20:41][D][wifi:272]: Starting scan...
[18:20:46][V][wifi_esp32:297]: Event: WiFi STAracker Goo[D][binary_sensor:033]: 'ESP32 BLE Tracker Google Home Mini': Sending state ON
[18:21:11][E][wifi:280]: Scan timeout!
[18:21:11]
[18:21:11][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 1 - SCAN_DONE
[18:21:11][V][wifi_esp32:293]: Event: WiFi Scan Done status=0 number=3 scan_id=128
[18:21:16][D][wifi:272]: Starting scan...
[18:21:16][D][wifi:287]: Found networks:
[18:21:16][I][wifi:322]: - 'WifiDomotica' (E4:95:6E:41:AB:A4) ▂▄▆█
[18:21:16][D][wifi:323]:     Channel: 3
[18:21:16][D][wifi:324]:     RSSI: -88 dB
[18:21:16][D][wifi:326]: - 'ulisse ' (84:16:F9:24:80:3B) ▂▄▆█
[18:21:16][D][wifi:326]: - 'Alice-F4bh85td2Ghz' (0C:B6:D2:15:F4:15) ▂▄▆█
[18:21:16][I][wifi:164]: WiFi Connecting to 'WifiDomotica'...
[18:21:16][V][wifi:166]: Connection Params:
[18:21:16][V][wifi:167]:   SSID: 'WifiDomotica'
[18:21:16][V][wifi:170]:   BSSID: E4:95:6E:41:AB:A4
[18:21:16][V][wifi:174]:   Password: 'domotica'
[18:21:16][V][wifi:176]:   Channel: 3
[18:21:16][V][wifi:184]:   Manual IP: Static IP=192.168.14.38 Gateway=192.168.14.254 Subnet=255.255.255.0 DNS1=8.8.8.8 DNS2=8.8.4.4
[18:21:16][V][wifi:188]:   Hidden: NO
[18:21:16][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 1 - SCAN_DONE
[18:21:16][V][wifi_esp32:293]: Event:0>, ap:<255,255>, sta:<3,0>, prof:1
[18:21:16]I (37206) wifi: new:<3,0>, old:<1,0>, ap:<255,255>, sta:<3,0>, prof:1
[18:21:17]I (37940) wifi: state: init -> auth (b0)
[18:21:17]I (37946) wifi: state: auth -> assoc (0)
[18:21:17]I (37950) wifi: state: assoc -> run (10)
[18:21:17]I (37965) wifi: connected with WifiDomotica, channel 3, bssid = e4:95:6e:41:ab:a4
[18:21:17]I (37966) wifi: pm start, type: 0
[18:21:17]
[18:21:46][W][wifi:394]: Timeout while connecting to WiFi.
[18:21:46][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
[18:21:46][D][WiFiGeneric.cpp:385] _eventCallback(): STA IP: 192.168.14.38, MASK: 255.255.255.0, GW: 192.168.14.254
[18:21:46][V][wifi:164]: WiFi Connecting to 'WifiDomotica'...[I][wifi:164]: WiFi Connecting to 'WifiDomotica'...
[18:21:46][V][wifi:166]: Connection Params:
[18:21:46][V][wifi:167]:   SSID: 'WifiDomotica'
[18:21:46][V][wifi:170]:   BSSID: E4:95:6E:41:AB:A4
[18:21:46][V][wifi:174]:   Password: 'domotica'
[18:21:46][V][wifi:176]:   Channel: 3
[18:21:46][V][wifi:184]:   Manual IP: Static IP=192.168.14.38 Gateway=192.168.14.254 Subnet=255.255.255.0 DNS1=8.8.8.8 DNS2=8.8.4.4
[18:21:46][V][wifi:188]:   Hidden: NO
[18:21:46]
[18:21:46][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
[18:21:46][D][WiFiGeneric.cpp:385] _eventCallback(): STA IP: 192.168.14.38, MASK: 255.255.255.0, GW: 192.168.14.254
[18:21:46][18:21:46]

```

manju-rn, 2019-08-12:
So you get the error between 18:21:17 and 18:21:46?  Since between that time I believe the it is indeed connected. As you have webserver activated, Can you verify you are able to display the webpage between that time? just to see whether that BLE component shows up there?

Have you tried without the static ip? does it give similar error?

[**18:21:17**]I (37965) wifi: **connected with WifiDomotica**, channel 3, bssid = e4:95:6e:41:ab:a4
[18:21:17]I (37966) wifi: pm start, type: 0
[18:21:17]
[**18:21:46**][W][wifi:394]: **Timeout while connecting to WiFi.**

OttoWinter, 2019-10-13:
Closing due to inactivity.

closed 2019-10-13 by OttoWinter