esphome/issues #570: Feature request: Support Platformio 4.0 / Python 3

grzeg8102, 2019-08-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip
**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32
**Affected component:**
all

**Description of problem:**
Feature request:
I think would be good to start working on support for Platformio 4.0, which finally supports Python 3.5+. Python 2.7 will be deprecated in less 5 months now. Seems platformio Team understood the impact and added support for Python 3.5. So the only reason why esphome runs on top of python 2.7 would be gone forever (hopefully).

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2019-08-07:
Duplicate of https://github.com/esphome/feature-requests/issues/320

glmnet, 2019-08-07:
FYI: there is a feature-request specific issue tracker.

closed 2019-08-07 by glmnet

fake-name, 2019-08-29:
Python 3 is EoL in 5 months. 

This isn't a feature at this point, it's a bug.

glmnet, 2019-08-29:
Yes. If esphome isn’t based on python 3 at this moment is mostly because platform io wasn’t supporting it before. People really want to get it working ASAP but there is a large user base with very different implementation scenarios and rush will not help here.

fake-name, 2019-08-29:
I wound up cloning https://github.com/fabaff/esphome/tree/py3 and am having a poke.