esphome/issues #572: Scripts are non-blocking, an issue or by design?

overho79, 2019-08-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

hassio 0.96.5
esphome from command line, intel nuc7, windows 10 pro

**ESP (ESP32/ESP8266, Board/Sonoff):**

olimex Esp8266-EVB (config in the software as modwifi)

**Affected component:**

**Description of problem:**
In my project, I have a long sequence of actions in my code for running an irrigation schedule.  I created a script to help simplify repetitive parts of the long chain of actions.  The resulting code looks something like this

- actions to setup irrigation zone 1
- script run zone 1
- actions to setup irrigation zone 2
- script run zone 2
- actions to setup irrigation zone 3
- ...

The problem is that my expectation was that these actions would run as an ordered sequence.  Instead, the script seems to be running in a non-blocking fashion, such that the actions to setup irrigation zone 2 (as well as irrigation zone 2 running) are happening concurrently with the script to run zone 1.  This is not the behavior I was expecting, as there is no mention of this behavior in the documentation.  

I'm convinced that the script is running in parallel with later actions in the block, as I later added the following after running the script runZone1

wait_until:
  condition:
    not:
      script.is_running: runZone1

and after doing so, and it successfully prevents later actions from running until the script completes.  

This basically brings us to the question, is script.execute non-blocking by design (e.g. this is expected behavior), or is this an issue to be resolved?  If this is expected behavior, it would be helpful to add some language to the script documentation so that others are aware of it.

**Problem-relevant YAML-configuration entries:**
```yaml
switch:
  - platform: gpio
    name: "Enable Relays"
    id: enableWatering
    restore_mode: ALWAYS_OFF
    pin: GPIO5
  - platform: gpio
    name: "Irrigation Zone 1"
    id: zone1
    restore_mode: ALWAYS_OFF
    pin: GPIO1
    inverted: True
  - platform: gpio
    name: "Irrigation Zone 2"
    id: zone2
    restore_mode: ALWAYS_OFF
    pin: GPIO2
    inverted: True

binary_sensor:
  # allows hassio user to cancel the irrigation program while it is running
  - platform: homeassistant
    name: "Cancel Irrigation"
    entity_id: input_boolean.cancel_irrigation
    id: cancel_irrigation
  - platform: template
    id: completed
    lambda: |-
      //if zone has been running longer than duration, return completed
      return (millis() - id(zoneStart) > id(zoneDuration));

globals:
  - id: zoneStart
    type: long unsigned int
  - id: zoneDuration
    type: long unsigned int

# 1) try to write scripts for running the zones individually
# 2) try to write a generic one
script:
  - id: runZone1
    then:
      - delay: 5s
      - delay: !lambda |-
          if(id(cancel_irrigation).state) //if irrigation is already cancelled, return without starting the zone
            return 200;
          else //if irrigation should run, update global duration variables and turn on the zone
            id(zoneStart) = millis();
            id(zone1).turn_on();
            return 200;  //give completed binary_sensor time to update
      # try to do the equivalent of the while loop in esphome       
      - wait_until:
          condition: 
            or:
              - binary_sensor.is_on: completed
              - binary_sensor.is_on: cancel_irrigation
      - switch.turn_off: zone1
  - id: runZone2
    then:
      - delay: 5s
      - delay: !lambda |-
          if(id(cancel_irrigation).state) //if irrigation is already cancelled, return without starting the zone
            return 200;
          else //if irrigation should run, update global duration variables and turn on the zone
            id(zoneStart) = millis();
            id(zone2).turn_on();
            return 200;  //give completed binary_sensor time to update
      # try to do the equivalent of the while loop in esphome       
      - wait_until:
          condition: 
            or:
              - binary_sensor.is_on: completed
              - binary_sensor.is_on: cancel_irrigation
      - switch.turn_off: zone2

################################################ 
#  see then block of the start_watering service for applicable yaml
#################################################
api:
  services: 
#write service to run the watering
    - service: start_watering
      variables:
        duration_z1: int
        duration_z2: int
        duration_z3: int
        duration_z4: int
        duration_z5: int
        duration_z6: int
        duration_z7: int
        duration_z8: int
      then:
        # power up the relays
        - switch.turn_on: enableWatering
        # pass duration for zone 1 to global duration variable that is in scope for the script
        - delay: !lambda 'id(zoneDuration) = duration_z1*6000;return 20;'
        - script.execute: runZone1
        - wait_until:    # the script doesn't block/execute in order with the rest, so have to wait until it finishes
            condition:
              not: 
                script.is_running: runZone1 
        - delay: !lambda 'id(zoneDuration) = duration_z2*6000;return 20;'
        - script.execute: runZone2
        - wait_until:    # the script doesn't block/execute in order with the rest, so have to wait until it finishes
            condition:
              not: 
                script.is_running: runZone2

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Thanks for creating ESPHome!  It has been a great help in my smart home/IOT efforts!

overho79, 2019-08-08:
Looks like my logs didn't come through.  Here is a snippet of the log where script is running concurrently with later actions:

[13:39:55][D][api:573]: Client 'Home Assistant 0.96.5 (192.168.113.17)' connected successfully!
[13:39:55][D][time:030]: Synchronized time: Thu Aug  8 13:39:55 2019
[13:39:55][D][homeassistant.binary_sensor:023]: 'input_boolean.cancel_irrigation': Got state OFF
[13:39:55][D][binary_sensor:033]: 'Cancel Irrigation': Sending state OFF
[13:40:03][D][switch:021]: 'Enable Relays' Turning ON.
[13:40:03][D][switch:045]: 'Enable Relays': Sending state ON
[13:40:08][D][switch:021]: 'Irrigation Zone 1' Turning ON.
[13:40:08][D][switch:045]: 'Irrigation Zone 1': Sending state OFF
[13:40:09][D][switch:021]: 'Irrigation Zone 2' Turning ON.
[13:40:09][D][switch:045]: 'Irrigation Zone 2': Sending state OFF
[13:40:09][D][binary_sensor:033]: 'completed': Sending state OFF
[13:40:14][D][binary_sensor:033]: 'completed': Sending state ON
[13:40:14][D][switch:025]: 'Irrigation Zone 1' Turning OFF.
[13:40:14][D][switch:045]: 'Irrigation Zone 1': Sending state ON
[13:40:14][D][switch:025]: 'Irrigation Zone 2' Turning OFF.
[13:40:14][D][switch:045]: 'Irrigation Zone 2': Sending state ON

Here is a snippet of the log when the sequence was working:

[14:15:19][D][api:573]: Client 'Home Assistant 0.96.5 (192.168.113.17)' connected successfully!
[14:15:19][D][time:030]: Synchronized time: Thu Aug  8 14:15:19 2019
[14:15:19][D][homeassistant.binary_sensor:023]: 'input_boolean.cancel_irrigation': Got state OFF
[14:15:19][D][binary_sensor:033]: 'Cancel Irrigation': Sending state OFF
[14:15:27][D][switch:021]: 'Enable Relays' Turning ON.
[14:15:27][D][switch:045]: 'Enable Relays': Sending state ON
[14:15:32][D][switch:021]: 'Irrigation Zone 1' Turning ON.
[14:15:32][D][switch:045]: 'Irrigation Zone 1': Sending state OFF
[14:15:32][D][binary_sensor:033]: 'completed': Sending state OFF
[14:15:38][D][binary_sensor:033]: 'completed': Sending state ON
[14:15:38][D][switch:025]: 'Irrigation Zone 1' Turning OFF.
[14:15:38][D][switch:045]: 'Irrigation Zone 1': Sending state ON
[14:15:43][D][switch:021]: 'Irrigation Zone 2' Turning ON.
[14:15:43][D][switch:045]: 'Irrigation Zone 2': Sending state OFF
[14:15:43][D][binary_sensor:033]: 'completed': Sending state OFF
[14:15:49][D][binary_sensor:033]: 'completed': Sending state ON
[14:15:49][D][switch:025]: 'Irrigation Zone 2' Turning OFF.
[14:15:49][D][switch:045]: 'Irrigation Zone 2': Sending state ON

glmnet, 2019-08-08:
Wait and delay for sure don’t block other things than the script itself. 
All other stuff keeps going on while a wait or delay is running, I mean the esp keeps performing all the tasks. 
So basically the way this works is, “non blocking” other than the same script itself.

overho79, 2019-08-08:
If I understand your response correctly, it sounds like you are saying this
is the expected behavior of the ESPHome script design.  If that is true, it
probably makes sense to update the documentation to make this clear (and
also share the snippet about waiting for the script to complete for those
that don't want this behavior).  This definitely would have saved me some
time debugging this issue.

Just to make sure, was my description of the issue clear?

If I run a sequence of actions without using a script they run in order.

When I run the same sequence of actions (except now some of them are run
within a script),  some actions will end up running out of order (the
script will run in parallel with some of the actions after it).

Thanks!

vicfergar, 2019-09-28:
Non-blocking scripts are great. But I think something like "script.wait" action is very usefull when we need to block execution until script ends running. What do you think?

glmnet, 2019-09-28:
Delay and wait until blocks the script. 

However script execute or other actions does not wait the script even if those actions take some time to complete. 

Op expected script to be blocked at script.execute because in there was a blocking action. But that’s is not the case.

brandond, 2019-09-29:
You can't actually busy-wait on ESPs for very long. Background tasks like wifi and watchdog timer need to be given time to execute or the device will crash and reboot.

vicfergar, 2019-09-30:
With script.wait action I mean do something like this in a single line:

```
 wait_until:
   condition:
     not: 
       script.is_running: script_id
```

glmnet, 2019-09-30:
Ahh so this is a feature request, you'd better fill in a feature request then :)

vicfergar, 2019-09-30:
Done :)

https://github.com/esphome/feature-requests/issues/416

closed 2019-10-20 by OttoWinter