esphome/issues #574: ESP32 more than one touch sensor causes connection issues

tomlut, 2019-08-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**Affected component:**

https://esphome.io/components/binary_sensor/esp32_touch.html

**Description of problem:**
Adding more than one binary touch sensor causes connection time-out errors and connection problems to home assistant via the api.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: kitchen_leak
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "WAPDI"
  password: !secret wifi_pwd
  manual_ip:
    static_ip: 10.1.1.87
    gateway: 10.1.1.1
    subnet: 255.255.255.0

logger:
#  level: WARN

api:
  password: !secret api_password

ota:
  password: !secret esp_pwd

esp32_touch:
#  setup_mode: True
  iir_filter: 10ms

binary_sensor:
  - platform: esp32_touch
    name: "Sink Leak"
    id: sink_leak
    pin: GPIO27
    threshold: 100

  - platform: esp32_touch
    name: "Dishwasher Leak"
    id: dishwasher_leak
    pin: GPIO32
    threshold: 100

```
This does not cause connection problems:

```yaml
esphome:
  name: kitchen_leak
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "WAPDI"
  password: !secret wifi_pwd
  manual_ip:
    static_ip: 10.1.1.87
    gateway: 10.1.1.1
    subnet: 255.255.255.0

logger:
#  level: WARN

api:
  password: !secret api_password

ota:
  password: !secret esp_pwd

esp32_touch:
#  setup_mode: True
  iir_filter: 10ms

binary_sensor:
  - platform: esp32_touch
    name: "Sink Leak"
    id: sink_leak
    pin: GPIO27
    threshold: 100
```

**Logs (if applicable):**

```
```
[19:17:40][I][app:096]: esphome version 1.13.6 compiled on Aug 10 2019, 19:17:15
[19:17:40][C][wifi:372]: WiFi:
[19:17:40][C][wifi:254]:   SSID: 

[redacted]
[19:17:40][C][wifi:255]:   IP Address: 10.1.1.87
[19:17:40][C][wifi:257]:   BSSID: 

[redacted]
[19:17:40][C][wifi:258]:   Hostname: 'kitchen_leak'
[19:17:40][C][wifi:262]:   Signal strength: -49 dB ▂▄▆█
[19:17:40][C][wifi:263]:   Channel: 11
[19:17:40][C][wifi:264]:   Subnet: 255.255.255.0
[19:17:40][C][wifi:265]:   Gateway: 10.1.1.1
[19:17:40][C][wifi:266]:   DNS1: 0.0.0.0
[19:17:40][C][wifi:267]:   DNS2: 0.0.0.0
[19:17:40][C][logger:137]: Logger:
[19:17:40][C][logger:138]:   Level: DEBUG
[19:17:40][C][logger:139]:   Log Baud Rate: 115200
[19:17:40][C][logger:140]:   Hardware UART: UART0
[19:17:40][C][esp32_touch:029]: Config for ESP32 Touch Hub:
[19:17:40][C][esp32_touch:030]:   Meas cycle: 8.19ms
[19:17:40][C][esp32_touch:031]:   Sleep cycle: 27.31ms
[19:17:40][C][esp32_touch:051]:   Low Voltage Reference: 0.5V
[19:17:40][C][esp32_touch:071]:   High Voltage Reference: 2.7V
[19:17:40][C][esp32_touch:091]:   Voltage Attenuation: 0V
[19:17:40][C][esp32_touch:094]:     IIR Filter: 10ms
[19:17:40][C][esp32_touch:104]:   Touch Pad 'Sink Leak'
[19:17:40][C][esp32_touch:105]:     Pad: T7
[19:17:40][C][esp32_touch:106]:     Threshold: 100
[19:17:40][C][esp32_touch:104]:   Touch Pad 'Dishwasher Leak'
[19:17:40][C][esp32_touch:105]:     Pad: T9
[19:17:40][C][esp32_touch:106]:     Threshold: 100
[19:17:40][C][status:034]: Status Binary Sensor 'Kitchen Leak Status'
[19:17:40][C][status:034]:   Device Class: 'connectivity'
[19:17:40][C][ota:029]: Over-The-Air Updates:
[19:17:40][C][ota:030]:   Address: 10.1.1.87:3232
[19:17:40][C][ota:032]:   Using Password.
[19:17:40][C][api:103]: API Server:
[19:17:40][C][api:104]:   Address: 10.1.1.87:6053
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 10.1.1.87:6053 (10.1.1.87)
INFO Successfully connected to 10.1.1.87
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 10.1.1.87:6053 (10.1.1.87)
INFO Successfully connected to 10.1.1.87
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 10.1.1.87:6053 (10.1.1.87)
INFO Successfully connected to 10.1.1.87
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 10.1.1.87:6053 (10.1.1.87)
INFO Successfully connected to 10.1.1.87
[19:18:25][D][api:573]: Client 'Home Assistant 0.96.5 (10.1.1.100)' connected successfully!
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 10.1.1.87:6053 (10.1.1.87)
WARNING Couldn't connect to API (Error connecting to 10.1.1.87: [Errno 111] Connection refused). Trying to reconnect in 1 seconds
INFO Connecting to 10.1.1.87:6053 (10.1.1.87)

```

```

**Additional information and things you've tried:**

tomlut, 2019-08-10:
Powering from my Rigol bench-top power supply made no difference to the behaviour (I thought it might be a noisy 5V switchmode mains adaptor).

OttoWinter, 2019-08-11:
If it's a system crash (ESP prints stack trace) then please connect the device via USB and post the serial logs here - I can't see a specific reason why it would crash from looking at the code.

tomlut, 2019-08-11:
I've used an oldschool work around. I'm using using GPIO binary sensors with very weak external pullup resistors (3.9M Ohm) and 0.1uF noise bypass capacitors. Works perfectly for detecting town water.

As this device is in place with the extra components and it's my last ESP32 I cant do any more testing for now.

I have 5 more ESP32s on order from China and will do as you ask when they arrive (up to 4 weeks away).

OttoWinter, 2019-10-13:
Ok, will close this issue then - if other people experience this please create a new issue and reference this one.

closed 2019-10-13 by OttoWinter