esphome/issues #575: ESP8266 Remote Transmitter Switches not shown in HomeAssistant

manju-rn, 2019-08-10:
HASS.io
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESP8266  esp01_1m
**ESP (ESP32/ESP8266, Board/Sonoff):**

Remote Transmitter
**Affected component:**

**Description of problem:**
I have defined the Remote transmitter Swtiches as templates as per the docs. However i do not see these entities shown in the Lovelace UI.  I checked the intergration page as well, and there is no switch component listed.  I see in the logs that the these switches are defined as "[10:28:02][C][template.switch:058]: Template Switch 'My Fan'".   The same code in the ESP32 shows up correctly in the lovelace UI.  How do I resolve this?

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: manjuesp8266
  platform: ESP8266
  board: esp01_1m

wifi:
  networks:
  - ssid: !secret myssid
    password: !secret mypass
    hidden: True
  ap:
    ssid: "secret"
    password: "secret"

web_server:
  port: 8080

# Enable logging
logger:
  level: DEBUG
  
# Enable Home Assistant API
api:

ota:

remote_transmitter:
  pin: 2
  carrier_duty_percent: 100%

switch:
  - platform: template
    name: My Fan
    optimistic: True
    assumed_state: True
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '11011000001'
          protocol: 4
  - platform: template
    name: Test sony
    optimistic: True
    assumed_state: True
    turn_on_action:
      - remote_transmitter.transmit_sony:
          data: 0x123
          nbits: 12
          repeat: 
            times: 10
  - platform: template
    name: My RC Raq
    optimistic: True
    assumed_state: True  
    turn_on_action:
       - remote_transmitter.transmit_raw:
           code: [279, -1853, 278, -788, 277, -788, 252, -814, 278, -788, 277, -789, 277, -788, 278, -1827, 278, -788, 276, -789, 279, -1827, 278, -1826, 254, -1851, 278, -789, 277, -1827, 253]

```

**Logs (if applicable):**

```
[10:28:02][C][template.switch:058]: Template Switch 'My Fan'
[10:28:02][C][template.switch:058]:   Assumed State: YES
[10:28:02][C][template.switch:059]:   Restore State: NO
[10:28:02][C][template.switch:060]:   Optimistic: YES
[10:28:02][C][template.switch:058]: Template Switch 'Test sony'
[10:28:02][C][template.switch:058]:   Assumed State: YES
[10:28:02][C][template.switch:059]:   Restore State: NO
[10:28:02][C][template.switch:060]:   Optimistic: YES
[10:28:02][C][template.switch:058]: Template Switch 'My RC Raq'
[10:28:02][C][template.switch:058]:   Assumed State: YES
[10:28:02][C][template.switch:059]:   Restore State: NO
[10:28:02][C][template.switch:060]:   Optimistic: YES
[10:28:02][C][logger:137]: Logger:
[10:28:02][C][logger:138]:   Level: DEBUG
[10:28:02][C][logger:139]:   Log Baud Rate: 115200
[10:28:02][C][logger:140]:   Hardware UART: UART0
[10:28:02][C][remote_transmitter:018]: Remote Transmitter...
[10:28:02][C][remote_transmitter:019]:   Carrier Duty: 100%
[10:28:02][C][remote_transmitter:020]:   Pin: GPIO2 (Mode: OUTPUT)

```

**Additional information and things you've tried:**

OttoWinter, 2019-08-11:
Please make sure you added the device in your integrations screen - if possible also post a screenshot of the integration screen.

Other steps:

- Restart Home Assistant and the ESP device.

manju-rn, 2019-08-12:
There are no entities shown up in the integration screen for this ESP8266 in the integration page.  See the attached screenshots.  Also see the ESP's webserver where the same entities show up and they work perfectly, see the last screenshot  (there are few more entities I have added from the Yaml I gave earlier.
I have done multiple haasio restarts (including hard restart on the RPI where this is hosted)

Point to note:  The same entities show up flawlessly on the ESP32 (Nodemcu), this is only for ESP8266 that it is not showing up

[image]

[image]

[image]

rradar, 2019-08-16:
I had this in the past. Deleting the node in the integration panel of home assistant and adding it again can do the trick sometimes!

manju-rn, 2019-08-18:
I don;t find option to delete in the integration page.  You mean to delete the node in ESPHome?  well, need to backup some code as I have overwritten some parts in main.cpp.  Will try that later this week

glmnet, 2019-08-18:
It’s a bit offending that you overwrite code in main and then come here asking for help of the software misbehaving  and you did not tell that in the beginning. 
I’m wondering now if these same changes in main you did in the other code in ESP32 but honestly I don’t care.

manju-rn, 2019-08-19:
glmnet - not sure why would you find it offending. The code changes in the main.cpp are outside of the code that is generated by ESPHome. And they were done sometime back with no issues to the components I added or deleted afterwards, it is the issue only with the remote transmitters / receivers.  Also, there are no issues logged whatsoever.

brandond, 2019-08-19:
If you've mucked about with the autogenerated stuff that says 'do not edit', it's possible you broke or removed the bits that register the switch components which would definitely prevent them from showing up in HA. 

As @glmnet said, it'd be nice if you'd call that out in your issue description so we at least know what we're dealing with.

manju-rn, 2019-08-20:
@brandond As I stated in previous post, code changes were outside the code generated by ESPHOme - meaning in main.cpp I edited "outside" the "do not edit" and it's a stable code as it is running for quite sometime. Also any code within the "Do not Edit' will be overwritten by any new changes and upload from ESPHome. 
I do agree that I should have pointed out earlier, but as I said, these changes are there for some time and and feels integrated. 

Anyway, I tried creating a new ESP8266 node " with no  additional code" and only one of the 3 transmitters are shown.  will do some debugging and see the pattern before posting

manju-rn, 2019-08-20:
This can be closed.  After several restarts of the Hass.io, all remote transmitters are showing up.  So no Idea what was the issue, I checked all logs and there was no specific error related to this.

closed 2019-08-24 by glmnet