esphome/issues #576: GPS time seems wrong

deltazerorsan, 2019-08-10:
Hi Otto,

Thanks a lot for this incredible tool!

Using Hass.io 0.97 ESPHome 1.13.6 over a raspi3.

I'm trying the GPS module and I'm obtaining this:

22:44:39.215 -> [0;36m[D][time:030]: Synchronized time: Tue Jul 20 07:36:28 1965[0m
22:49:39.222 -> [0;36m[D][time:030]: Synchronized time: Tue Jul 20 07:36:28 1965[0m

Every five minutes. If I disconnect the GPS, there is no synchro, and a few seconds after I reconnect the GPS there comes the same synchro time... The GPS and the UART are both at 9600 bauds, and the GPS has fix.

Where can I look deeply?

Thanks!

rsan

rradar, 2019-08-16:
Could your share your gps board you use? Some of them have a rtc on it but sometimes without battery...

deltazerorsan, 2019-08-16:
> Could your share your gps board you use? Some of them have a rtc on it but sometimes without battery...

I'm using ublock 6 and ublock 8 boards and all of then works when connected to a PC or other micros including that one with a different soft. 

And please read the NMEA messages posted, they contains a correct date and time and as I described above are captured from the logs and decoded by hand...

In general in any board, if you have fix you have correct time, despite the battery status. And as you can see I have fix, logically checked and physically viewing the onboard led indicator.

Cheers!

closed 2019-08-27 by OttoWinter