esphome/issues #577: Nextion boot loop

gijsje, 2019-08-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**Affected component:**

**Description of problem:**
I am working on a nice Nextion setup with a lot of data from ha
It works great with the code i have below but i still have a lot of errors in the log but it works.
The errors look like this:
```
[10:39:34][W][nextion:083]: Sending command 'door_slide.bco=2016' failed because no ACK was received
[10:39:34][W][nextion:176]: Nextion reported variable name invalid!
```
That is not my main issue.
So the bellow code works but as soon as i add this line to the bottom of the below code the esp goes in to some sort of a boot loop as you see the screen flicker a lot and i do not understand why as the setup of the line is the same as the others but it does come back online.

The line i add: `it.send_command_printf("%s.bco=%.0f", "window_bed",id(color_window_bedroom).state);`

Could it be that the there are to many error logs going on the it does not work anymore?

**Problem-relevant YAML-configuration entries:**
```yaml
logger:
  baud_rate: 0

uart:
  rx_pin: D7
  tx_pin: D4
  baud_rate: 9600

time:
  - platform: sntp
    id: sntp_time
  - platform: homeassistant
    id: homeassistant_time

sensor:
#--------------------Temp Sensors-----------------------------------------------
  - platform: homeassistant
    id: temperature_tuin
    entity_id: sensor.tuin_temperatuur
    
  - platform: homeassistant
    id: temperature_slaapkamer
    entity_id: sensor.slaapkamer_temperatuur  

  - platform: homeassistant
    id: temperature_huiskamer
    entity_id: sensor.family_room_thermostat_temperature
    unit_of_measurement: "°C"
    
  - platform: homeassistant
    id: temperature_kantoor
    entity_id: sensor.kantoor_temperatuur
    
  - platform: homeassistant
    id: temperature_buiten
    entity_id: sensor.dark_sky_temperature  
    unit_of_measurement: "°C"
#--------------------Dim Level--------------------------------------------------    
  - platform: homeassistant
    id: dim_level
    entity_id: input_number.nextion_dim_level 
#--------------------Weather Icon-----------------------------------------------    
  - platform: homeassistant   # Forecast icon
    id: today_icon
    entity_id: sensor.today_icon    
#--------------------Color Motion-----------------------------------------------    
  - platform: homeassistant
    id: color_motion_office
    entity_id: input_number.nextion_color_motion_office
    
  - platform: homeassistant
    id: color_motion_bedroom
    entity_id: input_number.nextion_color_motion_bedroom
    
  - platform: homeassistant
    id: color_motion_gang
    entity_id: input_number.nextion_color_motion_gang
    
  - platform: homeassistant
    id: color_motion_garage
    entity_id: input_number.nextion_color_motion_garage
    
  - platform: homeassistant
    id: color_motion_garden
    entity_id: input_number.nextion_color_motion_garden
    
  - platform: homeassistant
    id: color_motion_hall
    entity_id: input_number.nextion_color_motion_hall
    
  - platform: homeassistant
    id: color_motion_kitchen
    entity_id: input_number.nextion_color_motion_kitchen
    
  - platform: homeassistant
    id: color_motion_living
    entity_id: input_number.nextion_color_motion_living
    
  - platform: homeassistant
    id: color_motion_toilet
    entity_id: input_number.nextion_color_motion_toilet
    
  - platform: homeassistant
    id: color_motion_washok
    entity_id: input_number.nextion_color_motion_washok

#--------------------Color Door-----------------------------------------------    
  - platform: homeassistant
    id: color_door_back
    entity_id: input_number.nextion_color_door_back  
    
  - platform: homeassistant
    id: color_door_front
    entity_id: input_number.nextion_color_door_front
    
  - platform: homeassistant
    id: color_door_garage
    entity_id: input_number.nextion_color_door_garage 
    
  - platform: homeassistant
    id: color_door_office
    entity_id: input_number.nextion_color_door_office 
    
  - platform: homeassistant
    id: color_door_slide
    entity_id: input_number.nextion_color_door_slide  
    
  - platform: homeassistant
    id: color_door_gard
    entity_id: input_number.nextion_color_door_garderobe
    
  - platform: homeassistant
    id: color_window_bedroom
    entity_id: input_number.nextion_color_window_bedroom
    
#--------------------Device Tracker---------------------------------------------    
text_sensor:
  - platform: homeassistant
    id: location_jeanette
    entity_id: device_tracker.google_maps_1
    
  - platform: homeassistant
    id: location_gijs
    entity_id: device_tracker.google_maps_1
#--------------------Home Status-----------------------------------------------    
  - platform: homeassistant
    id: home_status
    entity_id: input_select.thuis_status    

     

display:
  - platform: nextion
    id: page_main
    lambda: |-
          auto time_text = id(sntp_time).now().strftime("%X");
          it.set_component_text("text_time", time_text.c_str());
          auto date_text = id(sntp_time).now().strftime("%x");
          it.set_component_text("text_date", date_text.c_str());
          
          it.send_command_printf("dim=%.0f",id(dim_level).state);

          it.set_component_text_printf("text_weather","Buiten %.1f C",id(temperature_buiten).state);
          it.set_component_text_printf("lokatie_jeane","Jeanette: %s" ,id(location_jeanette).state.c_str());
          it.set_component_text_printf("locatie_gijs", "Gijsje: %s" ,id(location_gijs).state.c_str());
          it.set_component_text_printf("text_status", "Status %s" ,id(home_status).state.c_str());
  - platform: nextion
    id: page_weather
    lambda: |-
          it.set_component_text_printf("txt_temp_tuin","Tuin %.1f C",id(temperature_tuin).state);
          it.set_component_text_printf("txt_temp_slaap","Slaapkamer: %.1f C",id(temperature_slaapkamer).state);
          it.set_component_text_printf("txt_temp_huisk","Huiskamer: %.1f C",id(temperature_huiskamer).state);
          it.set_component_text_printf("txt_temp_kant","Kantoor: %.1f C",id(temperature_kantoor).state);

  - platform: nextion
    id: page_motion
    lambda: |-    
          it.send_command_printf("%s.bco=%.0f", "mot_kantoor",id(color_motion_office).state);  
          it.send_command_printf("%s.bco=%.0f", "mot_slaapk",id(color_motion_bedroom).state);  
          it.send_command_printf("%s.bco=%.0f", "mot_gang",id(color_motion_gang).state);  
          it.send_command_printf("%s.bco=%.0f", "mot_garage",id(color_motion_garage).state);  
          it.send_command_printf("%s.bco=%.0f", "mot_tuin",id(color_motion_garden).state);  
          it.send_command_printf("%s.bco=%.0f", "mot_hal",id(color_motion_hall).state);  
          it.send_command_printf("%s.bco=%.0f", "mot_keuken",id(color_motion_kitchen).state);  
          it.send_command_printf("%s.bco=%.0f", "mot_huiskamer",id(color_motion_living).state);  
          it.send_command_printf("%s.bco=%.0f", "mot_washok",id(color_motion_washok).state);  
          it.send_command_printf("%s.bco=%.0f", "mot_toilet",id(color_motion_toilet).state);  
          
  - platform: nextion
    id: page_door
    lambda: |-  
          it.send_command_printf("%s.bco=%.0f", "door_back",id(color_door_back).state);  
          it.send_command_printf("%s.bco=%.0f", "door_front",id(color_door_front).state); 
          it.send_command_printf("%s.bco=%.0f", "door_garage",id(color_door_garage).state); 
          it.send_command_printf("%s.bco=%.0f", "door_slide",id(color_door_slide).state);  
          it.send_command_printf("%s.bco=%.0f", "door_office",id(color_door_office).state); 

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2019-10-13:
> Could it be that the there are to many error logs going on the it does not work anymore?

no.

> The errors look like this but there are many more with similar text:

That looks like you're setting values on nextion components that do not exist in the layout editor (but unrelated to this issue).

> The line i add: `it.send_command_printf("%s.bco=%.0f", "window_bed",id(color_window_bedroom).state);`

That line looks fine to me, shouldn't be an issue as far as I see.

> So the bellow code works but as soon as i add this line to the bottom of the below code the esp goes in to some sort of a boot loop as you see the screen flicker a lot and i do not understand why as the setup of the line is the same as the others but it does not come back online.

If you connect it via USB, what does the serial monitor show? If you view the serial logs via esphome then the crash log will also be decoded which helps immensely.

Andoramb, 2019-10-23:
I was also struggling with this behavior. The 3.5inch screen flashes upon boot/reboot, and it is not possible to log via OTA. I can try to debug via usb. 
However, I've copied the exact same code from a Wemos D1 mini to Lolin32 and the flashing/rebooting was gone.
(I've changed the uart pins though, but I think that's not relevant)

closed 2020-12-05 by stale[bot]