esphome/issues #580: cant start esphome wizzard

nelbs, 2019-08-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
 Hassbian (installed with pip)

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**
I am using Hassbian and installed esphome succesfully with `pip install esphome`
I tried to start esphome with `esphome livingroom.yaml wizard` for the first time but I get the following error:

**Problem-relevant YAML-configuration entries:**

**Logs (if applicable):**

```
pi@hassbian:/ $ esphome livingroom.yaml wizard
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 6, in <module>
    from esphome.__main__ import main
  File "/usr/local/lib/python2.7/dist-packages/esphome/__main__.py", line 10, in <module>
    from esphome import const, writer, yaml_util
  File "/usr/local/lib/python2.7/dist-packages/esphome/writer.py", line 7, in <module>
    from esphome.config import iter_components
  File "/usr/local/lib/python2.7/dist-packages/esphome/config.py", line 15, in <module>
    from esphome import core, core_config, yaml_util
  File "/usr/local/lib/python2.7/dist-packages/esphome/core_config.py", line 6, in <module>
    import esphome.config_validation as cv
  File "/usr/local/lib/python2.7/dist-packages/esphome/config_validation.py", line 1242, in <module>
    Optional(CONF_COMMAND_TOPIC): All(requires_component('mqtt'), subscribe_topic),
  File "/usr/local/lib/python2.7/dist-packages/esphome/voluptuous_schema.py", line 198, in extend
    ret = super(_Schema, self).extend(schema, extra=extra)
  File "/usr/local/lib/python2.7/dist-packages/voluptuous/schema_builder.py", line 782, in extend
    return result_cls(result, required=result_required, extra=result_extra)
TypeError: __init__() got an unexpected keyword argument 'required'

```

**Additional information and things you've tried:**

maxmib, 2019-08-13:
pip2 install voluptuous==0.11.5
might solve this

nelbs, 2019-08-13:
Thanks that did the trick!

brandond, 2019-08-13:
It looks like this may have been broken by: https://github.com/alecthomas/voluptuous/pull/380

mahonni, 2019-08-13:
I was getting crazy because I'm not good at python, didn't know why it worked to me a few weeks ago and couldn't find any references about it. Thanks, @maxmib !!

closed 2019-08-19 by brandond

donpdonp, 2019-10-11:
first time esphome user.  really frustrating to hit this error right out of the gate. 0.11.5 fixed it.

```
$ sudo pip install esphome
Collecting esphome
  Downloading https://files.pythonhosted.org/packages/df/bd/a96abf8fb626b21793c42d5750a1503520d33a11fc4947408845fa7a4a24/esphome-1.13.6-py2.py3-none-any.whl (1.1MB)
Requirement already satisfied: PyYAML<6,>=5.1 in /usr/local/lib/python3.6/dist-packages (from esphome) (5.1.2)
Requirement already satisfied: paho-mqtt<2,>=1.4 in /usr/local/lib/python3.6/dist-packages (from esphome) (1.4.0)
Requirement already satisfied: voluptuous<0.12,>=0.11.5 in /usr/local/lib/python3.6/dist-packages (from esphome) (0.11.7)
...
$ esphome
Traceback (most recent call last):
...
  File "/usr/local/lib/python3.6/dist-packages/voluptuous/schema_builder.py", line 782, in extend
    return result_cls(result, required=result_required, extra=result_extra)
TypeError: __init__() got an unexpected keyword argument 'required'
```