esphome/issues #584: Compile error

neographikal, 2019-08-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
1.13.6 docker image

**ESP (ESP32/ESP8266, Board/Sonoff):**
esp32

**Affected component:**
asyncmqtt client

**Description of problem:**
Haven't been able to compile in a long time (weeks) because of

[image]

I understand development of 1.14.0 is underway but can we please get a minor patch on 1.13.6 so it can be compiled again? I need to roll out some nodes but I can't. :)

**Problem-relevant YAML-configuration entries:**
No difference

**Logs (if applicable):**
```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

brandond, 2019-08-14:
Have you tried the dev image rather than waiting for a new release?
`docker pull esphome/esphome:dev`

neographikal, 2019-08-14:
Just tried, confirmed I pulled the dev: 1.14.0-dev. Cleaned build, recompiled:

[image]

Still, with such a bug it would be nice to have a stable that will compile while the dev branch features are sorted out.

brandond, 2019-08-15:
I haven't seen anyone else report this recently on either stable or dev so it may be unique to your system. Have you tried removing your platformio files?

rdehuyss, 2019-08-15:
@brandond I have the same issue. The github issue was closed however (https://github.com/esphome/issues/issues/476). Also waiting on the fix...

neographikal, 2019-08-16:
> I haven't seen anyone else report this recently on either stable or dev so it may be unique to your system. Have you tried removing your platformio files?

Yes I cleaned the build before compiling. It got all the dependencies again, tried to build, same error unfortunately.

neographikal, 2019-08-20:
On the 1.14.0-dev I'm down to a warning, but it compiles.

`
Compiling .pioenvs/screensfront/lib1e7/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PingRespPacket.cpp.o
Compiling .pioenvs/screensfront/lib1e7/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PubAckPacket.cpp.o
.piolibdeps/AsyncMqttClient_ID346/src/AsyncMqttClient.cpp: In constructor 'AsyncMqttClient::AsyncMqttClient()':
.piolibdeps/AsyncMqttClient_ID346/src/AsyncMqttClient.cpp:40:61: warning: format '%x' expects argument of type 'unsigned int', but argument 3 has type 'uint64_t {aka long long unsigned int}' [-Wformat=]
sprintf(_generatedClientId, "esp32%06x", ESP.getEfuseMac());
^
.piolibdeps/AsyncMqttClient_ID346/src/AsyncMqttClient.cpp:40:61: warning: format '%x' expects argument of type 'unsigned int', but argument 3 has type 'uint64_t {aka long long unsigned int}' [-Wformat=]
Compiling .pioenvs/screensfront/lib1e7/AsyncMqttClient_ID346/AsyncMqttClient/Packets/PubCompPacket.cpp.o
`

1.13.6 is still giving a hard error and refuses to compile

`
Compiling .pioenvs/screensfront/libe24/AsyncTCP_ID1826/AsyncTCP.cpp.o
Compiling .pioenvs/screensfront/lib1e7/AsyncMqttClient_ID346/AsyncMqttClient.cpp.o
Compiling .pioenvs/screensfront/lib1e7/AsyncMqttClient_ID346/AsyncMqttClient/Packets/ConnAckPacket.cpp.o
.piolibdeps/AsyncTCP_ID1826/src/AsyncTCP.cpp: In function 'bool _start_async_task()':
.piolibdeps/AsyncTCP_ID1826/src/AsyncTCP.cpp:205:141: error: 'xTaskCreateUniversal' was not declared in this scope
xTaskCreateUniversal(_async_service_task, "async_tcp", 8192 * 2, NULL, 3, &_async_service_task_handle, CONFIG_ASYNC_TCP_RUNNING_CORE);
^
*** [.pioenvs/screensfront/libe24/AsyncTCP_ID1826/AsyncTCP.cpp.o] Error 1
.piolibdeps/AsyncMqttClient_ID346/src/AsyncMqttClient.cpp: In constructor 'AsyncMqttClient::AsyncMqttClient()':
.piolibdeps/AsyncMqttClient_ID346/src/AsyncMqttClient.cpp:40:61: warning: format '%x' expects argument of type 'unsigned int', but argument 3 has type 'uint64_t {aka long long unsigned int}' [-Wformat=]
sprintf(_generatedClientId, "esp32%06x", ESP.getEfuseMac());
^
.piolibdeps/AsyncMqttClient_ID346/src/AsyncMqttClient.cpp:40:61: warning: format '%x' expects argument of type 'unsigned int', but argument 3 has type 'uint64_t {aka long long unsigned int}' [-Wformat=]
========================== [ERROR] Took 24.67 seconds ==========================
`

I'd still prefer a minor patch version for 1.13.6 to run until 1.14.0 is stable and released.

tomlut, 2019-08-24:
Yep. ESP32 boards are now failing to compile for me too. Same error:

```
/data/lane_gate/.piolibdeps/AsyncTCP_ID1826/src/AsyncTCP.cpp:205:141: error: 'xTaskCreateUniversal' was not declared in this scope
```

OttoWinter, 2019-08-27:
Yes this is a duplicate of https://github.com/esphome/issues/issues/476 which has been fixed in 1.14 - I will try to create a small patch 1.13 release in the next couple of days that includes some problems once I get through the message backlog, and also cut off 1.14 at the same time.

closed 2019-08-27 by OttoWinter