esphome/issues #586: NodeMCU w/ PN532 - Wake Up Command Failed

anthonyjestin, 2019-08-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 NodeMCU

**Affected component:**

https://esphome.io/components/binary_sensor/pn532.html?highlight=pn532

**Description of problem:**

My error is the following: "Wake Up Command Failed" and can be seen in my log below, and additionally if I reboot the NodeMCU with a power cycle while the PN532 is connected, it won't show as online in the ESPHome dashboard, but will boot fine if board is not connected. I got exactly one successful tag read at one point yesterday, but have been unsuccessful in repeating that again.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: boardgames_nfc
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: [redacted]
  password: [redacted]

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:

ota:

spi:
  clk_pin: D5
  miso_pin: D6
  mosi_pin: D7

pn532:
  cs_pin: D8
  update_interval: 1s

```

**Logs (if applicable):**

```
INFO Reading configuration...
INFO Starting log output from boardgames_nfc.local using esphome API
INFO Connecting to boardgames_nfc.local:6053 (192.168.1.66)
INFO Successfully connected to boardgames_nfc.local
[20:30:30][I][app:096]: esphome version 1.13.6 compiled on Aug 14 2019, 20:13:05
[20:30:30][C][wifi:372]: WiFi:
[20:30:30][C][wifi:254]:   SSID: [redacted]
[20:30:30][C][wifi:255]:   IP Address: [redacted]
[20:30:30][C][wifi:257]:   BSSID: [redacted]
[20:30:30][C][wifi:258]:   Hostname: 'boardgames_nfc'
[20:30:30][C][wifi:262]:   Signal strength: -59 dB ▂▄▆█
[20:30:30][C][wifi:263]:   Channel: 11
[20:30:30][C][wifi:264]:   Subnet: 255.255.255.0
[20:30:30][C][wifi:265]:   Gateway: [redacted]
[20:30:30][C][wifi:266]:   DNS1: [redacted]
[20:30:30][C][wifi:267]:   DNS2: [redacted]
[20:30:30][C][spi:096]: SPI bus:
[20:30:30][C][spi:097]:   CLK Pin: GPIO14 (Mode: OUTPUT)
[20:30:30][C][spi:098]:   MISO Pin: GPIO12 (Mode: INPUT)
[20:30:30][C][spi:099]:   MOSI Pin: GPIO13 (Mode: OUTPUT)
[20:30:30][C][logger:137]: Logger:
[20:30:30][C][logger:138]:   Level: VERY_VERBOSE
[20:30:30][C][logger:139]:   Log Baud Rate: 115200
[20:30:30][C][logger:140]:   Hardware UART: UART0
[20:30:30][C][pn532:340]: PN532:
[20:30:30][E][pn532:345]: Wake Up command failed!
[20:30:30][C][pn532:352]:   CS Pin: GPIO15 (Mode: OUTPUT)
[20:30:30][C][pn532:353]:   Update Interval: 1.0s
[20:30:30][C][ota:029]: Over-The-Air Updates:
[20:30:30][C][ota:030]:   Address: boardgames_nfc.local:8266
[20:30:30][C][api:103]: API Server:
[20:30:30][C][api:104]:   Address: boardgames_nfc.local:6053

```

**Additional information and things you've tried:**

I have reviewed issued #491  and #492 , however my issues persist.
I have the following connections made between the two boards (NodeMCU -> PN532):
D5 (SCLK) -> SCK
D6 (MISO) -> MISO
D7 (MOSI) -> MOSI
D8 (CS) -> SS
3.3V -> VCC
GND -> GND

Haringstad, 2019-08-19:
I face the same issue, and I tried several nodeMCU's, ESP32, and 3 different RFID scanners.

anthonyjestin, 2020-01-11:
Marking issue as closed.  Not sure how I didn't clue into this earlier, but the PN532 board needed 5V, not 3.3V.  As soon as I hooked up with proper voltage my issue was resolved.

closed 2020-01-11 by anthonyjestin