esphome/issues #587: Bang Bang Climate keeps on when temperature reading is NAN

john-arvid, 2019-08-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

https://esphome.io/components/climate/bang_bang.html

**Description of problem:**
If the temperature sensor malfunctions when the switch is turned on it never turns off. (Never say never, but at least not for the 15 minutes I tested)

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: esp01_mill04

esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret ssid
  password: !secret wpa

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password
  
web_server:
  port: 80

switch:
  - platform: gpio
    restore_mode: RESTORE_DEFAULT_OFF
    pin: GPIO0
    id: relay1
    name: $devicename " Relay"
    
sensor:
  - platform: dht
    pin: GPIO3
    model: AM2302
    temperature:
      id: temp1
      name: $devicename " Temperature"
    humidity:
      name: $devicename " Humidity"
    update_interval: 10s

climate:
  - platform: bang_bang
    name: test
    sensor: temp1
    default_target_temperature_low: 22 °C
    default_target_temperature_high: 26 °C

    heat_action:
      - switch.turn_on: relay1
    idle_action:
      - switch.turn_off: relay1

```

**Logs (if applicable):**

```
[18:50:37][W][dht:120]: Requesting data from DHT failed!
[18:50:37][W][dht:058]: Invalid readings! Please check your wiring (pull-up resistor, pin number).

```

glmnet, 2019-08-23:
Yeah if the sensor component is marked as failed the climate component should be failed too thus call idle action for safety

OttoWinter, 2019-08-31:
Duplicate of https://github.com/esphome/issues/issues/590

OttoWinter, 2019-08-31:
@glmnet FYI, the bang_bang controller already automatically switches to IDLE mode when it receives NAN values, but DHT apparently doesn't do that, see #590

closed 2019-08-31 by OttoWinter