esphome/issues #588: Bang Bang Climate target_temperature resets after reboot

john-arvid, 2019-08-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

https://esphome.io/components/climate/bang_bang.html

**Description of problem:**
Not sure if feature or a bug but after a reboot the target_temperatures are not stored when they are changed in frontend.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: esp01_mill04

esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret ssid
  password: !secret wpa

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password
  
web_server:
  port: 80

switch:
  - platform: gpio
    restore_mode: RESTORE_DEFAULT_OFF
    pin: GPIO0
    id: relay1
    name: $devicename " Relay"
    
sensor:
  - platform: dht
    pin: GPIO3
    model: AM2302
    temperature:
      id: temp1
      name: $devicename " Temperature"
    humidity:
      name: $devicename " Humidity"
    update_interval: 10s

climate:
  - platform: bang_bang
    name: test
    sensor: temp1
    default_target_temperature_low: 22 °C
    default_target_temperature_high: 26 °C

    heat_action:
      - switch.turn_on: relay1
    idle_action:
      - switch.turn_off: relay1

```

glmnet, 2019-08-23:
This is by design, the ESP8266 retain memory is not used by default by ESPHome due to wearing issues, and this info is stored in the ESPHome normally, not in Home Assistant.
If you want to change this behavior try setting the `esp8266_restore_from_flash` option. If it doesn't work post again and I'll reopen this issue.

closed 2019-08-23 by glmnet

john-arvid, 2019-08-23:
Is it possible to change this design? So that the default_target_temperature is updated when changed from HA.
Cause as you say, retain memory causes wearing issues, so I would say this is not a solution.

glmnet, 2019-08-23:
There is no easy solution. 
The same thing happens eg with lights. You turn on a light then esp is power cycled and the light is by default off.

glmnet, 2019-08-23:
Of course default_target_temperature is hard coded in the yaml thus hard coded in esphome generated code

glmnet, 2019-08-23:
There is an idea of using home assistant esphome client (home assistant itself) as a state server but it is just an idea so far

john-arvid, 2019-08-23:
Ok, understand.
Thank you for your replies.

My solution for now is to either use NodeRed to keep the temperature settings updated, or not use Bang Bang at all and just use a generic thermostat in HA.

OttoWinter, 2019-08-27:
Why not use `esp8266_restore_from_flash`? Please post how often you would change the target temperature values, I doubt it will be anywhere near enough to get close to memory wear.

john-arvid, 2019-08-27:
Ok, if I take a normal schedule it would have one setting at night, a different one in the morning, a different one during the workours, then a different during the evening, then back to night settings. Then it will add extra if I manually choose to change the temperature during the day.
So a minimum of 4 changes pr day. (What is the limit of memory wear for the esp8266?)

But on the other hand, since I have node red to control the automation I could also have node red to check if the temperature is correct. But this again breaks when I manually change the settings.

OttoWinter, 2019-08-27:
@john-arvid Please read the [docs](https://esphome.io/components/esphome.html#esp8266-restore-from-flash), at 4 updates per day it's much more likely the cheap chip breaks from other sources.

john-arvid, 2019-08-28:
I have read the docs and it is stated there that it should not be used for switches. It also says that you can disable restore mode, is this config correct? (So it will only keep the `default_target_temperature` in the flash)

```
substitutions:
  devicename: esp01_mill04

esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m
  esp8266_restore_from_flash: true

wifi:
  ssid: !secret ssid
  password: !secret wpa

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password
  
web_server:
  port: 80

switch:
  - platform: gpio
    restore_mode: RESTORE_DEFAULT_OFF
    pin: GPIO0
    id: relay1
    name: $devicename " Relay"
    
sensor:
  - platform: dht
    pin: GPIO3
    model: AM2302
    temperature:
      id: temp1
      name: $devicename " Temperature"
    humidity:
      name: $devicename " Humidity"
    update_interval: 10s

climate:
  - platform: bang_bang
    name: test
    sensor: temp1
    default_target_temperature_low: 22 °C
    default_target_temperature_high: 26 °C

    heat_action:
      - switch.turn_on: relay1
    idle_action:
      - switch.turn_off: relay1
```

john-arvid, 2019-08-28:
Thank you for the feedback.
I can confirm that it works as expected. 
The only thing I haven't tested is how many writes to flash it does. But I am not sure if that is possible to test.

OttoWinter, 2019-08-29:
@john-arvid It saves each time you change a configurable value (target temperature). If you crank the log verbosity, you will see `SAVE ...` messages.