esphome/issues #589: (SlidingWindowMovingAverage)Filter get same input value twice

hSoubry, 2019-08-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io on a Synology nas

**ESP (ESP32/ESP8266, Board/Sonoff):**
Wemos D1 mini

**Affected component:**
Ultrasonic Distance Sensor
SlidingWindowsMovingAverageFilter 

**Description of problem:**
The SlidingWindowsMovingAverageFilter is updated with te same sensor reading twice. I have the Ultrasonic Distance Sensor setup for a reading every 5seconds. And the SlidingWindowsMovingAverageFilter setup to a windows_size of 12 samples. This should give an output every 60seconds. But in the logging I see an output from the filter every 30 seconds.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: my_first_esp_node
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "----------"
  password: "----------"

# Enable logging
logger:
  level: VERY_VERBOSE
  
# Example configuration entry
sensor:
  - platform: ultrasonic
    trigger_pin: D5
    echo_pin: D0
    name: "Regenwater"
    unit_of_measurement: "m³"
    update_interval: 5s
    timeout: 2.5m
    filters:
      - sliding_window_moving_average:
          window_size: 12
          send_every: 12
          send_first_at: 12

      - lambda: return (2.22 - x) * 7.5;
  
    
# Enable Home Assistant API
api:

ota:```

**Logs (if applicable):**
<!--
Please copy the debug log here. If possible, also connect to the ESP over USB and copy those logs into the backticks.
-->
```
[09:17:06][I][app:096]: esphome version 1.13.6 compiled on Aug 15 2019, 09:14:03
[09:17:06][C][wifi:372]: WiFi:
[09:17:06][C][wifi:254]:   SSID: [redacted]
[09:17:06][C][wifi:255]:   IP Address: 192.168.1.62
[09:17:06][C][wifi:257]:   BSSID: [redacted]
[09:17:06][C][wifi:258]:   Hostname: 'my_first_esp_node'
[09:17:06][C][wifi:262]:   Signal strength: -89 dB ▂▄▆█
[09:17:06][C][wifi:263]:   Channel: 6
[09:17:06][C][wifi:264]:   Subnet: 255.255.255.0
[09:17:06][C][wifi:265]:   Gateway: 192.168.1.1
[09:17:06][C][wifi:266]:   DNS1: 192.168.1.1
[09:17:06][C][wifi:267]:   DNS2: 0.0.0.0
[09:17:06][C][logger:137]: Logger:
[09:17:06][C][logger:138]:   Level: VERY_VERBOSE
[... 169 lines cut ...]
[09:18:14][VV][sensor.filter:013]: Filter(0x3fff130c)::input(0.771407)
[09:18:14][VV][sensor.filter:062]: SlidingWindowMovingAverageFilter(0x3fff130c)::new_value(0.771407) -> 0.771893
[09:18:16][VV][api:598]: on_ping_request_
[09:18:19][VV][component:128]: Running interval 'update':0 with interval=5000 last_execution=91338 (now=96347)
[09:18:19][V][ultrasonic.sensor:022]: Echo took 4500µs
[09:18:19][V][sensor:013]: 'Regenwater': Received new state 0.771750
[09:18:19][VV][sensor.filter:013]: Filter(0x3fff130c)::input(0.771750)
[09:18:19][VV][sensor.filter:062]: SlidingWindowMovingAverageFilter(0x3fff130c)::new_value(0.771750) -> 0.771821
[09:18:19][D][ultrasonic.sensor:030]: 'Regenwater' - Got distance: 0.77 m
[09:18:19][V][sensor:013]: 'Regenwater': Received new state 0.771750
[09:18:19][VV][sensor.filter:013]: Filter(0x3fff130c)::input(0.771750)
[09:18:19][VV][sensor.filter:062]: SlidingWindowMovingAverageFilter(0x3fff130c)::new_value(0.771750) -> 0.771750
[09:18:19][VV][sensor.filter:066]: SlidingWindowMovingAverageFilter(0x3fff130c)::new_value(0.771750) SENDING
[09:18:19][VV][sensor.filter:023]: Filter(0x3fff130c)::output(0.771750) -> 0x3fff1574
[09:18:19][VV][sensor.filter:013]: Filter(0x3fff1574)::input(0.771750)
[09:18:19][VV][sensor.filter:107]: LambdaFilter(0x3fff1574)::new_value(0.771750) -> 10.861876
[09:18:19][VV][sensor.filter:020]: Filter(0x3fff1574)::output(10.861876) -> SENSOR
[09:18:19][D][sensor:092]: 'Regenwater': Sending state 10.86188 m³ with 2 decimals of accuracy
[09:18:21][VV][api:598]: on_ping_request_

```

**Additional information and things you've tried:**

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

brandond, 2019-08-15:
This is a bug with the Ultrasonic sensor. `publish_state` is called twice:
https://github.com/esphome/esphome/blob/dev/esphome/components/ultrasonic/ultrasonic_sensor.cpp#L29

hSoubry, 2019-08-16:
Thank you for finding the problem. Will this be fixed in an upcomming release?
In the meantime, I doubled the filters window_size to work around this.

closed 2019-08-19 by brandond