esphome/issues #590: id(some_sensor).state is not updated when reading failed

john-arvid, 2019-08-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

https://esphome.io/components/sensor/dht.html

**Description of problem:**
I am trying to check when a temperature read fails, for example if the sensor or wire breaks.
So when disconnecting a DHT22 sensor the id(some_sensor).state is not updated, but keeps the last value.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: esp01_mill04

esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret ssid
  password: !secret wpa

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password
  
web_server:
  port: 80

switch:
  - platform: gpio
    restore_mode: RESTORE_DEFAULT_OFF
    pin: GPIO0
    id: relay1
    name: $devicename " Relay"
    
sensor:
  - platform: dht
    pin: GPIO3
    model: AM2302
    temperature:
      id: temp1
      name: $devicename " Temperature"
                    
    humidity:
      name: $devicename " Humidity"
    update_interval: 10s
    
script:
  - id: test1
    then:
      - lambda: !lambda |-
          ESP_LOGD("main", "State is: %f", id(temp1).state);

interval:
  - interval: 10s
    then:
      - script.execute: test1

climate:
  - platform: bang_bang
    name: test
    id: my_climate
    sensor: temp1
    default_target_temperature_low: 20 °C
    default_target_temperature_high: 22 °C
    visual:
      min_temperature: 5 °C
      max_temperature: 25 °C
      temperature_step: 0.5 °C

    heat_action:
      - switch.turn_on: relay1
    idle_action:
      - switch.turn_off: relay1

```

middelink, 2019-08-22:
as far as i can tell, that's intended. the state is only updated when correctly reading the sensor, I guess mainly to facilitate expiry and stuff.

However, it looks like you're in luck, the dht sensor calls `this->status_set_warning()` when failing to read the hardware. You could test that programmaticly with `id(temp1).status_has_warning()`

john-arvid, 2019-08-22:
Thank you for the input, I can use that instead of a counter I made to check if the sensor still works.

My thought is that it should give back NAN to Home Assistant so the heater can turn off. Or one could have a very dangerous situasion.

middelink, 2019-08-22:
I would think (not verified it yet) that if the sensor stops updating the value, the mqtt_sensor will at expiry time mark the the mqtt topic as empty, at which point HA should do something useful, like run around in circles and yell :)

https://esphome.io/components/sensor/index.html (look for expire_after)

john-arvid, 2019-08-22:
Yes for MQTT it looks like it is being handled (as long HA does something with the expired information, I can't confirm this).
But I am using the API, and there HA keeps the last state.

closed 2019-08-31 by OttoWinter

HFsi, 2019-10-05:
> as far as i can tell, that's intended. the state is only updated when correctly reading the sensor, I guess mainly to facilitate expiry and stuff.
> 
> However, it looks like you're in luck, the dht sensor calls `this->status_set_warning()` when failing to read the hardware. You could test that programmaticly with `id(temp1).status_has_warning()` in some automation rule
> 
> Anyways, looking at the code of DHT, I also see it logging a warning: "Invalid readings! Please check your wiring (pull-up resistor, pin number)", so you should be able to see that in your logging by default.

How do you call this warning status?
I've tried this (and several other combinations) but didn't work:
`- interval: 1s

    then:
      - if:
          condition:
            lambda: |-
              if (id(temp).status_has_warning()) {
                return true;
              } else {
                return false;
              }
          then:
            - logger.log: "Error DHT"`