esphome/issues #591: Esphome fails to start produces a  TypeError

JCrankWV, 2019-08-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

BrianGilbert, 2019-08-16:
this error also occurs with Python 2.7.16

tested on both MacOS 10.14 and 10.15 beta same error on both.

radunCZ, 2019-08-16:
Same errors visible on fresh raspbian with Python 2.7.13 and 2.7.16 (under venv)

radunCZ, 2019-08-16:
cheked the modules version against the vorking environment and voluptous seems to be the issue. Once replaced the newest 0.11.7 by 0.11.5 from the working one, esphome is able to start.

rradar, 2019-08-16:
Had this issue yesterday on a linux machine as well.

To fix install voluptous in version 0.11.5 (neweset 0.11.7 which is installed automatically doesn't work)

brandond, 2019-08-16:
Duplicate of #580

closed 2019-08-16 by brandond

SonnyJim, 2019-09-07:
On a Pi Zero W running Hassbian I took a slightly approach, starting from the Pi user:

sudo -u homeassistant -H -s
source /srv/homeassistant/bin/activate
pip install voluptuous==0.11.5
pip install esphome

neuman1812, 2019-09-27:
Having the same issues.  I upgrade to python 3  and install followed the options from @dannysauer .  

Full log of what I did here: https://hastebin.com/ipozilahuy.sql

Steps: 
Install esphome
Uninstall voluptuous
install python-voluptuous

martin3000, 2019-09-28:
```
pip3 uninstall voluptuous
sudo apt install python3-voluptuous 

```
This fixed it in ubuntu.

rmrfus, 2019-10-02:
It appears that esphome works fine with 0.10.5

```
pip3 uninstall voluptuous
pip install --user voluptuous==0.10.5
```