esphome/issues #592: ESP does not have enough space to store OTA file

robertsorensen, 2019-08-16:
I've posted this on the Home Assistant forum but didn't get any suggestions so hopefully this approach will provide me with some things to try. The remaining post is pasted from the HA post:

I’ve spend some time debugging an issue with an AM2302 temperature and humidity sensor connected to an ESP01.
Without having done anything to my code I suddenly started getting this error message when trying to download code:

Memory Usage -> http://bit.ly/pio-memory-usage 2
DATA: [==== ] 39.6% (used 32424 bytes from 81920 bytes)
PROGRAM: [======= ] 65.3% (used 326480 bytes from 499696 bytes)
========================= [SUCCESS] Took 9.83 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of raindance_esp1_temp3.local
INFO -> 10.0.1.127
INFO Uploading /data/raindance_esp1_temp3/.pioenvs/raindance_esp1_temp3/firmware.bin (330624 bytes)
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)

Here is the code:

> esphome:
  name: raindance_esp1_temp2
  platform: ESP8266
  board: esp01

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

sensor:
  - platform: dht
    pin: GPIO2
    model: AUTO_DETECT
    update_interval: 10s
    temperature:
      name: "Temperature"
      filters:
        - sliding_window_moving_average:
            window_size: 15
            send_every: 15
            send_first_at: 15
    humidity:
      name: "Humidity"
      filters:
        - sliding_window_moving_average:
            window_size: 15
            send_every: 15
            send_first_at: 15

Nothing fancy and nothing advanced and keep in mind that the code used to download.
I tried a couple of things.

I did remove everything below the ota: line - still I get error message
I tried another ESP01 and do get the same error message.
I tried to download the code over a serial port but the download appear to hang after compilation but before download starts. (I use the serial port to initialize the ESP01 the first time and that works OK)
I rebooted my Raspberry Pi where I run HassIO but I doesn’t change anything.
I am running 0.97.2

Any suggestions?

closed 2019-08-16 by close-issue-app[bot]

brandond, 2019-08-16:
Running an OTA update temporarily stores a second copy of the firmware in flash. If your firmware occupies more than half of the flash space, (yours says it's occupying 65%) then there is insufficient space to store a second copy. You can, as the error message suggests try flashing a more minimal build with less features enabled, and the reflash your full build, but this is unlikely to work given how little you already have. Your best bet is to find a device with larger flash size.

robertsorensen, 2019-08-17:
Hi Brandon

Thanks for responding. I did notice that my question were auto closed.
Anyway - I felt that I had tried everything possible. Eventually I did
power cycle my Raspberry Pi where HA is running. This did not address the
issue but it did allow me to download the program via USB. Previously that
had failed and it had failed despite rebooting my HA several times.
After successfully downloading code over USB I was able to download the
same code OTA and while I haven't tried to repeat the OTA download several
times I have a feeling that the power cycle did the trick.

Thanks and really appreciate your willingness to provide help.

Thanks,
Robert

glmnet, 2019-08-17:
Ahhh yes. This is the issue about the docker image not recognizing usb devices? 
May be a warning on the ui would be useful for those souls.

brandond, 2019-08-17:
Yeah, USB flash will always work as long as the image fits in flash. You just have to keep it under 50% if you want to OTA. No idea why the serial adapter wasn't showing up though. Sounds like Docker on Pi can be a bit fiddly.

glmnet, 2019-08-17:
I’m talking about this warning in the middle of the page. I don’t use the hassio add on so I’m not familiar. 

https://esphome.io/guides/getting_started_hassio.html

BarrettLowe, 2019-09-07:
Make sure you've selected the right board in your yaml. The esp-01 has a 512k and a 1M version. It seems like the 512k version is incapable of OTA updates due to size - not 100% sure on that though.