esphome/issues #593: Comments in lambda causes compiling warning

john-arvid, 2019-08-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

**Description of problem:**
When compiling it gives a warning that a comment is a multiline comment. The code still works as expected.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: esp01_mill04

esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret ssid
  password: !secret wpa

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password
  
web_server:
  port: 80

globals:
  - id: counter
    type: int
    restore_value: no
    initial_value: '0'
  - id: check_counter
    type: int
    restore_value: no
    initial_value: '0'
  - id: error_bool
    type: bool
    restore_value: no
    initial_value: 'false'
  - id: max_temp
    type: int
    restore_value: no
    initial_value: '27'

switch:
  - platform: gpio
    restore_mode: RESTORE_DEFAULT_OFF
    pin: GPIO0
    id: relay1
    name: $devicename " Relay"
    
sensor:
  - platform: dht
    pin: GPIO3
    model: AM2302
    temperature:
      id: temp1
      name: $devicename " Temperature"
      on_value:
        then:
          - lambda: !lambda |-
              //If there has been an error before
              if (id(counter) != id(check_counter)) { 
                // Reset counters, since everything is back to normal 
                id(check_counter) = 0;
                id(counter) = 0;
              }
              
              id(counter) += 1;

    humidity:
      name: $devicename " Humidity"
    update_interval: 10s
    
script:
  - id: test1
    then:
      - lambda: !lambda |-
          // Increase counter to be used in error checks
          id(check_counter) += 1; 
          // Check if any difference (If there is a problem with the sensor the counter won't be increased) 
          if (id(counter) != id(check_counter)) {
            // Put a message in the logs //
            ESP_LOGE("main", "Something is wrong with the sensor, turning off heater");
            // Set the heater in OFF mode //
            auto call = id(my_climate).make_call();
            call.set_mode("OFF");
            call.perform();
            
            id(error_bool) = true;
          }
          // Just as an absolute safeguard, not really necesarry (Not tested) 
          else if (id(temp1).state > id(max_temp)) {
            // Put a message in the logs //
            ESP_LOGE("main", "It is too hot here, turning off heater");
            // Set the heater in OFF mode 
            auto call = id(my_climate).make_call();
            call.set_mode("OFF");
            call.perform();
            
            id(error_bool) = true;
          }
          else if (id(error_bool)) {
            // Reset the error bool 
            id(error_bool) = false;
            // Set the heater back in AUTO mode 
            auto call = id(my_climate).make_call();
            call.set_mode("AUTO");
            call.perform();
          }
          
          // Reset variables once in a while
          if (id(check_counter) > 1000) {
            id(check_counter) = 0;
            id(counter) = 0;
          }

interval:
  - interval: 10s
    then:
      - script.execute: test1

climate:
  - platform: bang_bang
    name: test
    id: my_climate
    sensor: temp1
    default_target_temperature_low: 20 °C
    default_target_temperature_high: 22 °C
    visual:
      min_temperature: 5 °C
      max_temperature: 25 °C
      temperature_step: 0.5 °C

    heat_action:
      - switch.turn_on: relay1
    idle_action:
      - switch.turn_off: relay1

```

**Logs (if applicable):**

```
Compiling .pioenvs/esp01_mill04/src/main.cpp.o
src/main.cpp:144:3: warning: multi-line comment [-Wcomment]
   //       - lambda: !lambda "//If there has been an error before\nif (id(counter) != id(check_counter))\
   ^
src/main.cpp:167:3: warning: multi-line comment [-Wcomment]
   //     - lambda: !lambda "// Increase counter to be used in error checks\nid(check_counter)\
   ^
Linking .pioenvs/esp01_mill04/firmware.elf

```

**Additional information and things you've tried:**

Tried to remove the first comment, but then it gives a warning about the next.
Have tried to comment inline with the code, tried with /* */.

brandond, 2019-08-16:
I think this is because the YAML is embedded into the source code as a comment. The warning is due to the fact that this creates a comment within a comment.

manju-rn, 2019-08-18:
Can you dump the entire main.cpp to check  how the code is generated?

glmnet, 2019-08-18:
Considering Brandon is right...
can a compiler flag be used in that case to avoid that warning? 
It would be nice if it could only target main.cpp

john-arvid, 2019-08-18:
main.cpp

```
// Auto generated code by esphome
// ========== AUTO GENERATED INCLUDE BLOCK BEGIN ===========
#include "esphome.h"
using namespace esphome;
using namespace switch_;
logger::Logger *logger_logger;
wifi::WiFiComponent *wifi_wificomponent;
ota::OTAComponent *ota_otacomponent;
api::APIServer *api_server;
using namespace sensor;
gpio::GPIOSwitch *relay1;
dht::DHT *dht_dht;
script::Script *test1;
Automation<> *automation_2;
interval::IntervalTrigger *interval_intervaltrigger;
Automation<> *automation_3;
sensor::Sensor *temp1;
sensor::SensorStateTrigger *sensor_sensorstatetrigger;
Automation<float> *automation;
script::ScriptExecuteAction<> *script_scriptexecuteaction;
globals::GlobalsComponent<int> *counter;
globals::GlobalsComponent<int> *check_counter;
globals::GlobalsComponent<bool> *error_bool;
globals::GlobalsComponent<int> *max_temp;
LambdaAction<float> *lambdaaction;
LambdaAction<> *lambdaaction_2;
sensor::Sensor *sensor_sensor;
// ========== AUTO GENERATED INCLUDE BLOCK END ==========="

void setup() {
  // ===== DO NOT EDIT ANYTHING BELOW THIS LINE =====
  // ========== AUTO GENERATED CODE BEGIN ===========
  // esphome:
  //   name: esp01_mill01
  //   platform: ESP8266
  //   board: esp01_1m
  //   build_path: esp01_mill01
  //   platformio_options: {}
  //   esp8266_restore_from_flash: false
  //   arduino_version: espressif8266@1.8.0
  //   includes: []
  //   libraries: []
  //   board_flash_mode: dout
  App.pre_setup("esp01_mill01", __DATE__ ", " __TIME__);
  // switch:
  // logger:
  //   logs: {}
  //   esp8266_store_log_strings_in_flash: true
  //   level: DEBUG
  //   baud_rate: 115200
  //   tx_buffer_size: 512
  //   hardware_uart: UART0
  //   id: logger_logger
  logger_logger = new logger::Logger(115200, 512, logger::UART_SELECTION_UART0);
  logger_logger->pre_setup();
  App.register_component(logger_logger);
  // wifi:
  //   fast_connect: false
  //   domain: .local
  //   power_save_mode: NONE
  //   reboot_timeout: 5min
  //   id: wifi_wificomponent
  //   networks:
  //   - password: xxx
  //     ssid: xxx
  //     id: wifi_wifiap
  //   use_address: esp01_mill01.local
[... 212 lines cut ...]
      }
  });
  dht_dht->set_temperature_sensor(temp1);
  sensor_sensor = new sensor::Sensor();
  App.register_sensor(sensor_sensor);
  sensor_sensor->set_name("esp01_mill01 \042 Humidity\042");
  sensor_sensor->set_unit_of_measurement("%");
  sensor_sensor->set_icon("mdi:water-percent");
  sensor_sensor->set_accuracy_decimals(0);
  automation_2->add_actions({lambdaaction_2});
  dht_dht->set_humidity_sensor(sensor_sensor);
  dht_dht->set_dht_model(dht::DHT_MODEL_AM2302);
  // =========== AUTO GENERATED CODE END ============
  // ========= YOU CAN EDIT AFTER THIS LINE =========
  App.setup();
}

void loop() {
  App.loop();
}

```

This webpage shows some flags that can be used. (Search for wcomment)
https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html

manju-rn, 2019-08-20:
yeah I guess using -Wcomments will remove those warnings in case the yellow text in Logs are annoying to you. I believe the other way is to define a custom program and call it using .h file.  Since the custom program will be as per the standard there would not be any warnings for comments

john-arvid, 2019-08-20:
Ok, so the solution for this issue would be to use a .h file when writing lambda that needs commenting?
Please close this issue if this is the case.

brandond, 2019-08-20:
Yeah if you want to get fancy or complicated you can put your function in an external file, include that, and call it from your lambda.

https://esphome.io/components/esphome.html#includes

Linwood-F, 2019-08-29:
Just in the FWIW department I just spent a while chasing this as well.  Glad to see the issue posted here, at least it's not something in my syntax.

closed 2019-08-31 by OttoWinter