esphome/issues #594: compiling error

maxmib, 2019-08-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

operating system :armbian aarch64 based debian buster
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
installation: pip
**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32 board
**Affected component:**

none

**Description of problem:**
compiling error ,sh: 1: xtensa-esp32-elf-g++: not found

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test
  platform: ESP32
  board: lolin_d32
wifi:
  ssid: "Xi"
  password: "123456"
# Enable logging
logger:
# Enable Home Assistant API
api:
  password: "123456"
ota:
  password: "123456"

```

**Logs (if applicable):**

```
root@N1-test:~# esphome config/test.yaml run
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d config/test
Processing test (platform: espressif32@1.6.0; framework: arduino; board: lolin_d32)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/lolin_d32.html
PLATFORM: Espressif 32 1.6.0 > WEMOS LOLIN D32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10001.190111 (1.0.1), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for Hash library in registry
Warning! Library `{'requirements': None, 'name': 'Hash'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'Hash'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Found 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <WiFi> 1.0
|-- <Update> 1.0
Compiling .pioenvs/test/src/esphome/components/api/api_message.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/test/src/esphome/components/api/api_server.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
Compiling .pioenvs/test/src/esphome/components/api/basic_messages.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/test/src/esphome/components/api/api_message.cpp.o] Error 127
Compiling .pioenvs/test/src/esphome/components/api/command_messages.cpp.o
*** [.pioenvs/test/src/esphome/components/api/api_server.cpp.o] Error 127
*** [.pioenvs/test/src/esphome/components/api/basic_messages.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [.pioenvs/test/src/esphome/components/api/command_messages.cpp.o] Error 127
========================== [ERROR] Took 5.56 seconds ===================

```

**Additional information and things you've tried:**

manju-rn, 2019-08-20:
you would need to install the toolchain - this post may help
https://github.com/espressif/arduino-esp32/issues/1024

manju-rn, 2019-08-20:
Alternatively, just install the HomeAssistant and configure ESPHome on that, it just works :-)

maxmib, 2019-08-22:
> you would need to install the toolchain - this post may help
> [espressif/arduino-esp32#1024](https://github.com/espressif/arduino-esp32/issues/1024)

When compiling esp32 board firmware ,the plartformio will download  toolchain-xtensa32 automaticaly under .platformio/packages .And the package.json  is below 

"description": "GCC for SoC Xtensa L108",
    "name": "toolchain-xtensa32",
        "system": [
                "linux_armv7l",
                "linux_armv8l",
                "linux_aarch64"
        ],
    "url": "https://github.com/espressif/esp-idf",
    "version": "2.50200.80"

According to this information ,this pacakage is ok with arch64 linux operation system. But when compiling, the logs suggest that  xtensa-esp32-elf-g++ is not found. However this file is under .platformio/packages/toolchain-xtensa32/bin . So I compile xtensa-esp32-elf  following the issue you mentioned above.After replacing some folders in toolchain-xtensa32 using the files I compile ,esphome works.So I think the problem is with toolchain-xtensa32.

OttoWinter, 2019-08-31:
Yes I think the official aarch64 toolchains are broken for some users (but only for certain setups when a specific dynamic library is missing, which has prevented me from sending the patch to platformio directly).

You can use these steps to manually download a toolchain I compiled to run on native aarch64 without armv7 compatibility systems: https://github.com/esphome/esphome-docker-base/blob/master/aarch64/Dockerfile#L36

closed 2019-08-31 by OttoWinter