esphome/issues #595: Hassio esphonme and Node Red addons

mcfarlde, 2019-08-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

brandond, 2019-08-19:
What does your add-on configuration look like? I don't use hassio myself, but it appears to me that the listen port is configured in your hassio add-on `config.json` as opposed to in esphome itself.

mcfarlde, 2019-08-19:
The esphome addon  is standard ssl = true with certfile and keyfile.  The Node Red adding is also standard with ssl, cert and key.  It’s web interface is configured to use port 1880.  The NGinx stuff shows in the logs, I have never used NGinx so I assume it part of adding framework

brandond, 2019-08-19:
It looks like the port used by nginx comes from the hassio addon config - either `direct_port` or `ingress_port`:
https://github.com/esphome/esphome/blob/dev/docker/rootfs/etc/cont-init.d/20-nginx.sh

See: https://developers.home-assistant.io/docs/en/hassio_addon_config.html (see the `ingress_port` option)

Either way this appears to be an issue with your hassio config assigning the same port to multiple addons. If you haven't configured this manually, they're probably both defaulting to the same thing, hence the conflict. I would try manually specifying it for one or both.

mcfarlde, 2019-08-20:
Thanks, I poked around in the set/nginx/servers folder and found a couple of conf files that looked promising.  Nothin calling out port 63418 tho.  I'm not comfortable editing as I'm sure I would cause harm

OttoWinter, 2019-08-31:
@brandond Is right, ingress_port is assigned by the Hass.io supervisor, it looks like either 

 a) you've somehow got two esphome's running at once
 b) hass.io is passing the wrong ingress_port into the container.

Both would be issues with Hass.io supervisor (https://github.com/home-assistant/hassio/issues)

closed 2019-08-31 by OttoWinter