esphome/issues #598: ESP32 Bootloop

crippit, 2019-08-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io running on Docker on a Ubuntu Machine
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**Affected component:**

Booting ESPHome

**Description of problem:**
When flashing an ESP32 with ESPHome the device bootloops and throws an error in the console. The ESP32 seems irrelevant and the issue has been replicated on both an ESP32 (NodeMCU) and ESP32 Cam (AI Thinker)

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

```

**Additional information and things you've tried:**

manju-rn, 2019-08-20:
I see that you have not used any SPI pins 6 - 11 - which is good. You pin config for ESP CAM looks good but I am not sure about the Gpio_4 for flashlight.  Can you remove both this components and see if there is a problem and then work out systematically.  I would first remove the entry for Power Down pin - 32 does not look right to me.

crippit, 2019-08-24:
Thanks!

I'm also encountering the same issue when flashing a regular ESP32 without any pins assigned...

Dbroek, 2019-08-26:
I have the same issue. The funny thing is that with an older ESPHOME yaml, i do not expierence the same issues. 
I'm not sure how i can revert back to an older version of ESPHOME so i can try to find out which build works and which don't. 

Currently i trick ESPHOME by installing the old .BIN file, find the IP address, manually add it in ESPHOME and then rename the device.

Sasso0101, 2019-09-10:
[This pull request](https://github.com/espressif/arduino-esp32/pull/3135) fixed it in the official Arduino library! Does it apply for ESPHome?

rpi3ha, 2019-09-19:
I'm experiencing the same with my ESP32 boards, nothing connected to the board and just a basic config.

elgrande007, 2019-09-21:
Seem to have the same issue I tried the nodemci-32s and esp-wrover-kit
When i flash the board from arduino IDE with a simple program it works.

`esphome:
  name: esp_hobbykamer
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "secret"
  password: "secret

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "secret"

ota:
  password: "secret"
`

rpi3ha, 2019-09-22:
As @Sasso0101 wrote above I believe this is the problem for my ESP32 boards as well. I see it's compiling for 240 MHz and my ESP32 boards are only 160 MHz. Here's from the compile log `HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)`
Any way to override this in ESPHome? So it's compiled for 160 MHz instead of 240 MHz. ref. https://github.com/espressif/arduino-esp32/pull/3135

SeByDocKy, 2019-09-29:
Same here .... got the inifinite boot loop with a DWROOM board. Works fine with any  arduino IDE samples .... :(

Maybe by playing with platformio_options:  ?

SeByDocKy, 2019-09-29:
I found a fix (Thanks @Eurotrash): esphomeflasher.exe your_esphome_firmware.bin --bootloader bootloader_qio_80m.bin  

where the boot loader can be found here: https://github.com/espressif/arduino-esp32/blob/b588aff6d6edfdc764c2d687a3f1b7242e96b176/tools/sdk/bin/bootloader_qio_80m.bin

After this first flashing, OTA works fine

rpi3ha, 2019-09-30:
Confirmed working here as well!

Sasso0101, 2019-09-30:
works for me, too :)

elgrande007, 2019-09-30:
Tried the "bootloader_qio_80m.bin" And below the flash and boot log. it keeps on going...

```Using 'COM3' as serial port.
Connecting....
Detecting chip type... ESP32
Connecting....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 160MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 84:0D:8E:0E:86:A8
Uploading stub...
Running stub...
Stub running...
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 80MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 7.6s
Compressed 13696 bytes to 8946...
Writing at 0x00001000... (100 %)Wrote 13696 bytes (8946 compressed) at 0x00001000 in 0.9 seconds (effective 122.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 143...
Writing at 0x00008000... (100 %)Wrote 3072 bytes (143 compressed) at 0x00008000 in 0.1 seconds (effective 393.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 1048.6 kbit/s)...
Hash of data verified.
Compressed 21360 bytes to 13219...
Writing at 0x00010000... (100 %)Wrote 21360 bytes (13219 compressed) at 0x00010000 in 1.3 seconds (effective 136.3 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[22:39:53]ets Jun  8 2016 00:22:57
[22:39:53]
[22:39:53]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:39:53]configsip: 0, SPIWP:0xee
[22:39:53]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:39:53]mode:DIO, clock div:1
[22:39:53]load:0x3fff0018,len:4
[22:39:53]load:0x3fff001c,len:808
[22:39:53]load:0x40078000,len:6084
[22:39:53]load:0x40080000,len:6696
[22:39:53]entry 0x400802e4
[22:39:53]user code done
[22:39:53]ets Jun  8 2016 00:22:57
[22:39:53]
[22:39:53]rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:39:53]configsip: 0, SPIWP:0xee
[22:39:53]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:39:53]mode:DIO, clock div:1
[22:39:53]load:0x3fff0018,len:4
[22:39:53]load:0x3fff001c,len:808
[22:39:53]load:0x40078000,len:6084
[22:39:53]load:0x40080000,len:6696
[22:39:53]entry 0x400802e4
[22:39:53]user code done
[22:39:53]ets Jun  8 2016 00:22:57

```

SeByDocKy, 2019-10-01:
Still Bootloop ?

elgrande007, 2019-10-01:
Yes, it keeps on rebooting.

crippit, 2019-10-11:
I'm still having the same issue here as well. I'm using a Mac so I can't use esphomeflasher.exe to flash the suggested file. Any updates on this issue?

SeByDocKy, 2019-10-11:
> I'm still having the same issue here as well. I'm using a Mac so I can't use esphomeflasher.exe to flash the suggested file. Any updates on this issue?

You hace to use the esphome flasher python files .... not the  exe

glmnet, 2019-10-11:
can you post the full command for reference?

SeByDocKy, 2019-10-11:
> can you post the full command for reference?

in python: esphomeflasher your_esphome_firmware.bin --bootloader bootloader_qio_80m.bin

glmnet, 2019-10-11:
@elgrande007 did you use that command? I got it working with this
thanks @SeByDocKy

glmnet, 2019-10-11:
I also notice that esphome cannot reset the ESP32, whereas esphomeflasher does it

OttoWinter, 2019-10-15:
@glmnet 

> I also notice that esphome cannot reset the ESP32, whereas esphomeflasher does it

Both use esptool inside, so there should be no difference - but I have noticed that it can depend on the host system. For example on an RPi the automatic reset failed when the RPi didn't have sufficient power.

The 80kb bootloader is something I will have to look into, probably will require an update of platformio-espressif32 package to add the new board types.

Also: This exact error message can show up too if the CLK pin on the ESP32 is pulled low. And it just so happens that there are some NodeMCU-ESP32 sellers out there selling boards with wrong silkscreen: it says GND on a pin but it really is CLK. If you connect anything there the device will not boot.

satekp, 2019-10-17:
I have same issue with some esp 32 modules.
But could upload esphom OTA ,after uploading web updater

Imperial-Guard, 2019-10-24:
> I found a fix (Thanks @eurotrash): esphomeflasher your_esphome_firmware.bin --bootloader bootloader_qio_80m.bin

Does this work with the ESPHome-Flasher-1.1.0-windows-x64 ? Or is this all via command line

glmnet, 2019-10-24:
@Imperial-Guard I got it working in the windows command line

SeByDocKy, 2019-10-24:
> > I found a fix (Thanks @eurotrash): esphomeflasher your_esphome_firmware.bin --bootloader bootloader_qio_80m.bin
> 
> Does this work with the ESPHome-Flasher-1.1.0-windows-x64 ? Or is this all via command line

I tried with the exe windows file but seems it not possible to pass arguments through.... so for me it worked with the python codes

SeByDocKy, 2019-10-24:
> @Imperial-Guard I got it working in the windows command line

Can you write the exact windows command line you used ?

OttoWinter, 2019-10-24:
Ok, I can update the bootloaders in ESPHome no problem. However, I would first like to check if the new bootloaders are backwards-compatible. Can someone check if an ESPHome binary with `arduino_version: 1.0.1` works with the new bootloader (with the command line arg `--bootloader bootloader_qio_80m.bin`).

glmnet, 2019-10-24:
> > @Imperial-Guard I got it working in the windows command line
> 
> Can you write the exact windows command line you used ?

I was testing using the command again and for some reason it does not work on Command line (cmd.exe) nor does it work on power shell

I have git bash installed and it works there, you download the file from the releases pages then the command:

`$ ./ESPHome-Flasher-1.1.0-windows-x64.exe -h` 

outputs usage information properly. (With PS it just writes nothing)

OttoWinter, 2019-10-24:
Ok, this should be fixed with https://github.com/esphome/esphome-flasher/pull/5 - that updates the bootloader to the latest version. Was also able to confirm that the newer bootloaders work with older esp-idf versions, so that's good. Also automated the ESPHome-Flasher build pipeline, because spinning up a windows VM each time I wanted to publish a release was making releasing new versions a hassle.

OttoWinter, 2019-10-24:
This issue is now fixed. I released [esphome-flasher v1.2.0](https://github.com/esphome/esphome-flasher/releases/tag/v1.2.0) that contains a fix for this issue.

closed 2019-10-24 by OttoWinter