esphome/issues #602: compile error 4

nelbs, 2019-08-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassbian
Esphome installed with pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Description of problem:**
Im trying to compile my first file but I get stuck. As you can see i get some erors. Any idea what Iam doing wrong?
Updating esphome did not do the trick.

**Problem-relevant YAML-configuration entries:**
```yaml
---
esphome:
  name: deurbel
  platform: ESP8266
  board: esp01_1m

# WiFi connection, correct these
# with values for your WiFi.
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable logging.
logger:

# Enable Home Assistant API.
api:

# Enable over-the-air updates.
ota:

# Enable Web server.
web_server:
  port: 80

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: Doorbell ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: Doorbell IP
    ssid:
      name: Doorbell SSID
    bssid:
      name: Doorbell BSSID

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: Doorbell Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: Doorbell WiFi Signal
    update_interval: 60s

# Global to store the on/off state of the chime
globals:
  - id: chime
    type: bool
    restore_value: true
    initial_value: 'true'

# Exposed switches.
switch:
  # Switch to restart the doorbell.
  - platform: restart
    name: Doorbell Restart

  # Switch to turn on/off the chime.
  - platform: gpio
    id: relay
    inverted: true
    name: Doorbell Chime
    pin: GPIO0

  # Switch to turn on/off chime when
  # doorbell button is pushed.
  #
  # It creates a "virtual" switch based
  # on a global variable.
  - platform: template
    name: Doorbell Chime Active
    id: chime_active
    restore_state: false
    turn_on_action:
      - globals.set:
          id: chime
          value: 'true'
    turn_off_action:
      - globals.set:
          id: chime
          value: 'false'
    lambda: |-
      return id(chime);

# Binary sensor representing the
# Doorbell button push.
binary_sensor:
  - platform: gpio
    id: button
    name: Doorbell Button
    pin:
      # Connected to GPIO on the ESP-01S.
      number: GPIO2
      mode: INPUT_PULLUP
      inverted: true
    filters:
      # Small filter, to debounce the button press.
      - delayed_on: 25ms
      - delayed_off: 25ms
    on_press:
      # Only turn on the chime when it is active.
      then:
        if:
          condition:
            - switch.is_on: chime_active
          then:
            - switch.turn_on: relay
    on_release:
      # On release, turn of the chime.
      - switch.turn_off: relay

```

**Logs (if applicable):**

```
INFO Reading configuration...
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 03/29/20 02:00:00 to 10/25/20 03:00:00
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d config/deurbel
Processing deurbel (framework: arduino; platform: espressif8266@1.8.0; board: esp01_1m)
--------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 1.8.0 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: toolchain-xtensa 1.40802.2 (4.8.2), tool-esptool 1.413.0 (4.13), framework-arduinoespressif8266 2.20402.4 (2.4.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for Update library in registry
Warning! Library `{'requirements': None, 'name': 'Update'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'Update'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP Async WebServer> 1.1.1
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/deurbel/src/esphome/components/api/api_message.cpp.o
Compiling .pioenvs/deurbel/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/deurbel/src/esphome/components/api/basic_messages.cpp.o
Compiling .pioenvs/deurbel/src/esphome/components/api/command_messages.cpp.o
as: loadlocale.c:130: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.
xtensa-lx106-elf-g++: internal compiler error: Aborted (program as)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [.pioenvs/deurbel/src/esphome/components/api/api_message.cpp.o] Error 4
as: loadlocale.c:130: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.
xtensa-lx106-elf-g++: internal compiler error: Aborted (program as)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [.pioenvs/deurbel/src/esphome/components/api/command_messages.cpp.o] Error 4
as: loadlocale.c:130: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.
xtensa-lx106-elf-g++: internal compiler error: Aborted (program as)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [.pioenvs/deurbel/src/esphome/components/api/basic_messages.cpp.o] Error 4
as: loadlocale.c:130: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.
xtensa-lx106-elf-g++: internal compiler error: Aborted (program as)
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [.pioenvs/deurbel/src/esphome/components/api/api_server.cpp.o] Error 4
=============================================== [ERROR] Took 12.23 seconds ===============================================

```

**Additional information and things you've tried:**

nelbs, 2019-08-21:
`export LC_ALL=C`

solved my problem!

closed 2019-08-21 by nelbs

birdie60, 2019-10-01:
Thanks, had the same issue.
In what script file did you place "export LC_ALL=C"?

nelbs, 2019-10-01:
No script... It is a environment variable of your system. You have to enter it in your terminal.

samrat3star, 2019-10-17:
I m using Docker on Raspbian Buster lite should i login to hassio SSH or Raspbian SSH to enter the export LC_ALL=C ?