esphome/issues #603: All devices showing as "offline" but they really aren't

finity69x2, 2019-08-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

non-hassio HA in Docker on NUC running ESPHome v0.13.6

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff, Shelly & NodeMCU

**Affected component:**

dashboard

**Description of problem:**

All devices show as "offline" in the ESPHome dashboard installed in docker.

As far as i can tell everything is working the way it should be - I can upload configs, see the OTA logs, etc.

It just shows that everything is offline all the time.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

caphm, 2019-08-22:
Same here. Running the dashboard in a docker container. When I execute ping in the container, the nodes all answer successfully.

Anonym-tsk, 2019-08-27:
Use `--network host` in your container.
Or set `ESPHOME_DASHBOARD_USE_PING` environment variable to `True` in your container.
Or set `use_address` in your YAML.

caphm, 2019-08-27:
@Anonym-tsk setting ESPHOME_DASHBOARD_USE_PING solved it. Thanks :)

finity69x2, 2019-08-30:
@Anonym-tsk 

thanks, I didn't try --net=host but adding the environment variable fixed it. 

Here is the complete variable to add:

```
-e ESPHOME_DASHBOARD_USE_PING="true"
```

closed 2019-08-30 by finity69x2