esphome/issues #604: Sun component: can't get timestamp value from sunrise

herrwusel, 2019-08-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

sun

**Description of problem:**
I want to put my ESP to sleep on daylight. Therefore I want to utilize the sun component to actually calculate each daylight time in seconds from the sun rise and set data. Since both values come in strftime and strptime can't be used, member function sunrise and sunset came to mind ([docs](https://esphome.io/api/classesphome_1_1sun_1_1_sun.html)). Whenever I try to call the timestamp value I get the error below. I am a bit confused: according to [this](https://esphome.io/api/structesphome_1_1time_1_1_e_s_p_time.html) there should be a time stamp available.

**Problem-relevant YAML-configuration entries:**
```yaml
sun:
  latitude: !secret latitude
  longitude: !secret longitude
  id: sun_sun

sensor:
  - platform: template
    name: "Template Sensor"
    id: test_sensor
    lambda: |-
      return id(sun_sun).sunrise().timestamp;

```

**Logs (if applicable):**

```
src/main.cpp: In lambda function:
src/main.cpp:289:33: error: 'class esphome::optional<esphome::time::ESPTime>' has no member named 'timestamp'
return sun_sun->sunrise().timestamp;
^
src/main.cpp:291:3: warning: control reaches end of non-void function [-Wreturn-type]
});
^

```

**Additional information and things you've tried:**

brandond, 2019-08-22:
You don't say what version you're on. Are you perhaps using an older release that has `time` instead of `timestamp`?

herrwusel, 2019-08-23:
Sorry, I forget to mention the version: ESPHome 1.13.6. `time` doesn't work either.

OttoWinter, 2019-08-27:
That would be because it's returning an `optional` object, so you need to use `return id(sun_sun).sunrise()->timestamp;` or better:

```cpp
auto sunrise = id(sun_sun).sunrise();
if (!sunrise.has_value())
  return 0;
return sunrise.value().timestamp;
```

closed 2019-08-27 by OttoWinter