esphome/issues #612: esp32_ble_tracker set_scan_interval doesn't change the interval immediately

westenberg, 2019-08-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32

**Affected component:**
https://esphome.io/components/esp32_ble_tracker.html

**Description of problem:**
I use an api service to set the set_scan_interval on the esp32_ble_tracker so I can change the polling interval when e.g. motion is detected. Changing set_scan_interval works but it looks like it finishes the current interval first before it sets the new interval. This makes the function unusable for this purspose because when setting a 24h interval and going back to 5s it needs to finish the 24h interval first.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: zolder
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: ""
  password: ""

logger:

api:
  password: ""
  services:
    - service: ble_start_scan
      then:
        - lambda: |-
            id(ble_tracker).set_scan_interval(10);
    - service: ble_stop_scan
      then:
        - lambda: |-
            id(ble_tracker).set_scan_interval(86400);

ota:
  password: ""
  
esp32_ble_tracker:
  id: ble_tracker
  scan_interval: 60s
  
binary_sensor:
  - platform: ble_presence
    mac_address: 12:34:56:78:90:AB
    name: Device

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I think the solution would be to cancel the previous interval immediately and set the new interval.

Lisheng2016, 2019-08-25:
would adding a update action after setting the interval work? 
```
  services:
    - service: ble_start_scan
      then:
        - lambda: |-
            id(ble_tracker).set_scan_interval(10);
            id(ble_tracker).update();
    - service: ble_stop_scan
      then:
        - lambda: |-
            id(ble_tracker).set_scan_interval(86400);
            id(ble_tracker).update();
```

[component_update action](https://esphome.io/guides/automations.html#component-update-action)

westenberg, 2019-08-25:
Will check it. As far as I can see in the [api docs](https://esphome.io/api/classesphome_1_1esp32__ble__tracker_1_1_e_s_p32_b_l_e_tracker.html) the ESP32BLETracker doesn't have the update function.

In the [esp_gap_ble docs](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/esp_gap_ble.html) there is a esp_ble_gap_stop_scanning function as well. I think something needs to be added to set_interval to make sure it calls esp_ble_gap_stop_scanning before setting the new interval. Next the ESP32BLETracker::gap_event_handler should perhaps handle something (didn't figure this out yet) and then ESP32BLETracker::start_scan should be called again to start the cycle again.

OttoWinter, 2019-08-31:
No, that function is for internal use to be called from the python code generator. It's a static value and does not support changing the scan interval at runtime.

closed 2019-08-31 by OttoWinter