esphome/issues #614: OTA FAIL

Uakkauakkafo, 2019-08-25:
hello,
I have a fail during OTA:

Traceback (most recent call last):
  File "/usr/local/bin/esphomeyaml", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 505, in main
    return run_esphomeyaml(sys.argv)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 495, in run_esphomeyaml
    return POST_CONFIG_ACTIONS[args.command](args, config)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 310, in command_run
    exit_code = upload_program(config, args, port)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 199, in upload_program
    res = espota2.run_ota(host, remote_port, password, CORE.firmware_bin)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/espota2.py", line 281, in run_ota
    return run_ota_impl_(remote_host, remote_port, password, filename)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/espota2.py", line 266, in run_ota_impl_
    file_handle = open(filename, 'rb')
IOError: [Errno 2] No such file or directory: u'config/esp32_dth11/.pioenvs/esp32_dth11/firmware.bin'
this is my file:

esphomeyaml:
  name: esp32_dth11
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: 'mySSID'
  password: 'myPWD'

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

sensor:
  - platform: dht
    pin: 15
    model: DHT22
    temperature:
      name: "Temperatura Studio"
    humidity:
      name: "Umidità Studio"
    update_interval: 60s

switch:
  - platform: gpio
    pin: 23
    name: "Green Light"
    id: led1
  - platform: gpio
    pin: 22
    name: "Blue Light"
    id: led2
  - platform: gpio
    pin: 21
    name: "Red Light"
    id: led3

USB flash work, but via OTA not.

I have "esphome" and "esphomeyaml" installed on a Raspberry Pi 3B+ with Raspbian.
I Run Home Assistant on another Raspberry.

I have also another problem and I don't know if is linked:

I use "esphomeyaml" and flash via USB, all works, I'm also able to link the all in Home Assistant and use it.

If I use "esphome" I had that issue:

~ $ esphome ESP32_DTH11.yaml run
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 7, in <module>
    from esphome.__main__ import main
  File "/usr/local/lib/python2.7/dist-packages/esphome/__main__.py", line 10, in <module>
    from esphome import const, writer, yaml_util
  File "/usr/local/lib/python2.7/dist-packages/esphome/writer.py", line 7, in <module>
    from esphome.config import iter_components
  File "/usr/local/lib/python2.7/dist-packages/esphome/config.py", line 15, in <module>
    from esphome import core, core_config, yaml_util
  File "/usr/local/lib/python2.7/dist-packages/esphome/core_config.py", line 6, in <module>
    import esphome.config_validation as cv
  File "/usr/local/lib/python2.7/dist-packages/esphome/config_validation.py", line 1242, in <module>
    Optional(CONF_COMMAND_TOPIC): All(requires_component('mqtt'), subscribe_topic),
  File "/usr/local/lib/python2.7/dist-packages/esphome/voluptuous_schema.py", line 198, in extend
    ret = super(_Schema, self).extend(schema, extra=extra)
  File "/usr/local/lib/python2.7/dist-packages/voluptuous/schema_builder.py", line 782, in extend
    return result_cls(result, required=result_required, extra=result_extra)
TypeError: __init__() got an unexpected keyword argument 'required'

Thanks for your support

glmnet, 2019-08-25:
Are you running up do date esphome version?
esphomeyaml is old stuff, please add version info of your install

Uakkauakkafo, 2019-08-25:
I made that command:
"sudo pip2 install -U esphome"

and all is up to date:

```
Requirement already up-to-date: esphome in /usr/local/lib/python2.7/dist-packages
Requirement already up-to-date: ifaddr<1,>=0.1.6 in /usr/local/lib/python2.7/dist-packages (from esphome)
Requirement already up-to-date: platformio>=3.6.5 in /usr/local/lib/python2.7/dist-packages (from esphome)
Requirement already up-to-date: protobuf<3.8,>=3.7 in /usr/local/lib/python2.7/dist-packages (from esphome)
Requirement already up-to-date: PyYAML<6,>=5.1 in /usr/local/lib/python2.7/dist-packages (from esphome)
Requirement already up-to-date: paho-mqtt<2,>=1.4 in /usr/local/lib/python2.7/dist-packages (from esphome)
Requirement already up-to-date: pyserial<4,>=3.4 in /usr/local/lib/python2.7/dist-packages (from esphome)
Requirement already up-to-date: tornado<6,>=5.1.1 in /usr/local/lib/python2.7/dist-packages (from esphome)
Requirement already up-to-date: voluptuous<0.12,>=0.11.5 in /usr/local/lib/python2.7/dist-packages (from esphome)
Requirement already up-to-date: colorlog>=4.0.2 in /usr/local/lib/python2.7/dist-packages (from esphome)
Requirement already up-to-date: typing>=3.6.6; python_version < "3.5" in /usr/local/lib/python2.7/dist-packages (from esphome)
Requirement already up-to-date: esptool<3,>=2.6 in /usr/local/lib/python2.7/dist-packages (from esphome)
Requirement already up-to-date: tzlocal>=1.5.1 in /usr/local/lib/python2.7/dist-packages (from esphome)
Requirement already up-to-date: ipaddress in /usr/local/lib/python2.7/dist-packages (from ifaddr<1,>=0.1.6->esphome)
Requirement already up-to-date: semantic-version<3,>=2.5.0 in /usr/local/lib/python2.7/dist-packages (from platformio>=3.6.5->esphome)
Requirement already up-to-date: requests<3,>=2.4.0 in /usr/local/lib/python2.7/dist-packages (from platformio>=3.6.5->esphome)
Requirement already up-to-date: bottle<0.13 in /usr/local/lib/python2.7/dist-packages (from platformio>=3.6.5->esphome)
Requirement already up-to-date: colorama in /usr/local/lib/python2.7/dist-packages (from platformio>=3.6.5->esphome)
Requirement already up-to-date: click<8,>=5 in /usr/local/lib/python2.7/dist-packages (from platformio>=3.6.5->esphome)
Requirement already up-to-date: tabulate>=0.8.3 in /usr/local/lib/python2.7/dist-packages (from platformio>=3.6.5->esphome)
Requirement already up-to-date: six>=1.9 in /usr/lib/python2.7/dist-packages (from protobuf<3.8,>=3.7->esphome)
Requirement already up-to-date: setuptools in /usr/local/lib/python2.7/dist-packages (from protobuf<3.8,>=3.7->esphome)
Requirement already up-to-date: singledispatch in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=5.1.1->esphome)
Requirement already up-to-date: futures in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=5.1.1->esphome)
Requirement already up-to-date: backports-abc>=0.4 in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=5.1.1->esphome)
Requirement already up-to-date: ecdsa in /usr/local/lib/python2.7/dist-packages (from esptool<3,>=2.6->esphome)
Requirement already up-to-date: pyaes in /usr/local/lib/python2.7/dist-packages (from esptool<3,>=2.6->esphome)
Requirement already up-to-date: pytz in /usr/local/lib/python2.7/dist-packages (from tzlocal>=1.5.1->esphome)
Requirement already up-to-date: idna<2.9,>=2.5 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.4.0->platformio>=3.6.5->esphome)
Requirement already up-to-date: chardet<3.1.0,>=3.0.2 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.4.0->platformio>=3.6.5->esphome)
Requirement already up-to-date: certifi>=2017.4.17 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.4.0->platformio>=3.6.5->esphome)
Requirement already up-to-date: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.4.0->platformio>=3.6.5->esphome)

```
I have that version:

esphome-1.13.6.dist-info  

I found that information under  "/usr/local/lib/python2.7/dist-packages/"

any suggestion?

Uakkauakkafo, 2019-08-25:
Solved!!!

I follow that comment: "@nelbs I ran into this issue too. I fixed by downgrading voluptuous using pip install voluptuous==0.11.5."

read here --> [https://github.com/esphome/issues/issues/579]

thanks!!!

closed 2019-08-25 by Uakkauakkafo

davey, 2019-10-09:
Hi,
is there another workaround for that, as downgrading voluptuous isn't an option for me any more? With homeassistant running on the same machine, I need to have voluptuous in version >=0.11.7, see this message when trying to downgrade:

```
$ pip install voluptuous==0.11.5
[...]
ERROR: homeassistant 0.99.3 has requirement voluptuous==0.11.7, but you'll have voluptuous 0.11.5 which is incompatible.
[...]
```

davey, 2019-10-09:
Not sure what I did - but now it works here too, sorry for the noise...