esphome/issues #618: Custom Climate component example is missing `ClimateTraits`

NdS-Research-Facilities, 2019-08-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on VMware on Win10 on Lenovo P50 laptop

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-S

**Affected component:**

https://esphome.io/components/climate/custom.html

**Description of problem:**

compilation fails with error:

```
INFO Reading configuration /config/esphome/807d3ab9d974.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/807d3ab9d974
Processing 807d3ab9d974 (board: nodemcu-32s; framework: arduino; platform: espressif32@1.6.0)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 1.6.0 > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10001.190111 (1.0.1), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for Hash library in registry
Warning! Library `{'requirements': None, 'name': 'Hash'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'Hash'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESP Async WebServer> 1.2.2
|   |-- <AsyncTCP> 1.0.3
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/807d3ab9d974/.pioenvs/807d3ab9d974/src/main.cpp.o
src/main.cpp: In lambda function:
src/main.cpp:216:56: error: invalid new-expression of abstract class type 'OpenThermClimate'
       auto my_opentherm_climate = new OpenThermClimate();
                                                        ^
In file included from src/main.cpp:35:0:
src/OpenThermClimate.h:3:7: note:   because the following virtual functions are pure within 'OpenThermClimate':
 class OpenThermClimate : public Component, public Climate {
       ^
In file included from src/esphome/core/application.h:27:0,
                 from src/esphome/components/api/api_connection.h:4,
                 from src/esphome.h:2,
                 from src/main.cpp:3:
src/esphome/components/climate/climate.h:185:25: note: 	virtual esphome::climate::ClimateTraits esphome::climate::Climate::traits()
   virtual ClimateTraits traits() = 0;
                         ^
src/main.cpp:218:35: error: could not convert '{my_opentherm_climate}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::climate::Climate*>'
       return {my_opentherm_climate};
                                   ^
*** [/data/807d3ab9d974/.pioenvs/807d3ab9d974/src/main.cpp.o] Error 1
========================== [FAILED] Took 2.56 seconds ==========================

```

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
  includes:
    - OpenThermClimate.h

climate:
- platform: custom
  lambda: |-
    auto my_opentherm_climate = new OpenThermClimate();
    App.register_component(my_opentherm_climate);
    return {my_opentherm_climate};

  climates:
    - name: "OpenThermClimate"

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
see above
```

**Additional information and things you've tried:**

adding 
```
  climate::ClimateTraits traits() {
    return ClimateTraits();
  }
```
solves compile issue

NdS-Research-Facilities, 2019-08-25:
esphome.io dev chain 1.13.6

glmnet, 2019-08-26:
Thanks, could you [edit the page on github](https://github.com/esphome/esphome-docs/blob/current/components/climate/custom.rst) and send the PR with this and your other issue?
Or otherwise put here the full working code and I can replace it.

closed 2020-12-05 by stale[bot]