esphome/issues #619: esp32 cam - DIY MORE -  camera video not working

gnkarn, 2019-08-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

esp32 cam DIY MORE , it  looks identical to AI thinker eso32 cam

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**
The cammera is not working , flash led, and status led work fine , but cammera has no image

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32_cam
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: 'xx'
  password: 'xx'
# Optional manual IP
  manual_ip:
    static_ip: 192.168.0.36
    gateway: 192.168.0.1
    subnet: 255.255.255.0

ota:
  password: 'xx'    

# Example configuration entry
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

web_server:
  port: 80
  
# Example configuration entry
esp32_camera:
  name: esp32_camera
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  test_pattern: True

  
#esp32_touch:
#  setup_mode: True

#binary_sensor:
#  - platform: esp32_touch
#    name: "boton camara esp"
#    pin: GPIO13
#    threshold: 0
    
#deep_sleep:
#  wakeup_pin: GPIO12

output:
  # Register the  LED as a dimmable output ....
  - platform: ledc
    id: flash_led
    pin: GPIO4
#    inverted: True

  # Register the  LED as a dimmable output ....
  - platform: ledc
    id: status_led
    pin: GPIO33
    inverted: True

light:
  # ... and then make a light out of it.
  - platform: monochromatic
    name: "camara32 LED"
    output: flash_led
  - platform: monochromatic
    name: "camara32 status"
    output: status_led

```

**Logs (if applicable):**

```
[20:27:43][C][wifi:372]: WiFi:
[20:27:43][C][wifi:254]:   SSID: [redacted]
[20:27:43][C][wifi:255]:   IP Address: 192.168.0.36
[20:27:43][C][wifi:257]:   BSSID: [redacted]
[20:27:43][C][wifi:258]:   Hostname: 'esp32_cam'
[20:27:43][C][wifi:262]:   Signal strength: -69 dB ▂▄▆█
[20:27:43][C][wifi:263]:   Channel: 11
[20:27:43][C][wifi:264]:   Subnet: 255.255.255.0
[20:27:43][C][wifi:265]:   Gateway: 192.168.0.1
[20:27:43][C][wifi:266]:   DNS1: 0.0.0.0
[20:27:43][C][wifi:267]:   DNS2: 0.0.0.0
[20:27:43][C][ledc.output:032]: LEDC Output:
[20:27:43][C][ledc.output:033]:   PinGPIO4 (Mode: OUTPUT)
[20:27:43][C][ledc.output:034]:   LEDC Channel: 0
[20:27:43][C][ledc.output:035]:   Frequency: 1000.0 Hz
[20:27:43][C][ledc.output:036]:   Bit Depth: 15
[20:27:43][C][ledc.output:032]: LEDC Output:
[20:27:43][C][ledc.output:033]:   PinGPIO33 (Mode: OUTPUT)
[20:27:43][C][ledc.output:034]:   LEDC Channel: 1
[20:27:43][C][ledc.output:035]:   Frequency: 1000.0 Hz
[20:27:43][C][ledc.output:036]:   Bit Depth: 15
[20:27:43][C][logger:137]: Logger:
[20:27:43][C][logger:138]:   Level: VERBOSE
[20:27:43][C][logger:139]:   Log Baud Rate: 115200
[20:27:43][C][logger:140]:   Hardware UART: UART0
[20:27:43][C][light:174]: Light 'camara32 LED'
[20:27:43][C][light:176]:   Default Transition Length: 1.0s
[20:27:43][C][light:177]:   Gamma Correct: 2.80
[20:27:43][C][light:174]: Light 'camara32 status'
[20:27:43][C][light:176]:   Default Transition Length: 1.0s
[20:27:43][C][light:177]:   Gamma Correct: 2.80
[20:27:43][C][esp32_camera:043]: ESP32 Camera:
[20:27:43][C][esp32_camera:044]:   Name: esp32_camera
[20:27:43][C][esp32_camera:045]:   Board Has PSRAM: YES
[20:27:43][C][esp32_camera:047]:   Data Pins: D0:5 D1:18 D2:19 D3:21 D4:36 D5:39 D6:34 D7:35
[20:27:43][C][esp32_camera:048]:   VSYNC Pin: 25
[20:27:43][C][esp32_camera:049]:   HREF Pin: 23
[20:27:43][C][esp32_camera:050]:   Pixel Clock Pin: 22
[20:27:43][C][esp32_camera:051]:   External Clock: Pin:0 Frequency:20000000
[20:27:43][C][esp32_camera:052]:   I2C Pins: SDA:26 SCL:27
[20:27:43][C][esp32_camera:053]:   Reset Pin: -1
[20:27:43][C][esp32_camera:074]:   Resolution: 640x480 (VGA)
[20:27:43][E][esp32_camera:091]:   Setup Failed: ERROR
[20:27:43][C][web_server:125]: Web Server:
[20:27:43][C][web_server:126]:   Address: 192.168.0.36:80
[20:27:43][C][ota:029]: Over-The-Air Updates:
[20:27:43][C][ota:030]:   Address: 192.168.0.36:3232
[20:27:43][C][ota:032]:   Using Password.
[20:27:43][C][api:103]: API Server:
[20:27:43][C][api:104]:   Address: 192.168.0.36:6053

```

**Additional information and things you've tried:**

there may be a difference , between those two boards, but couldnt find it yet , they look identical  .

OttoWinter, 2019-08-31:
Try without the ledc platform - the camera uses ledc internally to generate the pixel clock and that might interfere here. Also see https://github.com/esphome/issues/issues/435 please

MiloszP, 2019-09-05:
Actually, there are few wersion of esp32 cam, while looking the same the gpio for camera are in different order. That was my case. I found a documentation file on shops website and changed GPIOs.

myoung34, 2019-09-17:
@MiloszP whats your working config look like?
Ive got a DIYMore esp32 cam that isnt working with my current config

Alex9779, 2019-10-21:
I got the same module, at least it looks just like the picture on the homepage of diymore.cc.
I didn't dig into too much and didn't care yet about LEDs or buttons but following config showed my a picture in HA:
```
esphome:
  name: test_esp32cam
  platform: ESP32
  board: esp32dev

wifi:
  ssid: !secret ponyhof_ssid
  password: !secret ponyhof_password

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

esp32_camera:
  name: test_esp32cam

  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
```

This seems to be the same as your just without the additional stuff... Just the board type is different...

r-jordan, 2020-01-03:
I was having the same problem with the flash light on an ESP32 CAM. It's related with the PWM channel that the camera uses on the ESP32. There's a note in the camera documentation https://esphome.io/components/esp32_camera.html that says channel in the ledc platform should be changed when a camera platform is used. I've introduced this config in the ledc platform to use another channel different of 1 (for example 2) and it works good now:

```
  - platform: ledc
    channel: 2
    id: flash_led
    pin: GPIO4
```
Remember to hard reboot the ESP32 after upload the new code. It seems that the camera PWM channel is not fixed with a soft reboot. And I was almost turning crazy until I noticed!

stanislav-brabec, 2020-11-24:
>     channel: 2

I uploaded a sketch without it to two of my boards. Both stopped to work. After this change, I realized that it fixes the issue, but uploading of the sketch without channel line causes permanent damage. One board seems to have a permanent damage on the board. Second board seems to have a permanent damage of the camera module. Combining two remaining parts, I have one working board and one broken now.

> Remember to hard reboot the ESP32 after upload the new code. It seems that the camera PWM channel is not fixed with a soft reboot. And I was almost turning crazy until I noticed!

What do you mean on hard reboot? Is power off/on and then reset button sufficient?

stanislav-brabec, 2021-02-12:
@r-jordan Actually, your solution works! It is the correct fix.

@OttoWinter It does not help. Even without ledc driver, it stays broken. To fix the module, you have to use ledc again, but with channel 2. Otherwise the channel 0 stays bound to GPIO4. If the channel is not used in the sketch, it seems that the bootloader does not touch these values at all.

closed 2021-06-26 by stale[bot]