esphome/issues #621: unexpected keyword argument 'required'

FauthD, 2019-08-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu 18.04 pip
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, Sonoff T2 dual (I get error long before that)
**Affected component:**

Cover (I get error long before that)

**Description of problem:**
In an empty dir I tried to getting started:
```
~/.local/bin/esphome test.yaml wizard
```
fails:
```
raceback (most recent call last):
  File "/home/fauthd/.local/bin/esphome", line 6, in <module>
    from esphome.__main__ import main
  File "/home/fauthd/.local/lib/python3.6/site-packages/esphome/__main__.py", line 10, in <module>
    from esphome import const, writer, yaml_util
  File "/home/fauthd/.local/lib/python3.6/site-packages/esphome/writer.py", line 7, in <module>
    from esphome.config import iter_components
  File "/home/fauthd/.local/lib/python3.6/site-packages/esphome/config.py", line 15, in <module>
    from esphome import core, core_config, yaml_util
  File "/home/fauthd/.local/lib/python3.6/site-packages/esphome/core_config.py", line 6, in <module>
    import esphome.config_validation as cv
  File "/home/fauthd/.local/lib/python3.6/site-packages/esphome/config_validation.py", line 1242, in <module>
    Optional(CONF_COMMAND_TOPIC): All(requires_component('mqtt'), subscribe_topic),
  File "/home/fauthd/.local/lib/python3.6/site-packages/esphome/voluptuous_schema.py", line 198, in extend
    ret = super(_Schema, self).extend(schema, extra=extra)
  File "/home/fauthd/.local/lib/python3.6/site-packages/voluptuous/schema_builder.py", line 782, in extend
    return result_cls(result, required=result_required, extra=result_extra)
TypeError: __init__() got an unexpected keyword argument 'required'
```

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
None yet
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
I get error long before that
```

**Additional information and things you've tried:**

Install (pip install esphome) had failed with:
```
ERROR: Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/usr/local/lib/python3.6/dist-packages/paho'
Consider using the `--user` option or check the permissions."
```

So I installed with:
pip install esphome --user

Is this the result of Python2/3 issues?  (seeing python3.6 in the path)

FauthD, 2019-08-26:
Trying the same on Rpi3 with Raspian Buster.
Same error (got an unexpected keyword argument 'required').

BTW:
Installation also failed with "got an unexpected keyword argument 'required'".
sudo apt remove python3-yaml
Then EspHome can be installed.

fabaff, 2019-08-26:
`esphome` is Python 2.x-only.

fabaff, 2019-08-27:
There is an open PR (https://github.com/esphome/esphome/pull/702) which will remove the Python 2 support.

FauthD, 2019-08-28:
There is another issue:
[https://github.com/esphome/issues/issues/580](url)
which recommends to downgrade voluptuous lib:
`pip2 install voluptuous==0.11.5`
Though it did not help in my case. (on both: Ubuntu 18.04 and Rpi3/Raspbian)

FauthD, 2019-08-28:
`pip3 install voluptuous==0.11.5`
Did solve the issue.
So it proofs we are using Python3 here.

OttoWinter, 2019-08-29:
@fabaff esphome is not python-2 only, it has supported python3 for a while now (using platformio's devel 4.0.0 branch). Now thatplatformio 4.0.0 has been out and doesn't appear to have any major issues.

OttoWinter, 2019-08-29:
Closing as duplicate of https://github.com/esphome/issues/issues/580

closed 2019-08-29 by OttoWinter