esphome/issues #622: ct_clamp sensor stops processing values from ads1115

jdaniel006, 2019-08-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Pip
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 DevKit
**Affected component:**

ct_clamp sensor

**Description of problem:**
I am trying to read energy consumption with two ct-clamps, an ADS1115 component and one ESP32 clone. It has been noticed several times that the values sent (via MQTT) by the ct-clamp sensors are missing or are incorrect.

Meanwhile, the ads1115 channel readings are, or seem to be, ok. As seen in the logs (sorry, via MQTT), electric/sensor/ads1115_channel_a2-a3/state does have values, while electric_general3 is NaN all the time.

These are the values from the ADS1115 that have been ingested in an InfluxDB:
https://pasteboard.co/IuAXeYp.png
And these are the corresponding current values
https://pasteboard.co/IuAYgMC.png
(notice the no value interval from 19h to 7h)

Power cycling the esp32 device restores the condition to normal, that is, values are in the expected range again.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: cuadro_oficina
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "XXXXXXXXXXX"
  password: "YYYYYYYYYYYYYYYYYYYYYYYYY"
  manual_ip:
    static_ip: ----------------
    gateway: --------------------
    subnet: 255.255.255.0
    dns1: 8.8.8.8
    dns2: 8.8.4.4

# Enable logging
logger:

ota:

mqtt:
  broker: IPIPIPIPIP
  username: 'AAAAAAAAAAAAAAA'
  password: 'BBBBBBBBBBBBB'
  topic_prefix: 'electric'
  birth_message:
    topic: 'electric/status'
    payload: Online
  will_message:
    topic: 'electric/status'
    payload: Offline
  log_topic:
  discovery: False

i2c:
  sda: 21
  scl: 22
  scan: True

ads1115:
  - address: 0x48
    id: 'ads48'

sensor:
  - platform: ct_clamp
    sensor: adc_sensor1
    name: "Measured Current"
    update_interval: 1s
    state_topic: 'electric/corriente'
    discovery: False
    filters:
      - calibrate_linear:
         - 0 -> 0
        - 1.00 -> 30.0
  - platform: ct_clamp
    sensor: adc_sensor2
    name: "Corriente línea 5"
    update_interval: 1s
    state_topic: 'electric/general3'
    discovery: False
    filters:
      - calibrate_linear:
        - 0 -> 0
        - 0.495 -> 100
  - platform: ads1115
    multiplexer: 'A0_A1'
    gain: 2.048
    name: "ADS1115 Channel A0-A1"
    ads1115_id: 'ads48'
    id: adc_sensor1
  - platform: ads1115
    multiplexer: 'A2_A3'
    gain: 1.024
    name: "ADS1115 Channel A2-A3"
    ads1115_id: 'ads48'
    id: adc_sensor2

```

**Logs (if applicable):**

```
Client mosqsub/3323-debian received PUBLISH (d0, q0, r0, m0, 'electric/corriente', ... (5 bytes))
electric/corriente 16.95
Client mosqsub/3323-debian received PUBLISH (d0, q0, r0, m0, 'electric/general3', ... (3 bytes))
electric/general3 nan
Client mosqsub/3323-debian received PUBLISH (d0, q0, r0, m0, 'electric/sensor/ads1115_channel_a2-a3/state', ... (5 bytes))
electric/sensor/ads1115_channel_a2-a3/state 0.108
Client mosqsub/3323-debian received PUBLISH (d0, q0, r0, m0, 'electric/corriente', ... (5 bytes))
electric/corriente 17.87
Client mosqsub/3323-debian received PUBLISH (d0, q0, r0, m0, 'electric/general3', ... (3 bytes))
electric/general3 nan

```

**Additional information and things you've tried:**

OttoWinter, 2019-08-31:
Will be fixed in 1.14 with the `continuous_measurement` option for ads1115

closed 2019-08-31 by OttoWinter

jdaniel006, 2019-09-02:
Thanks a lot!` continuous_measurement ` will be a great improvement. Looking forward to it.