esphome/issues #624: Cannot start hass.io docker container

parrel, 2019-08-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io V0.97.2 running on Proxmox V4.5 in an LXC container.

**ESP (ESP32/ESP8266, Board/Sonoff):**

N/A

**Affected component:**

[https://esphome.io/guides/getting_started_hassio.html](url) ?

**Description of problem:**
I installed the Hass.io add-on and after the installation it could not start. I guess it has something to do with it being ran in an LXC container. 

**Problem-relevant YAML-configuration entries:**
```yaml
N/A
```

**Logs (if applicable):**

```
19-08-27 13:28:32 INFO (SyncWorker_9) [hassio.docker.interface] Clean addon_15ef4d2f_esphome application
19-08-27 13:29:19 ERROR (SyncWorker_9) [hassio.docker] Can't start addon_15ef4d2f_esphome: 500 Server Error: Internal Server Error ("linux runtime spec devices: error gathering device information while adding custom device "/dev/ttyS31": no such file or directory")

```

**Additional information and things you've tried:**

- Updated Hass.io to the latest version
- Rebooted after installation

parrel, 2019-08-28:
This is my config at the moment:
`{ "ssl": false }`

OttoWinter, 2019-08-29:
That sounds like an issue with Hass.io

esphome uses the `auto_uart` option in the addon config to instruct Hass.io to auto-map all host serial devices into the container.

Please post your contents of /dev/ttyUSB* with `ls -lh`, I think you have a broken file there (potentially created with `touch /dev/ttyS31`)

parrel, 2019-08-29:
Thank you for your answer.

```
root@hass-io:~# ls -lh /dev/ttyUSB*
ls: cannot access '/dev/ttyUSB*': No such file or directory
```

```
root@hass-io:~# ls -lh /dev/tty*
crw-rw-rw- 1 root root 5, 0 Aug 28 07:17 /dev/tty
lrwxrwxrwx 1 root root    8 Aug 28 07:17 /dev/tty1 -> lxc/tty1
lrwxrwxrwx 1 root root    8 Aug 28 07:17 /dev/tty2 -> lxc/tty2
```

Is there a way to disable the `auto_uart` option? I am not planning to flash the device via serial using the addon.

parrel, 2019-08-29:
Okay, I forked your hassio repository, disabled `auto_uart` in the config files, committed changes, and added the forked repository as an addon repository in hassio. 

Seems to work fine now!

OttoWinter, 2019-08-31:
@parrel Ok that's good to know.

But that also confirms it's an issue with Hass.io and not esphome - no code on the esphome side has run yet when the error occurs, the container fails to start.

My suspicion is that Hass.io for some reason finds /dev/ttyS31 as a valid serial device when it does not exist. Please post an issue here with all details you have: https://github.com/home-assistant/hassio/issues

closed 2019-08-31 by OttoWinter