esphome/issues #627: Error doing job: Task exception was never retrieved caused by binary_sensor?

voydz, 2019-08-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

hass.io running on a Debian 9 VM

**ESP (ESP32/ESP8266, Board/Sonoff):**

The issue affects one of two exactly the same NodeMCU (lolin) + DHT22 configurations and one out of one Shelly 2.5 device. See YAML config below.

**Affected component:**

As far as I can tell, the issue might be related to the `binary_sensor:` component.

**Description of problem:**
After running weeks without issues after updating to hass 0.97.2 I noticed a few of my ESPhome devices appeared to be offline. I checked my network configuration and found no problems.

I checked the dashboard of the ESPhome addon. All my ESPhome devices are booting up, are displayed as online and spitting out logs OTA just fine. From ESPhome side of things, all seems fine, no issues logged there.

But home assistant is not able to get sensor values for **some** of my devices. Those are displayed as "disconnected" by the `binary_sensor:` status platform. Every sensor and switch affected is "unavailable".

So the strange thing is, I got two other devices with binary_sensors, working just fine. One is an exact clone of the sensor02 device (see YAML config below), which started working after unplugging it for a longer period of time. (Nope, this was not working for the other two.) And I got one RGB Controller D1mini with the `binary_sensor:` status platform which does not seem to be affected at all. Never needed to restart it or something.

[Here are some screenshots](https://imgur.com/a/VWopuCP)

**Problem-relevant YAML-configuration entries:**
### .config.yaml
```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_psk

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_pw

ota:
  password: !secret api_pw
```

### workroom_switch01.yaml (Shelly 2.5)
```yaml
esphome:
  name: workroom_switch01
  platform: ESP8266
  board: esp01_1m

<<: !include .config.yaml
  
sensor:
  - platform: uptime
    name: "workroom_switch01_uptime"
    update_interval: 60s
    
  - platform: wifi_signal
    name: "workroom_switch01_wifi"
    update_interval: 60s
  
text_sensor:
  - platform: version
    name: "workroom_switch01_version"
    
binary_sensor:
  - platform: status
    name: "workroom_switch01_status"
    
  - platform: gpio
    pin:
      number: GPIO5
    name: "workroom_wallswitch_right"

  - platform: gpio
    pin:
      number: GPIO13
    name: "workroom_wallswitch_left"

output:
  - platform: gpio
    pin: GPIO4
    id: relay_1
    
  - platform: gpio
    pin: GPIO15
    id: relay_2

light:
  - platform: binary
    name: "workroom_ceiling_light_1"
    output: relay_1

  - platform: binary
    name: "workroom_ceiling_light_2"
    output: relay_2
```

### sensor02.yaml (Simple NodeMCU + DHT22)
```yaml
esphome:
  name: sensor02
  platform: ESP8266
  board: nodemcuv2

<<: !include .config.yaml

sensor:
  - platform: dht
    pin: D5
    model: DHT22
    temperature:
      name: "livingroom_sensor_temperature"
    humidity:
      name: "livingroom_sensor_humidity"
    update_interval: 60s

  - platform: uptime
    name: "sensor02_uptime"
    update_interval: 60s
    
  - platform: wifi_signal
    name: "sensor02_wifi"
    update_interval: 60s
  
text_sensor:
  - platform: version
    name: "sensor02_version"
    
binary_sensor:
  - platform: status
    name: "sensor02_status"
```

**Logs (if applicable):**

```
Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 673, in _async_remove_listener
    self._listeners[event_type].remove(listener)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 504, in __eq__
    if self.unique_id is None or other.unique_id is None:
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 548, in unique_id
    if not self._static_info.unique_id:
  File "/usr/src/homeassistant/homeassistant/components/esphome/binary_sensor.py", line 32, in _static_info
    return super()._static_info
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 516, in _static_info
    return self._entry_data.info[self._component_key][self._key]
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 512, in _entry_data
    return self.hass.data[DATA_KEY][self._entry_id]
KeyError: 'f09efc14f0734df585c5d04704685864'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 426, in async_remove_entity
    await self.entities[entity_id].async_remove()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 441, in async_remove
    self._on_remove.pop()()
  File "/usr/src/homeassistant/homeassistant/core.py", line 616, in remove_listener
    self._async_remove_listener(event_type, listener)
  File "/usr/src/homeassistant/homeassistant/core.py", line 681, in _async_remove_listener
    _LOGGER.warning("Unable to remove unknown listener %s", listener)
  File "/usr/local/lib/python3.7/logging/__init__.py", line 1390, in warning
    self._log(WARNING, msg, args, **kwargs)
  File "/usr/local/lib/python3.7/logging/__init__.py", line 1514, in _log
    self.handle(record)
  File "/usr/local/lib/python3.7/logging/__init__.py", line 1524, in handle
    self.callHandlers(record)
  File "/usr/local/lib/python3.7/logging/__init__.py", line 1586, in callHandlers
    hdlr.handle(record)
  File "/usr/local/lib/python3.7/logging/__init__.py", line 894, in handle
    self.emit(record)
  File "/usr/src/homeassistant/homeassistant/components/system_log/__init__.py", line 176, in emit
    record, stack, _figure_out_source(record, stack, self.hass)
  File "/usr/src/homeassistant/homeassistant/components/system_log/__init__.py", line 102, in __init__
    self.message = record.getMessage()
  File "/usr/local/lib/python3.7/logging/__init__.py", line 369, in getMessage
    msg = msg % self.args
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 519, in __repr__
    return "<Entity {}: {}>".format(self.name, self.state)
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 562, in name
    return self._static_info.name
  File "/usr/src/homeassistant/homeassistant/components/esphome/binary_sensor.py", line 32, in _static_info
    return super()._static_info
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 516, in _static_info
    return self._entry_data.info[self._component_key][self._key]
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 512, in _entry_data
    return self.hass.data[DATA_KEY][self._entry_id]
KeyError: 'f09efc14f0734df585c5d04704685864'
```

**Additional information and things you've tried:**
* checked network configuration
* readded devices as hass esphome integration
* nothing but to find the log

voydz, 2019-08-28:
Hey @glmnet ,
thank you for the quick response. 

1. Accessing the ESPhome devices through the browser works fine.
2. The connection quality as reported by esphome logs vary between -44dB to -66dB depending on device. Not the best but it seems to work.
3. No OTA problems at all. I can flash them whenever I want.

I did a bit more research and was curious if this issue might be coherent to https://community.home-assistant.io/t/name-resolution-problem-since-0-97-2/132201

Where are those Hass integration configurations saved? I wanted to check what IP/Hostname I gave my ESPhome devices. If I did omit `.local` this might actually be a problem, according to the thread above.

voydz, 2019-08-28:
Ok, I am actually stupid. I tried to readd a faulty ESPhome device again, but deleting the "old" entry in the integrations panel first. Thus no error about colliding entity_ids. Now it is working again.

But that tells me, that it might be actually an mDNS issue. I think I might have added those devices without `.local` and thus the correct IP address could not be resolved after the 0.97.2 issue posted above.

@glmnet thanks for your time and help anyway :)

## How to Fix

1. Delete the old ESPhome integration in home-assistant
2. Readd the ESPhome integration using `<devicename>.local` or a static ip (see esphome docs for this)
3. Restart home-assistant
4. Make sure the entity_ids match or adjust them in the integration config in home-assistant

closed 2019-08-28 by voydz

OttoWinter, 2019-08-29:
Still this should not necessarily happen, the error points to the entry data being destroyed before the entities are unregistered, which could happen due to two reasons:

 1. Something in HA changed recently with respect to the order of calls in `async_unload_entry`
 2. Something messed up HA's entry_data file in .storage