esphome/issues #629: semantic-version fails when Installing ESPHome

ikout, 2019-08-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

RPi3 with raspbian buster

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

Clean install

**Description of problem:**
I'm doing a new install on a RPi3 with raspbian buster and "sudo pip install esphome" but the installation fails/stops at the following point:
```
Collecting semantic-version<3,>=2.5.0 (from platformio>=3.6.5->esphome)
  Using cached https://www.piwheels.org/simple/semantic-version/semantic_version-2.7.1-py2.py3-none-any.whl
semantic-version requires Python '>=3.4' but the running Python is 2.7.16
```

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2019-08-29:
This is an issue with a subpackage mostly, see https://github.com/rbarrois/python-semanticversion/commit/1655f9cdd72dd3e6dcaf071e84aa5e897e06d39d#commitcomment-34871066

I'll close this to track the issue there.

closed 2019-08-29 by OttoWinter

ikout, 2019-08-29:
Issue is fixed with the 2.8.0 version as mentioned at: https://github.com/rbarrois/python-semanticversion/issues/78

Just did another attempt to install ESPHome and it worked since version-2.8.0 was available: 
```
Collecting semantic-version<3,>=2.5.0 (from platformio>=3.6.5->esphome)
  Downloading https://files.pythonhosted.org/packages/7b/3d/13cd8dd752fd172caa57e6428dff3014643301dc9b5b30752d7b86fe952f/semantic_version-2.8.0-py2.py3-none-any.whl
```

Thanks!