esphome/issues #630: Zero Crossing not being detected

HFsi, 2019-08-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU ESP8266 and ESP32

**Affected component:**

https://esphome.io/components/binary_sensor/gpio.html

**Description of problem:**
I´ve a RobotDyn AC Dimmer, and I can control the PWM pin to turn on and off, but I´m not able to read the pulses from the ZC pin to start correctly the dimming

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO25
      mode: INPUT_PULLUP
      inverted: True
    name: "zc"
    on_release:
      then:
        - switch.turn_on: pwm
        - delay: 1 ms
        - switch.turn_off: pwm

**Logs (if applicable):**
<!--
Please copy the debug log here. If possible, also connect to the ESP over USB and copy those logs into the backticks.
-->
```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

The pin is working since I tested it with a LED and GND, and it brightness is lower than with VCC and GND, so I assume that it´s receiving power through the generated pulses, thus it shines less than with continuous power from VCC

This is the link of the module I´m using:
https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

glmnet, 2019-08-29:
You should not use PWM, an standard output is required here.

**I'm not 100% sure about what I'm going to explain here, if anyone knows better please advice**
The way TRIACs works they turn on with a signal you send, then they will turn off if the signal is no longer present once the AC Waves goes thru the 0 crossing.
So once it is "on" you cannot turn it off, it will turn by itself off, so no sense to use a pwm signal.
About timming:
Assume the wave is 60hz, that is 1000 milliseconds / 60 hz = 16.6 ms per full wave cycle so actually just 8.3 ms per half cycle. Binary sensors works by sampling the input pin as quickly as the ESP can, if the ESP misses the pulse your code will not be triggered, I don't know if this is a problem so far, in this cases you usually use an interrupt, that is a hardware "device" watching the pin and interrupting the CPU to work on it.
Now remember the triac turns itself off at 0 crossing, then if you turn it on as soon as you see it crossing, you'll effectively turn it on for the whole cycle, so what you can do is to wait some time after you see the zero crossing and then turn on the triac.

Assuming your on_release works, you can have your script do something like first thing delay: 4ms then turn on the output delay 1ms and then turn off the output. 
If that works you should have a light at 50% on.

```yaml
    on_release:
      then:
        - delay: 4 ms
        - switch.turn_on: gpio_output
        - delay: 1 ms
        - switch.turn_off: gpio_output
``` 

Ideally this should be done with an integration that take in account the maths for timmings and the interrupt pin attachment.
Again I'm writing all this with no experience with triacs at all, so beware.

OttoWinter, 2019-08-29:
> read the pulses from the ZC pin to start correctly the dimming

`binary_sensor.gpio` polls the pin in the main esphome loop (60Hz), so if the ZC pin has a very short pulse length (which is very likely) this will not work.

For an AC dimmer you need a specialized integration (for example custom output platform), you cannot do it straight from YAML. It's probably even relatively simple to create a custom component for it (as in LOC), someone just needs to do it. see also the AC dimmer feature request discussion for progress.

This is the correct behavior of the GPIO binary sensor, it's only meant for monitoring this like buttons or window switches, not for high-speed data.

closed 2019-08-29 by OttoWinter

HFsi, 2019-08-30:
I'm making the custom component....so far so good, but the attachInterupt function doesn't compile. Is this mode supported by the custom component?

glmnet, 2019-08-30:
do you wanna share your code?
the function is `attach_interrupt(...`

HFsi, 2019-08-30:
Sorry, I'll folow the thread in the pull requests, I believe this is not the right place since it's not an issue