esphome/issues #632: esp32 always rebooting after flashing esphome firmware

thylive, 2019-08-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
hassio

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32

**Affected component:**

ESPHome-flasher,flashing

**Description of problem:**

I use esphome firmware flash it to my  esp32 , i just find it rebooting all the time , i do not know where am wrong

PASTE DEBUG LOG HERE

[12:12:14.606] rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:12:14.606] configsip: 0, SPIWP:0xee
[12:12:14.606] cl
[12:12:14.657] k_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:12:14.657] mode:QIO, clock div:2
[12:12:14.657] load:0x3f400020,len:236584
[12:12:14.657] ets Jun  8 2016 00:22:57
[12:12:15.005] 
[12:12:15.005] rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:12:15.015] configsip: 0, SPIWP:0xee
[12:12:15.015] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:12:15.022] mode:QIO, clock div:2
[12:12:15.022] load:0x3f400020,len:236584
[12:12:15.022] 
[12:12:15.416] ets Jun  8 2016 00:22:57
[12:12:15.416] 
[12:12:15.416] rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:12:15.416] configsip: 0, SPIWP:0xee
[12:12:15.416] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:12:15.426] mode:QIO, clock div:2
[12:12:15.426] load:0x3f400020,len:236584
[12:12:15.426] ets Jun  8 2016 00:22:57
[12:12:15.816] 
[12:12:15.816] rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:12:15.827] configsip: 0, SPIWP:0xee
[12:12:15.827] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:12:15.827] mode:QIO, clock div:2
[12:12:15.832] load:0x3f400020,len:236584
[12:12:15.832] 
[12:12:16.216] ets Jun  8 2016 00:22:57
[12:12:16.234] rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:12:16.236] configsip: 0, SPIWP:0xee
[12:12:16.236] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:12:16.236] mode:QIO, clock div:2
[12:12:16.236] load:0x3f400020,len:236584
[12:12:16.236] 
[13:16:23.336] ets Jun  8 2016 00:22:57
[13:16:23.336] 
[13:16:23.336] rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:16:23.344] flash read err, 1000
[13:16:23.344] ets_main.c 371 
[13:16:23.344] ets Jun  8 2016 00:22:57
[13:16:23.737] 
[13:16:23.737] rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:16:23.745] flash read err, 1000
[13:16:23.745] ets_main.c 371 
[13:16:23.745] 
[13:16:24.137] ets Jun  8 2016 00:22:57
[13:16:24.147] 
[13:16:24.147] rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:16:24.147] flash read err, 1000
[13:16:24.147] ets_main.c 371 
[13:16:24.150] ets Jun  8 2016 00:22:57
[... 770 lines cut ...]
[13:41:45.528] rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:41:45.528] configsip: 0, SPIWP:0xee
[13:41:45.528] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:41:45.537] mode:DIO, clock div:2
[13:41:45.541] load:0x3f400020,len:188412
[13:41:45.541] 
[13:41:45.927] ets Jun  8 2016 00:22:57
[13:41:45.927] 
[13:41:45.927] rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:41:45.927] configsip: 0, SPIWP:0xee
[13:41:45.937] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:41:45.943] mode:DIO, clock div:2
[13:41:45.943] load:0x3f400020,len:188412
[13:41:45.943] ets Jun  8 2016 00:22:57
[13:41:46.327] 
[13:41:46.327] rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:41:46.337] configsip: 0, SPIWP:0xee
[13:41:46.337] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:41:46.346] mode:DIO, clock div:2
[13:41:46.346] load:0x3f400020,len:188412
[13:41:46.346] 

**Additional information and things you've tried:**

Not relevant.

thylive, 2019-08-29:
Problem-relevant YAML-configuration entries:

esphome:
  name: esp_gateway
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "xxxxxxxxx"
  password: "xxxxxxx"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxxxxx"

ota:
  password: "xxxxxxx"

Dbroek, 2019-08-30:
This issue has also been reported in ticket "ESP32 Bootloop #598".

OttoWinter, 2019-08-31:
Closing as duplicate of #598

closed 2019-08-31 by OttoWinter