esphome/issues #634: ESP32 high power usage in deep sleep.

craigueh, 2019-08-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

Deep Sleep

**Description of problem:**
I have an issue that happens across several different ESP32 devices all consuming high power when in deep sleep mode. This makes them unusable for battery projects as typical 3000ma batteries will drain in a few weeks or even days sometimes.

Similar to this issue - espressif/arduino-esp32#1113 I have for example a Lolin32 board which is capable of 150 μA in deep sleep but mine draws 1.7ma. I have other modules from different manufacturers all the same issue with 1.5ma to 2.5ma absolutely minimum when in deep sleep but far from the microamp draw these boards are capable of.

To eliminate outside influence I have an ESP32 board with uart chip, LDO, LEDs and all unnecessary components removed leaving basically just the ESP32 chip and some jumper wires for power. It too draws 1.5ma in deep sleep. This setup should be capable of sub 100μA.

So my conclusion is that ESPhome must be leaving the wifi/ble powered on as described in the link above.

Is there any way to issue a wifi turn off command through esphome? The solutions in the link are all based on Arduine IDE native commands.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

brandond, 2019-08-30:
Can you provide the YAML your using, as requested? Otherwise it's hard to know if you're using the deep_sleep component correctly.

craigueh, 2019-08-30:
Hi 

I think I have just figured it out. The issue is not related to WiFi after all. It’s the wake up mode of the gpio pin selected. Once i changed to esp32_ext1_wakeup mode all peripherals are powered off correctly in deep sleep and now it’s consuming just 135μA.

glmnet, 2019-08-30:
Maybe something to add to docs?

OttoWinter, 2019-08-31:
@craigueh I doubt that's it - ext1 is just a different wakeup *source* within the RTC unit. It shouldn't affect power use much. Also, I doubt wifi can use 1ma - the wifi peripheral usually uses more along the lines of 100ma.

our30K, 2020-05-27:
using Arduino's wifi turn on methods leads to ADC somehow start consuming power. 
so when turn off the wifi, it should turn off the ADC too. to save power.  It is the ADC consumes that 1mA you are experiencing

see my post in :
[(https://github.com/espressif/arduino-esp32/issues/1113)]

closed 2020-12-05 by stale[bot]

nikistring, 2021-08-10:
Hi guys, I'm creating a weather station with an esp32 board from wavehsare, my problem is that when I call the deep sleep, even after 
`adc_power_off();
  //esp_bluedroid_disable();
  esp_wifi_stop();
  esp_bt_controller_disable();

  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
`
it keeps consuming 6.8 mA, I'm struggling because I can't make it work under that value. I wonder if any of you has any clue.
Thank you in advance for the help.