esphome/issues #636: ESP-01 OTA does not work

EspenT, 2019-08-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
NUC with Hass.io and Laptop with Ubuntu

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**
Unable to update firmware over OTA

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: 32_mb_test
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "20B_IOT"
  password: "87654321"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```

**Logs (if applicable):**

```
INFO Uploading 32_mb_test/.pioenvs/32_mb_test/firmware.bin (369392 bytes)
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)

```

**Additional information and things you've tried:**

When trying OTA to a esp-01 I got this error.
```
INFO Uploading 32_mb_test/.pioenvs/32_mb_test/firmware.bin (369392 bytes)
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (remove everything except ota)
```

As requested I tried to remove everything exept ota. It did not make a difference.
OTA is crucial in my project so i ordered some winbond w25q32 Flash ICs. 

While waiting for those i discovered two things
1 Using esp01_1m as bord type it should be plenty of space for a 300k but it does not event try.
2 Sometimes it will actually upload the file and everything works as i should.
```
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  39.8% (used 32568 bytes from 81920 bytes)
PROGRAM: [===       ]  32.3% (used 330436 bytes from 1023984 bytes)
========================= [SUCCESS] Took 2.05 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of garage_d00r_sensor.local
INFO  -> 172.20.1.57
INFO Uploading /data/garage_d00r_sensor/.pioenvs/garage_d00r_sensor/firmware.bin (334576 bytes)
```

I then received the 4MegaByte flash IC, but most of the time it will not try to do the OTA.
from CLI and HA/WEB does not seem to make any difference.

brandond, 2019-08-30:
You need to have at least 50% of your flash space available to do OTA, since it temporarily stores a second copy of the firmware on flash. You also need to be using a board definition that configures a flash layour that matches your flash chip, since the board definitions determine the flash memory allocations for user code and OTA space.

If you have a 1MB chip, you should use `esp01_1m` - the `esp01` board definition is set up for a 512KB chip and will not reserve sufficient space for OTA, even if the actual chip is larger.

Similarly, if you've replaced the SPI flash with a 4MB chip then you should use the `esp07` board definition, or another one configured for 4MB flash.

So:
* Pick a board definition that matches your SPI flash size
* Upload the initial firmware via UART (to force a rewrite of the flash layout)
* Try doing an OTA update

OttoWinter, 2019-08-31:
@brandond Is right - this is all due to the choice of the `board` setting.

You need to select the correct board (see also https://docs.platformio.org/en/latest/platforms/espressif8266.html) that matches the size on your device.

Also note that if you flashed the device with a `esp01` binary you can't upload a 4mb version (because the binary accepting the OTA doesn't know the device has 4mb of storage). So re-flash with 4mb binary via serial and then OTA should work.

closed 2019-08-31 by OttoWinter