esphome/issues #637: I2C sensors not working (HDC1080, htu21d)

michalszymura, 2019-08-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

https://esphome.io/components/sensor/hdc1080.html
https://esphome.io/components/sensor/htu21d.html
**Description of problem:**
Sensors are seen by the I2C scan, but communication with sensors fail. (tested one at a time since they have the same address). I2C display works without any problems.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: sensor_node01
  platform: ESP8266
  board: nodemcuv2
  
wifi:
  ssid: "removed"
  password: "removed"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'removed'

ota:
  password: 'removed'

i2c:
  sda: D1
  scl: D2
  scan: True
  frequency: 100kHz
  
sensor:
  - platform: hdc1080
    address: 0x40
    # setup_priority: -100
    temperature:
      name: "HDC1080 Temperature"
    humidity:
      name: "HDC1080 Pressure"
    update_interval: 5s
  - platform: homeassistant
    id: inside_temperature
    entity_id: sensor.bedroom_temperature
    internal: true
  # - platform: htu21d
  #   temperature:
  #     name: "htu21d-01 Temperature"
  #   humidity:
  #     name: "htu21d-01 Humidity"
  #   update_interval: 3s

font:
  - file: 'arial.ttf'
    id: font1
    size: 14
    
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x32"
    # model: "SH1106 128x64"
    # reset_pin: D0
    address: 0x3C
    lambda: |-
      if (id(inside_temperature).has_state()) {
        it.printf(0, 0, id(font1), TextAlign::TOP_LEFT,  "%.1f°", id(inside_temperature).state);
      }
```

**Logs (if applicable):**

```
INFO Successfully connected to sensor_node01.local
[21:14:13][I][app:096]: esphome version 1.13.6 compiled on Aug 31 2019, 20:58:54
[21:14:13][C][wifi:372]: WiFi:
[21:14:13][C][wifi:254]:   SSID: [redacted]
[21:14:13][C][wifi:255]:   IP Address: 192.168.0.201
[21:14:13][C][wifi:257]:   BSSID: [redacted]
[21:14:13][C][wifi:258]:   Hostname: 'sensor_node01'
[21:14:13][C][wifi:262]:   Signal strength: -50 dB ▂▄▆█
[21:14:13][C][wifi:263]:   Channel: 13
[21:14:13][C][wifi:264]:   Subnet: 255.255.255.0
[21:14:13][C][wifi:265]:   Gateway: 192.168.0.1
[21:14:13][C][wifi:266]:   DNS1: 192.168.0.3
[21:14:13][C][wifi:267]:   DNS2: 0.0.0.0
[21:14:13][C][i2c:028]: I2C Bus:
[21:14:13][C][i2c:029]:   SDA Pin: GPIO5
[21:14:13][C][i2c:030]:   SCL Pin: GPIO4
[21:14:13][C][i2c:031]:   Frequency: 100000 Hz
[21:14:13][I][i2c:033]: Scanning i2c bus for active devices...
[21:14:13][I][i2c:040]: Found i2c device at address 0x3C
[21:14:13][I][i2c:040]: Found i2c device at address 0x40
[21:14:13][C][logger:137]: Logger:
[21:14:13][C][logger:138]:   Level: DEBUG
[21:14:13][C][logger:139]:   Log Baud Rate: 115200
[21:14:13][C][logger:140]:   Hardware UART: UART0
[21:14:13][C][hdc1080:028]: HDC1080:
[21:14:13][C][hdc1080:029]:   Address: 0x40
[21:14:13][E][hdc1080:031]: Communication with HDC1080 failed!
[21:14:13][C][hdc1080:033]:   Update Interval: 5.0s
[21:14:13][C][hdc1080:034]:   Temperature 'HDC1080 Temperature'
[21:14:13][C][hdc1080:034]:     Unit of Measurement: '°C'
[21:14:13][C][hdc1080:034]:     Accuracy Decimals: 1
[21:14:13][C][hdc1080:034]:     Icon: 'mdi:thermometer'
[21:14:13][C][hdc1080:035]:   Humidity 'HDC1080 Pressure'
[21:14:13][C][hdc1080:035]:     Unit of Measurement: '%'
[21:14:13][C][hdc1080:035]:     Accuracy Decimals: 0
[21:14:13][C][hdc1080:035]:     Icon: 'mdi:water-percent'
[21:14:13][C][ssd1306_i2c:023]: I2C SSD1306
[21:14:13][C][ssd1306_i2c:023]:   Rotations: 0 °
[21:14:13][C][ssd1306_i2c:023]:   Dimensions: 128px x 32px
[21:14:13][C][ssd1306_i2c:024]:   Address: 0x3C
[21:14:13][C][ssd1306_i2c:025]:   Model: SSD1306 128x32
[21:14:13][C][ssd1306_i2c:027]:   External VCC: NO
[21:14:13][C][ssd1306_i2c:028]:   Update Interval: 1.0s
[21:14:13][C][ota:029]: Over-The-Air Updates:
[21:14:13][C][ota:030]:   Address: sensor_node01.local:8266
[21:14:13][C][ota:032]:   Using Password.
[21:14:13][C][api:103]: API Server:
[21:14:13][C][api:104]:   Address: sensor_node01.local:6053
[21:14:13][C][homeassistant.sensor:024]: Homeassistant Sensor 'inside_temperature'
[21:14:13][C][homeassistant.sensor:024]:   Unit of Measurement: ''
[21:14:13][C][homeassistant.sensor:024]:   Accuracy Decimals: 1
[21:14:13][C][homeassistant.sensor:025]:   Entity ID: 'sensor.bedroom_temperature'
[21:14:20][D][homeassistant.sensor:019]: 'sensor.bedroom_temperature': Got state 25.80
[21:14:20][D][sensor:092]: 'inside_temperature': Sending state 25.80000  with 1 decimals of accuracy
[21:15:49][I][ota:046]: Boot seems successful, resetting boot loop counter.

```

**Additional information and things you've tried:**
Tried on 3 ESP8266 boards.
Tried sensors one at a time.
I2C display works fine.
Tried different I2C speeds and pins.
Sensors work fine on the same ESP with arduino sketch.

Hsxsky, 2019-09-02:
Same problem.
ESP32 D1 R32

Tested at 2 board and 3 sensor 2 different port set.
Add to the test a TSL2561 at same port.

[14:06:32][C][i2c:028]: I2C Bus:
[14:06:32][C][i2c:029]:   SDA Pin: GPIO25
[14:06:32][C][i2c:030]:   SCL Pin: GPIO17
[14:06:32][C][i2c:031]:   Frequency: 50000 Hz
[14:06:32][I][i2c:033]: Scanning i2c bus for active devices...
[14:06:32][I][i2c:040]: Found i2c device at address 0x39
[14:06:32][I][i2c:040]: Found i2c device at address 0x40
...
[14:06:32][C][tsl2561:040]: TSL2561 'TSL2561 Ambient Light'
[14:06:32][C][tsl2561:040]:   Unit of Measurement: 'lx'
[14:06:32][C][tsl2561:040]:   Accuracy Decimals: 1
[14:06:32][C][tsl2561:040]:   Icon: 'mdi:brightness-5'
[14:06:32][C][tsl2561:041]:   Address: 0x39
[14:06:32][C][tsl2561:048]:   Gain: 1x
[14:06:32][C][tsl2561:049]:   Integration Time: 402.0 ms
[14:06:32][C][tsl2561:051]:   Update Interval: 10.0s
[14:06:32][C][hdc1080:028]: HDC1080:
[14:06:32][C][hdc1080:029]:   Address: 0x40
[14:06:32][E][hdc1080:031]: Communication with HDC1080 failed!
[14:06:32][C][hdc1080:033]:   Update Interval: 5.0s
[14:06:32][C][hdc1080:034]:   Temperature 'HDC1080 Temperature'
[14:06:32][C][hdc1080:034]:     Unit of Measurement: '°C'
[14:06:32][C][hdc1080:034]:     Accuracy Decimals: 1
[14:06:32][C][hdc1080:034]:     Icon: 'mdi:thermometer'
[14:06:32][C][hdc1080:035]:   Humidity 'HDC1080 Pressure'
[14:06:32][C][hdc1080:035]:     Unit of Measurement: '%'
[14:06:32][C][hdc1080:035]:     Accuracy Decimals: 0
[14:06:32][C][hdc1080:035]:     Icon: 'mdi:water-percent'

TSL2561 working:

[14:06:36][D][sensor:092]: 'TSL2561 Ambient Light': Sending state 17.97440 lx with 1 decimals of accuracy
[14:06:46][D][tsl2561:126]: Got illuminance=18.0lx

Hsxsky, 2019-09-06:
During the compile, there is a warning, when VERY_VERBOSE loggin is set.

Compiling .pioenvs/dolgozo/FrameworkArduino/esp32-hal-i2c.c.o
/home/t/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c.c: In function 'i2cDumpFifo':
/home/t/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c.c:415:35: warning: ' ' flag used with '%X' gnu_printf format [-Wformat=]
       } else k += sprintf(&buf[k],"% 4X ",fifoBuffer[i]);
                                   ^
/home/t/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c.c:415:35: warning: ' ' flag used with '%X' gnu_printf format [-Wformat=]

Therefore it can not debug.

Hsxsky, 2019-09-06:
After upgrading to bleeding edge version.
https://beta.esphome.io/guides/faq.html?highlight=beta#how-do-i-use-the-latest-bleeding-edge-version

[I][app:028]: Running through setup()...
[V][app:029]: Sorting components by setup priority...
[V][esp32-hal-i2c.c:1483] i2cInit(): num=0 sda=25 scl=17 freq=0
[V][esp32-hal-i2c.c:1677] i2cSetFrequency(): freq=100000Hz
[V][esp32-hal-i2c.c:1692] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[V][esp32-hal-i2c.c:1698] i2cSetFrequency(): Fifo delta=1
[V][esp32-hal-i2c.c:1677] i2cSetFrequency(): freq=50000Hz
[V][esp32-hal-i2c.c:1692] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=50000Hz
[V][esp32-hal-i2c.c:1698] i2cSetFrequency(): Fifo delta=1
[C][hdc1080:015]: Setting up HDC1080...
[VV][i2c:056]: Beginning Transmission to 0x40:
[VV][i2c:093]:     Writing 0b00000010 (0x02)
[VV][i2c:093]:     Writing 0b00000000 (0x00)
[VV][i2c:093]:     Writing 0b00000000 (0x00)
[VV][i2c:061]:     Transmission ended. Status code: 0x00
[E][component:092]: Component was marked as failed.

http://www.ti.com/lit/ds/symlink/hdc1080.pdf
According the datasheet commands is ok.

Hsxsky, 2019-09-06:
Solution:
/usr/lib/python2.7/site-packages/esphome/components/hdc1080/hdc1080.cpp

There is an HDC1080Component::setup function.

  if (this->write_bytes(HDC1080_CMD_CONFIGURATION, data, 2)) {
    this->mark_failed();
    return;
  }

I suppose this->write_bytes (i2c memory write) return 1 if OK and return 0 at ERROR.
If you negate this statement, the result will be ok. 

Changed code:
*****
void HDC1080Component::setup() {
  ESP_LOGCONFIG(TAG, "Setting up HDC1080...");

  const uint8_t data[2] = {
      0b00000000,  // resolution 14bit for both humidity and temperature
      0b00000000   // reserved
  };

  if (!this->write_bytes(HDC1080_CMD_CONFIGURATION, data, 2)) {
    this->mark_failed();
    return;
  }
}
****

Result in my environment:
[VV][i2c:056]: Beginning Transmission to 0x40:
[VV][i2c:093]:     Writing 0b00000001 (0x01)
[VV][i2c:061]:     Transmission ended. Status code: 0x00
[VV][i2c:083]: Requesting 2 bytes from 0x40:
[VV][i2c:126]:     Received 0b1000000110110000 (0x81B0)
[V][sensor:013]: 'Living Room Pressure': Received new state 50.659184
[D][sensor:092]: 'Living Room Pressure': Sending state 50.65918 % with 0 decimals of accuracy
[D][hdc1080:054]: Got temperature=27.8°C humidity=50.7%

Hsxsky, 2019-09-06:
There is some other i2c sensor esphome.
For example TSL2561.

Is checked tsl2561.cpp.

  if (!this->tsl2561_read_byte(TSL2561_REGISTER_TIMING, &timing)) {
    this->mark_failed();
    return;
  }

As I see here, need to be an negation (!). :)

To solve the problem you doesn't need bleeding edge version. I used it only for debugging.

xppx99, 2019-09-08:
Hello Hsxsky,
Is there a way to apply fix this for those of us who use the hassio plugin? 
Thanks

larsenglund, 2019-10-14:
> Hello Hsxsky,
> Is there a way to apply fix this for those of us who use the hassio plugin?
> Thanks
 @Hsxsky

Hsxsky, 2019-10-17:
Hello larsenglund and xppx99,

As I see the Issue is merged to the development (bleeding edge) version.
You can change to that version in the Hass.io  add-on repository.

Manual is here for developer version:
https://next.esphome.io/

larsenglund, 2019-10-17:
@Hsxsky Awesome, it's working now, thanks!

OttoWinter, 2019-11-07:
Has been fixed in 1.14, closing.

closed 2019-11-07 by OttoWinter