esphome/issues #638: UART interface is super noisy for gap-delimited protocols.

fake-name, 2019-09-01:
I have a piece of software that uses gap-delimited serial messages. Basically, you read until you don't get any bytes for a period of time, and then parse the received message. 

Right now, every time I try to read from the device, I get

```
[00:36:56][E][uart:090]: Reading from UART timed out at byte 0!
```

Now, if there was a way to turn this off, that'd be fine, but there isn't. You're stuck with it. You can't change the hard-coded timeouts either.

Basically, a timeout should not be assumed to be an error, and it shouldn't be logged as such. 

------

Irrelevant issue template cruft:

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

brandond, 2019-09-01:
Call `available()` to see if there's any data in the RX buffer instead of blindly calling `read()`.

OttoWinter, 2019-09-01:
@brandond Is right, with available you can check how much is in the receive buffer. That is the correct way to read from the bus - the timeout is there to protect the main esphome loop from getting caught up waiting for data that never arrives.

closed 2019-09-01 by OttoWinter