esphome/issues #639: Captive Portal Not Working

alexyao2015, 2019-09-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, Nodemcu
**Affected component:**

AP Captive Portal

**Description of problem:**
When ESPhome is unable to find the wifi network, it creates an AP, however connecting to it doe s not allow for reflashing. Serial just shows it constantly scanning for a network with no way to access the web server even though it is enabled in the config. Accessing 192.168.4.1 while connected to the AP will not load.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: demo
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "I do not exist"
  password: "fake password"

  ap:
   ssid: "DEMO"

# Enable logging
logger:
  baud_rate: 115200

web_server:
  port: 80

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Related to esphome/feature-requests#86

alexyao2015, 2019-09-01:
Seems like the captive portal hasn't been released yet...

glmnet, 2019-09-01:
Never tried the captive portal but if you add here the logs (from usb) we might get going faster

alexyao2015, 2019-09-01:
I've found that the captive portal doesn't exist yet in the master branch but it does in dev. So the issue relates to the web server and ap components, not the captive portal.

OttoWinter, 2019-09-01:
There have been many changes to get captive_portal working - and they overlap with what you're seeing.

It doesn't really make sense to look into this issue because it's probably already working with captive_portal.

If you experience with dev and captive_portal then please create a bug report for that.

closed 2019-09-01 by OttoWinter