esphome/issues #640: Using GPIO0 / GPIO2 pins on the ESP-01

manju-rn, 2019-09-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESP8266  ESP-01
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP-01
**Affected component:**

Remote Transmitter https://www.esphome.io/components/remote_transmitter.html
However, this would affect any component if that component pull the GPIO / GPIO2 pins LOW

**Description of problem:**
ESP-01 does not boot when a remote transmitter (or any other component) for that matter connected to the GPIO2 pin. However, if the component is attached (signal pin for Remote transmitter in this case) after the ESP is booted successfully, then everything works fine including the operation of the Remote Transmitter.

Please see the below test cases for details:
Negative:
1. ESP-01 is connected via UART adapter and have a Remote Transmitter (3 Pin) signal pin connected to the GPIO 2 of the ESP-01. 
2. ESP-01 shows a constant blue light and does not boot up.  
Believe that the GPIO 2 is pulled Low and hence does not allow the ESP to boot up.  

Positive:
1. ESP-01 is connected via UART adapter.  Remote Transmitter is NOT yet connected to the GPIO2.
2. After ESP-01 successfully boots up, the Signal pin of the Remote transmitter is then connected to the GPIO2 pin.  This works successfully.

So the question is how do I make sure that the GPIO 2 pin is not Low during boot up and subsequently made after boot up?  I can do that in the Arduino code, but how do we specify this in the YAML file?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: esp8266node
  platform: ESP8266
  board: esp01_1m
  includes:

remote_transmitter:
  pin: 2
  carrier_duty_percent: 50%

wifi:
  networks:
  - ssid: !secret myssid
    password: !secret mypass
    hidden: True
  ap:
    ssid: !secret standssid
    password: !secret standpass

web_server:
  port: 8080

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:

switch:
  - platform: template
    name: espfanspeed
    turn_on_action:
       - remote_transmitter.transmit_raw:
           code: [1258, -442,  1240, -442,  398, -1284,  1242, -442,  1240, -442,  400, -1284,  398, -1284,  398, -1284,  398, -1284,  398, -1284,  1242, -442,  398, -8018]
           carrier_frequency: 38kHz
           repeat: 
             times: 3
sensor:

**Logs (if applicable):**
<!--
Please copy the debug log here. If possible, also connect to the ESP over USB and copy those logs into the backticks.
-->
```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Tried to directly hard code the GPIO 2 to low in the main.cpp before the App.setup is called.

  // ========= YOU CAN EDIT AFTER THIS LINE =========
  digitalWrite(2,LOW);
  App.setup();
}

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

manju-rn, 2019-09-01:
Please note that the above issue in Remote Transmitter only happens during the first boot / cold boot. Subsequent resets or upload of the new binary does not affect it.

OttoWinter, 2019-09-01:
As you say yourself, this happens in a cold boot phase - in this phase no code from esphome whatsoever runs. It is not possible to fix this in software, you have to use resistors to pull the pin to the correct state on bootup.

closed 2019-09-01 by OttoWinter

manju-rn, 2019-09-02:
Just to so who are watching this thread.  This is resolved. I just used the RX pin (GPIO 3) as this is the only other pin that is not pulled high or low during the cold boot.  GPIO 2 has an issue with any remote transmitters (aka any direct output ) electrically as it tries pass 1.2 V across the pin during boot up. However, GPIO 2 can always be used for the inputs - like Remote Receivers

remote_transmitter:
  pin: 3
  carrier_duty_percent: 50%

brandond, 2019-09-02:
gpio2 controls the boot mode. If you pull it down during power-on or reset the device goes into the serial bootloader instead of running user code. That's hardwired into the chip's firmware.

manju-rn, 2019-09-03:
Yes, am aware of that.  However, what puzzled me is that the same behavior should have occurred for both Remote Transmitter and Remote Receiver, but somehow the Receiver works fine but transmitter does not