esphome/issues #641: Status

tvwerkhoven, 2019-09-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Home-built

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP8266, Wemos D1 mini pro v2.0.0

**Affected component:** esphome dashboard status light/board ping

**Description of problem:** The status light on the esphome dashboard does not appear to work for boards on a different subnet/zones.

I have two network zones:
1. Default: 172.16.0.0/24, access to everything
2. IoT: 172.16.1.0/24, access to LAN only, where I run my IoT devices.

Everything else works:
1. I can ping the board across zones
2. I can reach the esphome webserver across zones
3. I can push OTA updates across zones
4. I can see the status of esphome.io boards on the **same** zone (either default or IoT)

the only thing not working is the status light **across** zones.

**Problem-relevant YAML-configuration entries:** N/A
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):** N/A

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

In the Safari Developer Tools I can see a fetch for `ping` being sent out every second, so some mechanism is working.

carlosmgr, 2019-09-17:
My nodes and my dashboard are on 2 diferent vlans and they are working fine, that is a problem on your rules.

brandond, 2019-09-18:
See status_use_ping option in docs. I believe it's covered in the hassio add-on documentation, if you need to set it when running the dashboard from shell you'll have to set an environment variable I believe.

carlosmgr, 2019-09-23:
I see this question here from time to time, so lets help someone.

ESPhome uses mDNS witch is a multicast protocol, multicast does not cross vlans.

You have 4 options:
- Use a Firewall/Router that permits mDNS to cross vlans
- Use Avahi reflector on a RPI connect to all vlans
- On hass.io add the "status_use_ping=true"
- On esphome docker dashboard use the env "ESPHOME_DASHBOARD_USE_PING=true"

brandond, 2019-09-23:
@carlosmgr that's a great writeup. Would you mind putting in a PR for the docs?

tvwerkhoven, 2019-09-24:
@carlosmgr Thanks! Solution 4 worked for me.

closed 2019-10-13 by OttoWinter