esphome/issues #642: No running services when in AP mode

alfredopironti, 2019-09-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip linux

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 - esp-wrover-kit

**Affected component:**

https://esphome.io/components/wifi.html

**Description of problem:**
I enabled the "ap:" option under the "wifi:" component, with the hope of using it as a recovery method if connection to the main gateway router gets lost. When no connection to the main gateway router can be established, the device correctly sets up the AP, with the specified SSID, and I can connect successfully.

However, no service seems to be running on the esphome device (crucially no web, no OTA, no API), so I'm not sure how to make use of the connection. I've tried three different setups under "ap:" (see below), one without specifying a "manual_ip:", a second where the assigned IP was 192.168.4.4, but the default gateway was 192.168.4.1, and a third where both the assigned IP and the default gateway were set to 192.168.4.4.

In all cases, the device I was using to connect to the esphome AP (a linux laptop) got assigned IP 192.168.4.100, and in all cases I could successfully ping 192.168.4.1 (but surprisingly not 192.168.4.4). However, scanning the entire /24 subnet, no service was running on ports 80, 3232 nor 6053, and the only two running hosts were 192.168.4.1 (even when 192.168.4.4 was set as the manual IP) and 192.168.4.100 (my own laptop).

My expectation was to find at least one device offering the services defined on the esphome configuration file, or at least crucially OTA. As it stands, I'm unable to leverage the functionality of the AP mode. Am I using this feature in the wrong way, or could this be a bug with the current implementation?

**Problem-relevant YAML-configuration entries:**
This is the base configuration for all three setups, enabling web, ota and other services
```yaml
esphome:
  name: <device_name_redacted>
  platform: ESP32
  board: esp-wrover-kit

ota:

web_server:

api:

logger:
```

In the following, the three different ap: setups used. The first with no manual_ip specified
```
wifi:
  ssid: <redacted>
  password: <redacted>
  ap:
    ssid: <recovery_ssid_name>
```
The second with ip 192.168.4.4 but gateway 192.168.4.1
```
wifi:
  ssid: <redacted>
  password: <redacted>
  ap:
    ssid: <recovery_ssid_name>
    manual_ip:
      static_ip: 192.168.4.4
      gateway: 192.168.4.1
      subnet: 255.255.255.0
```
The third with both ip and gateway to 192.168.4.4
```
wifi:
  ssid: <redacted>
  password: <redacted>
  ap:
    ssid: <recovery_ssid_name>
    manual_ip:
      static_ip: 192.168.4.4
      gateway: 192.168.4.4
      subnet: 255.255.255.0
```

**Logs (if applicable):**

Unfortunately, I couldn't get any logs as I wasn't able to connect to the device (albeit I was able to connect to the AP set up by esphome)

**Additional information and things you've tried:**
I haven't been able to dig this further. But I remain available to perform additional tests or provide additional information as needed.

manju-rn, 2019-09-02:
Not sure if you understood the meaning of AP well - in this mode, ESP will broadcast its own AP. As you have enabled the webserver,  it will allow usage of the services configured within the ESP - for e.g., lets say you have connected a switch or a light directly to ESP, it would still trigger through the default buttons in the webserver (or any app you may have written which directly connects to ESP's AP). 
The ESP will NOT show up in the Home Assistant / openHAB or any other.  Hence, any integration through HA / OpenHAB will not be available

glmnet, 2019-09-02:
Which ESPHome version?

I wouldn't expect Home Assistant to work, only because it is not joined to the same network.

However I see no reason why no services are run.
I'd expect a command like

`esphome file.yaml run --upload-port 192.168.4.4`

would OTA properly

alfredopironti, 2019-09-02:
Sorry for the confusion in my wording! I'm using latest esphome 1.13.6.

I do understand the esphome is setting up its own AP, and that my laptop is connecting to that AP. Once connected, I'm joining a network where only my own laptop (192.168.4.100 for what I could see yesterday) and the esphome device are connected.

As I get from your comments, you would also expect that any service configured on the esphome device (maybe OTA, or web server) should be reachable from my laptop over the recovery AP set up by esphome. However, in my experiments, there's no service at all available on 192.168.4.1 or 192.168.4.4 (in fact, the latter gives a "no route to host" error under all three configuration options I've tried, like no host is active at that IP address).

Could you try to reproduce the scenario locally, and kindly share your findings with me? Reproducing should be as easy as follows:
- In any device you have running, add the "ap:" option with a SSID (and make sure they have at least web or ota enabled)
- Turn off the regular gateway AP, so that esphome, failing to connect, will set up its own AP
- Connect your laptop to the recovery AP set up by the esphome, and try interacting with the esphome device (either web or ota, according to its config)

If the issue I'm having is repeatable, your connection attempts at the last step should fail. (Else, I should get enough info to make it work locally ;-) )

Hope this clarifies, and many thanks for helping out!

glmnet, 2019-09-02:
well I have never tried the ap, however you should be able to ping 192.168.4.1 or 192.168.4.4 depending on the ip you give to it, otherwise there is no sense of the ap function at all.

There is coming a new captive portal for ESPHome in v1.14 and if there are issues in 1.13 those won't be fixed in that version as there have been many changes related to that feature.

You might like to try the dev branch too

I can try when I'm back at home in about 8 hours.

alfredopironti, 2019-09-02:
> well I have never tried the ap, however you should be able to ping 192.168.4.1 or 192.168.4.4 depending on the ip you give to it, otherwise there is no sense of the ap function at all.

We're on the same page. In all of the three ap configurations listed above, from my own laptop I was able to ping and get a response from 192.168.4.1. No host at 192.168.4.4 was responding in any scenario. However, when trying to connect to port 80, 3232, 6053 (web server, OTA and API, all enabled in the yaml config) the ports were closed on 192.168.4.1 (and no host was answering at 192.168.4.4). This was the same for all scenarios.

As the AP logic is being reworked in 1.14, it would make perfect sense to wait and see -- in case this issue gets confirmed as repeatable, it would at least remain documented.

Thanks for looking deeper into this. I'll be awaiting news :-)

OttoWinter, 2019-11-07:
Please try again with 1.14.2

alfredopironti, 2019-11-10:
Hi @OttoWinter,

I've now tried with 1.14.2 on a ESP32 module I have at home. I've also enabled the captive portal. The relevant part of the configuration is as follows:

```
wifi:
  ssid: "Non_existent_SSID" #so that the device won't connect
  password: !secret wifi_password
  ap:
    ssid: "device_recovery"
    password: !secret ap_password

captive_portal:
```

After a few minutes, the "device_recovery" SSID shows up and I can eventually connect to it. On connecting to the recovery AP, please note several attempts end up in a connection failure, but say 1 in 10 attempts is successful. Not sure why this happens (I speculate the device is rebooting, but haven't looked into this further, and would probably be a different issue).

The captive portal works fine, and eventually allowed me to change the main wifi SSID, so that the device would finally connect to the home network. I haven't tried uploading a new firmware via the captive portal web interface.

However, to the point of this issue, it still seems the OTA and API ports are closed on the device. Please see below (ran when connected to the "device_recovery" SSID):
```
% nmap -p 80,3232,6053 192.168.4.1
Starting Nmap 7.80 ( https://nmap.org ) at 2019-11-10 16:52 CET
Nmap scan report for 192.168.4.1
Host is up (0.013s latency).

PORT     STATE  SERVICE
80/tcp   open   http
3232/tcp closed mdtp
6053/tcp closed x11

Nmap done: 1 IP address (1 host up) scanned in 14.15 seconds
```
Port 80 (captive portal) is open and working, but no OTA nor API.

This is confirmed by the fact that, when connected to the "device_recovery" SSID, I can't upload the firmware via the command line esptool manually specifying the address. That is:
```
% esphome cfg_file.yaml run --upload-port 192.168.4.1
[snip]
INFO Successfully compiled program.
INFO Connecting to 192.168.4.1
ERROR Connecting to 192.168.4.1:3232 failed: [Errno 61] Connection refused
```

This is now less critical, as captive portal offers a working alternative (albeit less scriptable).

Can you reproduce this issue?

alfredopironti, 2019-11-11:
Giving further thoughts on the matter, one fundamental question is whether we want the services running during normal operation to be active when the recovery AP is up. For example, if I had "web_server:" and "captive_portal:" defined within the same file, such two services would be conflicting to use port 80 when the recovery AP goes up.

In fact, could it even make sense to merge the "web server" and "captive portal" functionalities into a single feature?

Also, we may list under the wifi -> ap setting which services we'd like to have running when the recovery AP is up? Say:
```
wifi:
  ap:
    captive_portal: #this says captive portal will be up when in recovery AP mode, but not in normal mode as no captive_portal: is defined at the top level

ota: #this says ota will be available in normal mode, but not in recovery AP as there's no OTA defined within ap:
```

I don't want to digress over the original issue, but this discussion could be part of the solution maybe?

OttoWinter, 2019-11-12:
Yes, so for a few services, being active during recovery mode doesn't make sense:

 - mqtt: makes no sense, won't be able to connect to broker anyway
 - web server: makes sense, but as you noted captive portal takes precedence.
 - ota: makes semi-sense - but the user could just use captive portal interface anyway.
 - api: makes sense, log function can be useful.

Right now, due to internal architectural reasons, the way it works is:

 - mqtt: disabled while captive portal active
 - web server: disabled while captive portal active
 - ota: disabled while captive portal active
 - api: disabled while captive portal active

> In fact, could it even make sense to merge the "web server" and "captive portal" functionalities into a single feature?

They are, it's just under the hoods in `web_server_base`

> Also, we may list under the wifi -> ap setting which services we'd like to have running when the recovery AP is up?

If you have a captive portal defined in your config, it doesn't make sense not to have captive portal active in AP mode.

alfredopironti, 2019-11-13:
Hi @OttoWinter ,

Thanks for the thorough response.

>> In fact, could it even make sense to merge the "web server" and "captive portal" functionalities into a single feature?

>They are, it's just under the hoods in web_server_base

Interesting! I was wondering if we could offer the same interface in both cases, but I guess it's intentional in recovery mode to have an interface more focused on recovery actions. Do I get it right?

Finally, if I understand correctly, captive_portal can start only in AP recovery mode. Hence, would it make sense to put the "captive_portal:" YAML option under "ap:", rather than top-level? It's somewhat cosmetic, I'll agree ;-)

That said, I think we can consider the issues closed, since the logic between captive portal and other networking features is now pretty clear. As time permit, I may look into a PR on the docs section.

Thanks for your continued support, and for your great, great software. It literally runs my house ;-)

closed 2019-11-13 by alfredopironti